openpnp/openpnp #777: Job runs without nozzle-tip / wrong nozzle-tip attached to the nozzle while using tool-changer

ldpgh, 2018-10-04:
# Problem Report
## Describe the Issue
While starting a Job ....
(1) and no nozzle-tip is assigned to the nozzle the job starts and runs witout any notification.
(2) and a wrong nozzle-tip is assigned there is no event to change to the appropriate nozzle-tip.

## Machine setp
(1) Tool changer
(2) Drag feeder with vision enabled

## Steps to Reproduce
1. Start OpenPnP
2. Click the Power On button.
3. Home the machine
4. Unload nozzle-tip.
5. Load the Job and start it
6. Job fails due to missing nozzle-top. In difference a manual feed operation on the drag-feeder issues a warning after lowering the nozzle ... means right before sending the "pick"-command.

## Expected Result
Tool change to load an appropriate nozzle if no/wrong nozzle-tip is loaded to the nozzle.

## Actual Result
No tool change when starting the job with wrong/no nozzle-tip.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
... files job,board,machine,parts,log  will be added later. (... need some extra work)

IRC discussion

ldpgh, 2018-10-05:
[image]

ldpgh, 2018-10-05:
The file-suffix have been renamed to '.txt' ...
* [board.txt](https://github.com/openpnp/openpnp/files/2449492/board.txt)
* [Job_2018_10_04_nur_r8_bis_r29_enabed.txt](https://github.com/openpnp/openpnp/files/2449500/Job_2018_10_04_nur_r8_bis_r29_enabed.txt)
* [parts.txt](https://github.com/openpnp/openpnp/files/2449503/parts.txt)
* [machine.txt](https://github.com/openpnp/openpnp/files/2449504/machine.txt)
* [packages.txt](https://github.com/openpnp/openpnp/files/2449505/packages.txt)

vonnieda, 2018-10-06:
I am not able to reproduce this. When I run the job you supplied it immediately loads NT1 before picking anything. Here is the log:

```
2018-10-06 10:47:47 SystemLogger INFO: isAllPlaced false
2018-10-06 10:47:47 ReferenceHead DEBUG: H1.moveToSafeZ(0.86)
2018-10-06 10:47:47 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-06 10:47:47 NullDriver DEBUG: moveTo(N1 N1, (NaN, NaN, 42.000000, NaN mm), 0.86)
2018-10-06 10:47:48 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.86)
2018-10-06 10:47:48 NullDriver DEBUG: moveTo(OpenPnpCaptureCameraDown, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:48 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 10:47:48 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_01, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:48 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 10:47:48 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_02, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:48 ReferencePnpJobProcessor DEBUG: Planned placements [N1 N1 -> R19]

NOZZLETIP CHANGE HAPPENS HERE

2018-10-06 10:47:48 ReferencePnpJobProcessor DEBUG: Change NozzleTip on Nozzle N1 N1 from null to NT1 NT1
2018-10-06 10:47:48 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): Start
2018-10-06 10:47:48 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo Start Location
2018-10-06 10:47:48 ReferenceHead DEBUG: H1.moveToSafeZ(0.86)
2018-10-06 10:47:48 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-06 10:47:48 NullDriver DEBUG: moveTo(N1 N1, (NaN, NaN, 42.000000, NaN mm), 0.86)
2018-10-06 10:47:49 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.86)
2018-10-06 10:47:49 NullDriver DEBUG: moveTo(OpenPnpCaptureCameraDown, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:50 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 10:47:50 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_01, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:50 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 10:47:50 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_02, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:50 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 205.255000, 0.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:50 NullDriver DEBUG: moveTo(N1 N1, (13.249000, 205.255000, 0.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:53 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 205.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:53 NullDriver DEBUG: moveTo(N1 N1, (13.249000, 205.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:53 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo Mid Location
2018-10-06 10:47:53 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 225.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:53 NullDriver DEBUG: moveTo(N1 N1, (13.249000, 225.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:54 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo Mid Location 2
2018-10-06 10:47:54 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 225.255000, 0.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:54 NullDriver DEBUG: moveTo(N1 N1, (13.249000, 225.255000, 0.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:54 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo End Location
2018-10-06 10:47:54 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 205.255000, 0.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:54 NullDriver DEBUG: moveTo(N1 N1, (13.249000, 205.255000, 0.000000, 0.000000 mm), 0.86)
2018-10-06 10:47:55 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-06 10:47:55 NullDriver DEBUG: moveTo(N1 N1, (NaN, NaN, 42.000000, NaN mm), 0.86)
2018-10-06 10:47:55 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): Finished
2018-10-06 10:47:55 ReferencePnpJobProcessor DEBUG: Attempt Feed id RESC2012N-33R, name null, heightUnits Millimeters, height 0.555000, packageId (RESC2012N) from ReferenceTapeFeeder id FDR1538500304947 with N1 N1.
2018-10-06 10:47:55 ReferenceDragFeeder DEBUG: feed(N1 N1)
2018-10-06 10:47:55 ReferenceHead DEBUG: H1.moveToSafeZ(0.86)
2018-10-06 10:47:55 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-06 10:47:55 NullDriver DEBUG: moveTo(N1 N1, (NaN, NaN, 42.000000, NaN mm), 0.86)
2018-10-06 10:47:55 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.86)
2018-10-06 10:47:55 NullDriver DEBUG: moveTo(OpenPnpCaptureCameraDown, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:56 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 10:47:56 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_01, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:56 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 10:47:56 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_02, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:47:56 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveTo((14.638000, 212.117000, NaN, NaN mm), 0.86)
2018-10-06 10:47:56 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_01, (14.638000, 212.117000, NaN, NaN mm), 0.86)
2018-10-06 10:47:56 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.actuate(true)
2018-10-06 10:47:56 NullDriver DEBUG: actuate(ZZZ_ReferenceActuator_Dummy_01, true)
2018-10-06 10:47:57 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveTo((14.638000, 212.117000, 0.000000, 0.000000 mm), 0.86)
[... cut ...]
2018-10-06 10:48:12 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 10:48:12 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_01, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:48:12 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 10:48:12 NullDriver DEBUG: moveTo(ZZZ_ReferenceActuator_Dummy_02, (NaN, NaN, 0.000000, NaN mm), 0.86)
2018-10-06 10:48:12 ReferencePnpJobProcessor INFO: Job finished 1 parts in 24.6 sec. This is 146.2 pph
```

vonnieda, 2018-10-06:
@ldpgh If you still think this is happening, please post a log showing the problem.

ldpgh, 2018-10-06:
Here comes the log ... and it contains "load of NT1 to the nozzle", but my outer context is a little bit weird.
1. AFAI can tell just NT3 has an assignment to the devices to be placed. Installing NT1 is some kind surprising.
2. Except NT3 all other NTn have "Allow Incompatible Packages?" enabled. This allows to use NT1. Will try again with "Allow Incompatible Packages?" disabled for all Nozzle Tips.
3. The used dragfeeder is mounted at the xy-location of the NT1 tool holder. It's hard to see that machine tries to change the nozzle and not to pick the part.

```
2018-10-06 18:22:35 Main DEBUG: Bienvenue, Willkommen, Hello, Namaskar, Welkom to OpenPnP version 2018-09-16_14-34-42.f8c9f2d.
2018-10-06 18:22:35 Scripting TRACE: Scripting.on Startup
2018-10-06 18:22:54 ReferenceMachine DEBUG: setEnabled(true)
2018-10-06 18:22:54 GcodeDriver DEBUG: sendCommand(null, 250)...
2018-10-06 18:22:55 GcodeDriver DEBUG: sendCommand(serial://COM220 null, 250) => []
2018-10-06 18:22:55 GcodeDriver DEBUG: sendCommand(connect, 5000)...
2018-10-06 18:22:55 GcodeDriver TRACE: [serial://COM220] >> connect
2018-10-06 18:22:55 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:22:55 GcodeDriver DEBUG: sendCommand(serial://COM220 connect, 5000) => [ok]
2018-10-06 18:22:55 GcodeDriver DEBUG: sendCommand(enable, 5000)...
2018-10-06 18:22:55 GcodeDriver TRACE: [serial://COM220] >> enable
2018-10-06 18:22:59 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:22:59 GcodeDriver DEBUG: sendCommand(serial://COM220 enable, 5000) => [ok]
2018-10-06 18:22:59 ReferenceHead DEBUG: H1.home()
2018-10-06 18:22:59 GcodeDriver DEBUG: sendCommand(home, -1)...
2018-10-06 18:22:59 GcodeDriver TRACE: [serial://COM220] >> home
2018-10-06 18:23:00 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:23:00 GcodeDriver DEBUG: sendCommand(serial://COM220 home, -1) => [ok]
2018-10-06 18:23:02 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:23:03 ReferenceHead DEBUG: H1.home()
2018-10-06 18:23:03 GcodeDriver DEBUG: sendCommand(home, -1)...
2018-10-06 18:23:03 GcodeDriver TRACE: [serial://COM220] >> home
2018-10-06 18:23:05 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:23:05 GcodeDriver DEBUG: sendCommand(serial://COM220 home, -1) => [ok, ok]
2018-10-06 18:23:38 SystemLogger INFO: isAllPlaced false
2018-10-06 18:23:38 FiniteStateMachine TRACE: Initialize => Uninitialized -> PreFlight
2018-10-06 18:23:38 FiniteStateMachine TRACE: StartOrPause => Stopped -> Running
2018-10-06 18:23:38 ReferenceHead DEBUG: H1.moveToSafeZ(0.86)
2018-10-06 18:23:38 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-06 18:23:38 GcodeDriver DEBUG: sendCommand(moveto   Z42.0000  F860, 5000)...
2018-10-06 18:23:38 GcodeDriver TRACE: [serial://COM220] >> moveto   Z42.0000  F860
2018-10-06 18:23:39 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:23:39 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto   Z42.0000  F860, 5000) => [ok]
2018-10-06 18:23:39 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 18:23:39 Scripting TRACE: Scripting.on Job.Starting
2018-10-06 18:23:39 FiniteStateMachine TRACE: Next => PreFlight -> FiducialCheck
2018-10-06 18:23:39 FiniteStateMachine TRACE: Next => FiducialCheck -> Plan
2018-10-06 18:23:39 ReferencePnpJobProcessor DEBUG: Planned placements [N1 N1 -> R20]

2018-10-06 18:23:39 FiniteStateMachine TRACE: Next => Plan -> ChangeNozzleTip
2018-10-06 18:23:39 ReferencePnpJobProcessor DEBUG: Change NozzleTip on Nozzle N1 N1 from null to NT1 NT1
2018-10-06 18:23:39 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): Start
2018-10-06 18:23:39 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo Start Location
2018-10-06 18:23:39 ReferenceHead DEBUG: H1.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceCamera DEBUG: OpenPnpCaptureCameraDown.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 18:23:39 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 205.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 18:23:39 GcodeDriver DEBUG: sendCommand(moveto X43.9430 Y204.1470   F860, 5000)...
2018-10-06 18:23:39 GcodeDriver TRACE: [serial://COM220] >> moveto X43.9430 Y204.1470   F860
2018-10-06 18:23:42 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:23:42 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto X43.9430 Y204.1470   F860, 5000) => [ok]
2018-10-06 18:23:42 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 205.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 18:23:42 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo Mid Location
2018-10-06 18:23:42 ReferenceNozzle DEBUG: N1.moveTo((13.249000, 225.255000, 42.000000, 0.000000 mm), 0.86)
2018-10-06 18:23:42 GcodeDriver DEBUG: sendCommand(moveto  Y224.1470   F860, 5000)...
2018-10-06 18:23:42 GcodeDriver TRACE: [serial://COM220] >> moveto  Y224.1470   F860
2018-10-06 18:23:43 GcodeDriver TRACE: [serial://COM220] << ok
2018-10-06 18:23:43 GcodeDriver DEBUG: sendCommand(serial://COM220 moveto  Y224.1470   F860, 5000) => [ok]
2018-10-06 18:23:43 ReferenceNozzle DEBUG: N1.loadNozzleTip(NT1): moveTo Mid Location 2
[... cut ...]
2018-10-06 18:24:29 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_01.moveToSafeZ(0.86)
2018-10-06 18:24:29 ReferenceActuator DEBUG: ZZZ_ReferenceActuator_Dummy_02.moveToSafeZ(0.86)
2018-10-06 18:24:29 ReferencePnpJobProcessor INFO: Job finished 2 parts in 50,5 sec. This is 142,5 pph
2018-10-06 18:24:29 Scripting TRACE: Scripting.on Job.Finished
2018-10-06 18:24:29 FiniteStateMachine TRACE: Next => Cleanup -> Stopped
2018-10-06 18:24:29 FiniteStateMachine TRACE: Reset => Stopped -> Uninitialized
```

ldpgh, 2018-10-06:
With all Nozzle Tips property "Allow Incompatible Packages?" disabled the machine starts with load of NT3 as it is desired ... So  I close the issue.

closed 2018-10-06 by ldpgh