openpnp/openpnp #779: Update of placement status after "one step of the job" depends on user activity

ldpgh, 2018-10-05:
# Problem Report
## Describe the Issue
After a "one step of the job" the "Placed" status is not updated automaticaly.

## Steps to Reproduce
1. Setup a regular Job.
2. "Process one step of the job and the pause."
3. Check "Placed" status ... should be still unchecked
4. Click in the next line of the part list ... status will change to checked.

## Expected Result
After "Process one step of the job and the pause." the "Placed" status should automatically updated.

## Actual Result
After "Process one step of the job and the pause." the "Placed" status is not automatically updated. It seems there is a "redraw" of the GUI required to get "Placed" status checked.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
... no real data, because the internal state is OK. Just the visualization is pending for some kind of a trigger.

[image]

ldpgh, 2018-10-05:
Note: After "Start processing the job" the "Placed" status has been update as expected.

vonnieda, 2018-10-17:
This is a bug, and is emblematic of the problems with the entire status system. Many of the changes that were made after initial development did not use events or property change listeners, and instead just called refresh. Now we have this issue where everything needs to call refresh after every update, which is not sustainable. This code should be cleaned up and fixed so that it responds to property change events, and the refresh stuff removed.