openpnp/openpnp #781: Home button remains yellow though "Home after ENABLED?" is checked

ldpgh, 2018-10-06:
# Problem Report
## Describe the Issue
If "Home after ENABLED?" in MachineSetup/ReferenceMachine is enabled and "home" command is issued right after "enable".

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Check color of the home-button in MachineControls.

## Expected Result
home-button in MachineControls filled/black color
Machine is "homed" perfectly.

## Actual Result
home-button in MachineControls filled/yellow color
Maybe the machine has still some pending settings to be reset.
A 2nd home is always issued manually after "enable" to turn the button color into black.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
... no real idea what to add in this visual color choose topic

ldpgh, 2018-10-06:
[image]

ldpgh, 2018-10-06:
[image]

vonnieda, 2018-10-06:
Should be an easy fix. Thanks for reporting it!

vonnieda, 2018-10-06:
Fixed in https://github.com/openpnp/openpnp/commit/7de60a14686536cf9098d344f357a4f7044b4c2f

closed 2018-10-06 by vonnieda

ldpgh, 2018-10-13:
Develop binary (download 2018-10-13) works for me too.

commit 7de60a14 (2018-10-06): Fixes #781 by changing the home icon indicator after the auto homing.
commit 665252ab (2019-05-19): Merge branch 'develop'

* develop: (63 commits)
  Note about reverting OpenCV upgrade.
  Revert "Updates required to move to opencv-3.4.2"
  serial fix
  Set ScriptEngine's FILENAME key to improve error messages.
  Escape angle brackets in exception messages.
  ReferenceDriverCommunications now allows you to specify the line ending type. Options are LF (default), CR, CRLF. No UI yet, change in machine.xml.
  Fixes #781 by changing the home icon indicator after the auto homing.
  Added welcome in Spanish
  Move translations into resources, where they belong. This fixes #773.
  Fixing an issue w [... cut]