openpnp/openpnp #782: Use bottom vision to check nozzle-tip after loading for offset

ldpgh, 2018-10-07:
# Feature Request
## Describe the Feature
Due to runout and misaligned loading of the nozzle-tip there is a chance of an unobeyed deviation.
1. misaligned ... means nozzle-tip not completly moved into the nozzle tip holder (some kind of tilt)
2. nozzle-tip is rotated in the holder or was taking out of the holder.

## How Will It Look?
After nozzle change use Bottom-Vision to check the tip of the nozzle-tip for perfect aligment and issue a warning if the error exceeds a value. Error value should be given to the nozzle-tip as property.

## Why Do We Need It?
1.1. Hat a tilted nozzle-tip causing 0.4mm offset
1.2. After taking out the nozzle and putting back the orientation has been changed causing 0.2mm offset
2. AFAI can tell it is useful for everyone having not a 101% percent perfect working tool changer.

## Provide Examples
Not known ... but may exist.

ldpgh, 2018-10-09:
Companion [#783](https://github.com/openpnp/openpnp/issues/783)

vonnieda, 2018-10-10:
For this, and for #783, I think it would be better to finish the mostly already finished nozzle calibration feature in #235. What you've described here, and in #783, only works for rotation = 0 (as far as I can tell) but many feeders pick with a non-zero rotation.

ldpgh, 2018-10-10:
Agree to [#235](https://github.com/openpnp/openpnp/issues/235). Could not catch it while searching the issues.

What (eg. branch) needs to be used/turned on to compile for [#235](https://github.com/openpnp/openpnp/issues/235)?

vonnieda, 2019-02-14:
Fixed by #804.

closed 2019-02-14 by vonnieda