openpnp/openpnp #783: Add offset property to nozzle-tip

ldpgh, 2018-10-09:
# Feature Request
## Describe the Feature
Due to runout the nozzle-tip may have a deviation. To store the deviation a nozzle-top owned offset property is needed.

## How Will It Look?
Add offset property to nozzle-tip similar to the nozzle-camera offset.
The offset should be handled as offset to the nozzle with orientation 0 degree. (Rotational offset has to be correct by the bottom-vision or it is ignored if bottom vision is disabled.)
In general I think of xy-offset, but for the sake of completeness it is better to have a xyz-triple.

## Why Do We Need It?
1. Due to manufacturing each nozzle tip may have its own variation.
2. AFAI can tell it is useful for everyone having a tool changer with more or less inaccurate nozzle-tips
3. companion for [#782](https://github.com/openpnp/openpnp/issues/782)

## Provide Examples
... see postprocessed (manipulated) screenshot
[image]

vonnieda, 2018-10-10:
@ldpgh Do I understand correctly that this offset would just be a simple offset in X, Y, Z, that is only applied when rotation = 0? What happens when rotation != 0?

vonnieda, 2018-10-17:
Related to #235. I think this is redundant if that is working correctly.

ldpgh, 2019-01-17:
Addressed by PR #803 ... but in this PR the fields are set to ReadOnly.

For me ... remove the ReadOnly settings in the source code and re-issue the PR

markmaker, 2021-04-25:
I think this is fully covered by improved run-out compensation, i.e. #804 and #825.

closed 2021-04-25 by vonnieda