openpnp/openpnp #784: angle-increment@machine.xml & angleIncrement@ReferenceNozzleTip.java dependency is not accessible by the GUI

ldpgh, 2018-10-14:
# Feature Request
## Describe the Feature
While working on issue #235 the modification of angleIncrement@ReferenceNozzleTip.java did not succeed, because it was always overridden at startup by angle-increment@machine.xml.

## How Will It Look?
1. The variable becomes an entry-field in panelCalibration @ ReferenceNozzleTipConfigurationWizard.java.
2. The range should be added as well (0...360 degree) at the moment, because my machine has just +/-95degree.

## Why Do We Need It?
1. It makes the dependeny between source code and machine.xml accessible through the GUI. At the moment just edit of machine.xml covers this setting.
2. Covers machine settings with special treatment/reduced range of the nozzle rotation.

## Provide Examples
Not known to me.

ldpgh, 2019-01-17:
PR #803 adds the parameter.

vonnieda, 2019-02-14:
I think this is resolved by #803 and it's associated PRs. Please reopen if not.

closed 2019-02-14 by vonnieda