openpnp/openpnp #795: Manual for automatic nozzle offset wizzard

mgrl, 2018-12-21:
Just a question related to the nozzle offset helper. Great addition, it helps to get the math right and makes the routine very simple.

My question is to step 4:
I use flour to get a mark on the table. Since we have some runout on the nozzle tips, I turn the nozzle in the flour at least by 360° to get the real center of the runout tainted nozzle tip. 
Am I right in doing it that way and it would be an improvement to add that advice to the wizzard?

vonnieda, 2018-12-22:
Sounds like a good idea to me. Maybe you could make a short video or GIF showing it and we could link to that as well?

pfried, 2019-01-02:
We could integrate the 360 turn thing (maybe a 360degree spin button), sounds like a good idea to me. I did the wizard, so maybe I can do a tutorial at some point (could be some weeks)

mgrl, 2019-01-05:
okay, updated the wiki according to the new steps: https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration:-Nozzle-Setup#head-offsets

for me it's okay like it is. maybe I make a photo and add it to the wiki next days (flour is still on the pcb ;)

regards, michael

closed 2019-01-05 by mgrl

mgrl, 2019-01-05:
finally just for reference. I just found that the 360° degree advice makes sense to get the machine better calibrated for accuracy than without. Its in conjunction to the findings here:

- https://groups.google.com/d/msg/openpnp/16YV9sPXPFM/WIaTbpIkBgAJ
- https://groups.google.com/d/msg/openpnp/16YV9sPXPFM/8kiTyXxGBgAJ

vonnieda, 2019-01-05:
Nice work @mgrl!