openpnp/openpnp #801: AutoSave delay after each placement

mgrl, 2019-01-10:
# Problem Report
## Describe the Issue
Coming back to my machine I found that the new autosave-feature adds a delay of about 1s on my windows machine after each placement.
[image]

I know it was introduced for having reliable placement-data after a sudden shutdown, but 1s adds much time... I would propose to make it at least configurable in the GUI to disable that feature.
Alternative, can these IO-operations be made faster by threading them off and continue with the processing? 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Run a job
3. see the delay

This is on win 7, 64 bit. Didn't anyone else having that delay until today?

vonnieda, 2019-01-10:
WRT threading it out, it's not an easy change. See https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/SKcP9xNd2_E/0kLOpvMMBwAJ

mgrl, 2019-01-10:
okay, will do a patch in the near future to expose that setting

mgrl, 2019-01-11:
I accidentally found out, that I had my job stored on a network share. Storing it local brings the delay down to about 20ms:
[image]

anyway I made a change and send a pull request #802

closed 2019-01-11 by mgrl