openpnp/openpnp #807: CameraID get lost and Reassign it brings just B/W setting back (minor issue)

ldpgh, 2019-01-18:
# Problem Report of a Minor issue
Camera setting get lost due to not connected camera. Reassign the camera brings it back, but with B/W setting.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
0. The machine is using the same camera for bottom and top vision (Trust Webcam)
1. Remove DownLooking camera
2. Start OpenPnP
3. Connect the Camera 
4. ReAssign the camera to DownLooking Camera
5 . B/W picture, because Saturation is set to Minimum (aka 1)
6. Move saturation slider back to 65 ... Color is back

## Expected Result
Get a Color setting after re-assign of the camera.

## Actual Result
see. Problem Report above
The saturation setting is gone ... not clear why (and not debugged yet !!)

ldpgh, 2019-01-19:
The camera settings are applied immediately ... no APPLY required. So this handling seems to take a different route compared to the regular parameters

ldpgh, 2019-01-19:
Once it happens the reassignment brings in a new ID for the camera ...
- usage yesterday
2019-01-19 13:31:29.980 OpenPnpCaptureCamera WARNING: No camera found with ID Trust Webcam \\?\usb#vid_0c45&pid_6340&mi_00#8&7a4ac3&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global for camera OpenPnpCaptureCameraDown
- usage today
               <camera class="org.openpnp.machine.reference.camera.OpenPnpCaptureCamera" id="CAM1536394175426" name="OpenPnpCaptureCameraDown" looking="Down" settle-time-ms="250" rotation="89.3" flip-x="false" flip-y="false" offset-x="0" offset-y="0" crop-width="0" crop-height="0" scale-width="0" scale-height="0" deinterlace="false" unique-id="Trust Webcam \\?\usb#vid_0c45&amp;pid_6340&amp;mi_00#8&amp;2a5beefa&amp;0&amp;0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global" format-id="0" fps="30">

ldpgh, 2019-02-16:
Additional issue in this context ... "DownLooking" camera is opened as "UpLooking" camera if the down-looking camera is not connected. Keep in mind both cameras are identical products.
- Surprisingly after startup the "UpLooking" camera ( `..." looking="Down" ...` ) is opened as "UpLooking" camera.

- usage 2916-02-12 .... camera settings from `machine.xml`
   - `<camera class="org.openpnp.machine.reference.camera.OpenPnpCaptureCamera" id="CAM1536394175426" name="OpenPnpCaptureCameraDown" looking="Down" settle-time-ms="250" rotation="89.3" flip-x="false" flip-y="false" offset-x="0" offset-y="0" crop-width="0" crop-height="0" scale-width="0" scale-height="0" deinterlace="false" unique-id="Trust Webcam \\?\usb#vid_0c45&amp;pid_6340&amp;mi_00#8&amp;2a5beefa&amp;0&amp;0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global" format-id="0" fps="10.0">`
   - `<camera class="org.openpnp.machine.reference.camera.OpenPnpCaptureCamera" id="CAM1538814163725" name="OpenPnpCaptureCameraUp" looking="Up" settle-time-ms="250" rotation="0.0" flip-x="false" flip-y="true" offset-x="0" offset-y="0" crop-width="300" crop-height="300" scale-width="0" scale-height="0" deinterlace="false" unique-id="Trust Webcam \\?\usb#vid_0c45&amp;pid_6340&amp;mi_00#8&amp;295b300b&amp;0&amp;0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global" format-id="0" fps="10.0">`

crashmatt, 2020-07-10:
I get a similar problem using two of the same type of cameras but a different type.  
Using ubuntu 18.04. Looks like the problem reported above is on windows.

Both hue and gamma values are reset.

This happens every time I reboot the computer which is frustrating.
On reboot the final digit of the camera id may be reassigned depending on...?
It looks like the values are reset when the new camera is picked, not when it is lost.  I tracked this as a git repository change.

`<camera class="org.openpnp.machine.reference.camera.OpenPnpCaptureCamera" id="CAM1591877223153" name="Top" looking="Down" settle-method="FixedTime" settle-time-ms="250" settle-timeout-ms="500" settle-threshold="0.0" settle-debounce="0" settle-full-color="false" settle-gaussian-blur="0" settle-gradients="false" settle-mask-circle="0.0" settle-contrast-enhance="0.0" settle-diagnostics="false" rotation="0.0" flip-x="false" flip-y="false" offset-x="0" offset-y="0" crop-width="0" crop-height="0" scale-width="0" scale-height="0" deinterlace="false" unique-id="USB 2.0 Camera: HD USB Camera usb-0000:0a:00.3-1" format-id="8" fps="10.0">`

         <camera class="org.openpnp.machine.reference.camera.OpenPnpCaptureCamera" id="CAM1591877398158" name="BottomCam" looking="Up" settle-method="FixedTime" settle-time-ms="250" settle-timeout-ms="500" settle-threshold="0.0" settle-debounce="0" settle-full-color="false" settle-gaussian-blur="0" settle-gradients="false" settle-mask-circle="0.0" settle-contrast-enhance="0.0" settle-diagnostics="false" rotation="-1.417841254939061" flip-x="false" flip-y="true" offset-x="0" offset-y="0" crop-width="0" crop-height="0" scale-width="0" scale-height="0" deinterlace="false" unique-id="USB 2.0 Camera: HD USB Camera usb-0000:02:00.0-2" format-id="8" fps="10.0">

OpenPnpCaptureCamera.java line 148 is suspicious.  That sets a new value for saturation when a new camera is set.  It is not clear to me where this value comes from.

markmaker, 2020-07-11:
This...
https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/OpenPnpCaptureCamera.java#L148

...is just setting up listeners:
https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/OpenPnpCaptureCamera.java#L489-L497

whereas `setPropertiesStream(stream)` here...
https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/OpenPnpCaptureCamera.java#L331

... is setting the values.
https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/OpenPnpCaptureCamera.java#L498-L516

And  `setPropertiesStream(stream)` is called 

- when the form Apply is pressed
- when the camera is opened, which in turn is called whenever a CameraView is opened or when a capture takes place (from camera stettling or a pipeline). 

\\?\usb#vid_0c45&amp;pid_6340&amp;mi_00#8&amp;**2a5beefa**&amp;0&amp;0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global

is not identical to

\\?\usb#vid_0c45&amp;pid_6340&amp;mi_00#8&amp;**295b300b**&amp;0&amp;0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global

So this should not happen, should it? Maybe that's a Linux bug?

_Mark

crashmatt, 2020-07-12:
Mark, thanks for your direction.  This is a whole lot more complex than I presumed.
The root cause for windows and linux may be different but the outcome seems a little similar.

If I have understood you right.  This line in the wizard binds to that property:

https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/wizards/OpenPnpCaptureCameraConfigurationWizard.java#L649

This appears to bind the device change to the camera

https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/wizards/OpenPnpCaptureCameraConfigurationWizard.java#L629

The only sign I can find of setting a value is:
https://github.com/openpnp/openpnp/blob/c6c100e175a71460ebfe969a7a880dd46f4e8c61/src/main/java/org/openpnp/machine/reference/camera/wizards/OpenPnpCaptureCameraConfigurationWizard.java#L490

I don't understand where the default values come from.   Maybe this is a linux system thing.

found a workaround by editing machine.xml to fix the camera port.  That way the settings don't get overwritten by selecting a different device.

/Matt

crashmatt, 2020-07-13:
Just tried my desktop machine.xml on my laptop using built in camera.  All camera settings get default to zero or whatever setting is closest to it.
This points again to there being something wrong with settings defaults when changing the camera device but not when adding a new camera.

crashmatt, 2020-07-13:
A theory.
1. Machine.xml is read at startup.
2. There is no camera device available to assign the settings to.  These settings are forgotten.
3. The gui sets zero settings
3. The device is modified.  The modified device does not use defaults since it is modified rather than new.  It uses the gui values.

markmaker, 2020-07-16:
The GUI code should never even be applied unless you explicitly go to the camera Wizard. And this works for many users, so there must be something else happen.

Do you have duplicated of your cameras defined? A third camera perhaps on the same USB device? Check both the machine and the head cameras and delete any surplus camera.

_Mark

markmaker, 2021-04-25:
@crashmatt is this still an issue?

crashmatt, 2021-04-28:
I have not unplugged the machine for a while.  Have not got much time to
check at the moment.  Close the issue for now and I will open a new one if
it re-appears.

vonnieda, 2021-04-28:
I think it's quite likely there is a bug here. I didn't add anything specifically for handling the case when a camera existed previously and then we load OpenPnP without the camera existing, and then it exists with the same ID in the future. At the least we could check on that behavior.

In addition, there is the known issue of cameras not having fixed IDs because so few have a real serial number. It's been suggested in the past to do some kind of heuristic to determine which camera is which based on the images coming from it and sort of "lazy load" the settings. I think it would be worth taking a stab at that.

So I think let's leave this one open for now.