openpnp/openpnp #810: Unit setup problem

hruijin, 2019-02-07:
When I use smoothie controller, I should choose the unit of mm, but the running distance of the stepper motor is not correct! Select inch unit, step motor running the correct distance! What is the reason for this?

vonnieda, 2019-02-07:
Hi @hruijin, can you please provide some more detail? 

1. Where are you selecting the unit? In OpenPnP's System Units, or in GcodeDriver settings?
2. Have you set and checked your steps per mm settings in Smoothie?
3. Are you initializing Smoothie in mm mode? (send G21 during connect)

Thanks,
Jason

hruijin, 2019-02-07:
Hello, thank you for your reply! In the Gcode Driver, smoothie has set the correct number of mm steps and executed G21!

vonnieda, 2019-02-07:
Hi @hruijin,

Please post your question to http://groups.google.com/group/openpnp. This way we can get more eyes on the problem. It will help, also, if you post your Smoothie config.txt and OpenPnP machine.xml there. 

Thanks,
Jason

closed 2019-02-08 by hruijin