openpnp/openpnp #813: Follow Up #648

vonnieda, 2019-02-18:
These are a couple minor issues that need to be followed up on after the commit of #648.

* [ ] Implement transform for the two point board locator.
* [ ] Add an indication to the user when a board has been located.
* [ ] Verify compatibility with panels.
* [ ] Implement inverted transform in calculateBoardPlacementLocationInverse so that manual training
  benefits from the transform.

vonnieda, 2020-01-13:
Some additional discussion in https://groups.google.com/d/msgid/openpnp/7a301f2e-7f61-4173-a841-8abe9d898b5d%40googlegroups.com?utm_medium=email&utm_source=footer

vonnieda, 2020-01-13:
Note regarding "Fix calculateBoardPlacementLocationInverse for board bottoms. Currently only works for top.", I think the solution is just to apply the steps in reverse.

markmaker, 2022-02-03:
@tonyluken, 

could you please look at this and tell me what you already implemented? So we can possibly close it?

Thanks!

_Mark

tonyluken, 2022-02-03:
>Implement transform for the two point board locator.

I believe when I added support for more than three fiducials I also covered the two fiducial case as well so I think this is complete.

>Fix calculateBoardPlacementLocationInverse for board bottoms. Currently only works for top.
I think the solution is just to apply the steps in reverse.

I believe @vonnieda fixed that here [5efa1b0](https://github.com/openpnp/openpnp/commit/5efa1b0c16f6931b43d457aa6004565e9446886e)

>Add an indication to the user when a board has been located.

I think that may still be open.

tonyluken, 2023-08-02:
@vonnieda,

Can we close this issue as I believe it was address by [PR1507](https://github.com/openpnp/openpnp/pull/1578)?

vonnieda, 2023-08-03:
Yep, thanks @tonyluken!

closed 2023-08-03 by vonnieda