openpnp/openpnp #816: Visual Homing returns true when no fiducial is found

cncmachineguy, 2019-03-15:
# Problem Report
## Describe the Issue
If no Home_fiducial is found, visualHoming returns true and the machine thinks it is homed.

## Steps to Reproduce
If you use visual homing already, just cover the fid so it can not be seen.
If you do not use visual homing, simply enable it

## Expected Result
I expect when the visual homing is done, if no fiducial is found an error might be thrown, or at the least do not set the machine as homed.

## Actual Result
Machine is set as homed and acts ready to go as if it is homed.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
2019-03-03 23:35:08.977 ReferenceFiducialLocator DEBUG: **_No matches found!_**
2019-03-03 23:35:08.981 GcodeDriver DEBUG: sendCommand(G92X0Y0Z0C0, -1)...
2019-03-03 23:35:08.982 GcodeDriver TRACE: [serial://COM7] >> G92X0Y0Z0C0
2019-03-03 23:35:08.986 GcodeDriver TRACE: [serial://COM7] << ok
2019-03-03 23:35:08.986 GcodeDriver DEBUG: sendCommand(serial://COM7 G92X0Y0Z0C0, -1) => [ok]
2019-03-03 23:35:08.986 GcodeDriver DEBUG: sendCommand(G4P0, -1)...
2019-03-03 23:35:08.986 GcodeDriver TRACE: [serial://COM7] >> G4P0
2019-03-03 23:35:08.990 GcodeDriver TRACE: [serial://COM7] << ok
2019-03-03 23:35:08.990 GcodeDriver DEBUG: sendCommand(serial://COM7 G4P0, -1) => [ok]
2019-03-03 23:35:08.992 ReferenceMachine DEBUG: **_setHomed(true)_**
-----------------------------------------------------------------------

doppelgrau, 2021-04-18:
I think this can be closed, reference head throws an exception if no location is found, reference machine does not catch it (only exceptions in scripting).
Guess had been solved without closing the bug?! (commit 3e64b5751e02d10e02c387b71b2e9b52f5febeab)

markmaker, 2021-04-25:
> Guess had been solved without closing the bug?! (commit 3e64b57)

Yes this was solved, but in 2fa440ca97dc8a7b6f4ea9e6f4a269e954279ef0, i.e. #1035.

closed 2021-04-25 by vonnieda

cncmachineguy, 2021-04-26:
Thanks All :)