openpnp/openpnp #819: Bottom Side Rotation Error w/ 3 Point Fid Check

vonnieda, 2019-03-30:
The new fiducial check code seems to be calculating a bad angle for bottom side boards.

A fix has been posted by @aneox in
https://github.com/openpnp/openpnp/commit/77f539aaf194d743be2a1a34f58c2dd57b6a7978#commitcomment-32973251 but I think this just hides the problem. 

It appears the fiducial check is adding 90 degrees to the result for some reason. @aneox's fix removes the 90 degrees but it would be better to find out where the 90 degrees is coming from in the first place.

This bug can be reproduced easily by loading the pnp-test job and running a fiducial check on the bottom board with angle 36.609. After the check completes the crosshair shows the correct XY but the rotation is +90.

vonnieda, 2019-03-30:
Note to self: I have a private ZIP from @aneox with more detailed information if needed. Email search term "tested with top side".

vonnieda, 2019-07-25:
Additional possible related example in https://groups.google.com/d/msgid/openpnp/bd72611a-5aa0-4a32-bf9b-21db7129bc4d%40googlegroups.com?utm_medium=email&utm_source=footer

closed 2019-10-19 by vonnieda

aneox, 2020-04-02:
recently I have updated to the last commit and would like to report, my [bug](https://github.com/openpnp/openpnp/commit/77f539aaf194d743be2a1a34f58c2dd57b6a7978#r32973251) is fixed, thx

PR #905 (2019-10-19): Bugfix/bottom fiducial
# Description
Fixes #819.

# Justification
Needs fixing
# Instructions for Use
None required
# Implementation Details
Invert X for bottom side fiducials and placements.

1. How did you test the change? 
Tested by doing 2 point and 3 point fiducial checks on all of the boards in the sample job.  Verified that the camera moved to the board origin and is in the correct orientation.  Moved to each component and verified that the camera is centered over the component and is in the correct orientation.  Did the same testing on actual machine using 6 board panel.
2. Did you fol [... cut]