openpnp/openpnp #823: Panel Board Placements Status Incorrect

vonnieda, 2019-04-21:
User reported that placement status is being shared across all boards in a panel, rather than being separate. 

> Hi guys
> I've noticed weird panelization bug today.
> I've panelized a board in two rows by 3 boards all together 6 boards. After starting a job openpnp marks all parts already placed on one board as placed on all boards. It actually places them on all boards but marks them as placed on all boards as soon as the part is placed on one. That's not a problem by itself but when the machine stops for error or you stop a job for whatever reason there is no way you can tell which parts are placed on all boards and which are not. Removing placed mark on one board immediately removes that mark on all boards.
> 
> Is this normal behavior or is this a bug? This way panelization does not make sense.
> I've even tried to input 6 separate boards from 6 separate boards files but this behavior ocuured again and marked all Ma parts placed on all boards when in fact they were placed only on one.
> 
> Any ideas.
> 
> TDarlic

See https://groups.google.com/d/msgid/openpnp/7ee97feb-c477-4995-b2da-d0d957515717%40googlegroups.com

CircuitSetup, 2019-08-15:
I noticed this as well

damancusonqn, 2019-09-27:
Same here, running in Ubuntu, my panel is 7x5 boards. It happens after a "No component detected after a pick" error or when the ReferenceStripFeeder reaches the end of the strip. Most often all components are wrongly marked as placed (in the actual board and the rest of the panel).

MatSpy, 2021-02-28:
Hello. 

Is there any progress in this topic?
Penalization is very useful thing. And with this bug it is way harder to use. 

Regards,
Mateusz

cyberwaelder1, 2021-06-01:
hi, I have the same problem. When one part is not picked, I have to remove all from the other boards to continue the step again.

tonyluken, 2023-08-02:
@vonnieda,

Can we close this issue as I believe it was address by [PR1507](https://github.com/openpnp/openpnp/pull/1578)?

vonnieda, 2023-08-03:
Wonderful, thank you @tonyluken!

closed 2023-08-03 by vonnieda