openpnp/openpnp #826: Exception in nozzle calibration

JulienDevillers, 2019-04-26:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
I try to calibrate a nozzle. After 2 rotations, one in each direction, the functions fails with a dialog box "java.lang.exception cannot be cast to java.util.List"

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. A machine with a nozzle and a nozzle tip.
2. navigate to nozzle calbration panel
3. move the nozzle above the camera
4. Set the pipeline :
<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="9" enabled="true" settle-first="true"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="1" enabled="true" conversion="Bgr2Gray"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.BlurMedian" name="4" enabled="true" kernel-size="3"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="3" enabled="true" diameter="200"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="2" enabled="true" threshold="150" auto="false" invert="false"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="5" enabled="false" diameter="100"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DetectCirclesHough" name="6" enabled="true" min-distance="50" min-diameter="40" max-diameter="60" dp="1.0" param-1="80.0" param-2="10.0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="results" enabled="false" threshold-step="10.0" threshold-min="50.0" threshold-max="220.0" repeatability="2" dist-between-blobs="10.0" color="true" color-value="0.0" area="true" area-min="25.0" area-max="5000.0" circularity="false" circularity-min="0.800000011920929" circularity-max="-1.0" inertia="true" inertia-ratio-min="0.10000000149011612" inertia-ratio-max="-1.0" convexity="true" convexity-min="0.949999988079071" convexity-max="-1.0"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="0" enabled="true" image-stage-name="9"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="7" enabled="false" key-points-stage-name="results">
         <color r="255" g="255" b="0" a="255"/>
      </cv-stage>
      <cv-stage class="org.openpnp.vision.pipeline.stages.DrawCircles" name="8" enabled="true" circles-stage-name="6" thickness="3"/>
   </stages>
</cv-pipeline>

4. click "calibrate" button

## Expected Result
No failure.

## Actual Result
Error message.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue

markmaker, 2019-04-26:
There is not enough information. Please add the log. 

What version of OpenPNP do you use?

Have you ever tried to reset the Pipeline? The default pipeline was dramatically changed in recent `develop` branch versions. If you have recently updated the version, maybe the pipeline stored on your nozzle tip is no longer compatible with the code. 

_Mark

JulienDevillers, 2019-05-08:
Ok, I added version number.
Sorry, I wrote the pipeline in the issue but as it is xml, it did not appear in the issue. I set it as code and now you can see it.

The default pipeline does not recognize the nozzle. That's why I'm using a custom one.

JulienDevillers, 2019-05-10:
I did some debug.
In ReferenceNozzleTip.java

In my pipeline, the stage named "result" is disabled. This causes
` Object results = pipeline.getResult(VisionUtils.PIPELINE_RESULTS_NAME).model;
`
to return " java.lang.exception : Stage not enabled." which can't be casted to List :
 `              if (0==((List) results).size()) {
 `
Fix by #833

JulienDevillers, 2019-05-12:
Fixed by #833

closed 2019-05-12 by JulienDevillers