openpnp/openpnp #831: Things I Learned on a Production SMT Line

vonnieda, 2019-05-10:
This is sort of a tongue in cheek but sort of serious list of things I've learned in the past several months working on a production SMT line. 

As background: When I designed OpenPnP I had never even seen a pick and place machine in person and I certainly had never run one. 

Now that I have more experience there are a lot of things I would change, so this is half wishlist, half mea culpa, and half plans for the future. Other folks should feel free to comment on things they think are problematic with core OpenPnP concepts.

This may someday become a feature list for OpenPnP 2.0.

I'll start:

* Touchscreen interface, if done well, is great. It's hard to do well.
* Probably very little value in showing cameras in color.
* Camera does not need to be shown all the time - just when targeting something.
* Graphical previews are gold. Parts, packages, boards, jobs, feeder locations, etc. A picture really is worth a thousand table rows.
* The placement -> part link should be soft, not hard. Store the text, not the ID. If the part doesn't exist in the database, warn the user but leave the text.
* Get rid of all the crazy saving of data constantly, saving config after every change, etc. If users need that level of safety, buy a [cheap UPS](https://www.amazon.com/AmazonBasics-Standby-UPS-400VA-Outlets/dp/B073Q48Z95/ref=sr_1_5?keywords=ups&qid=1557457854&s=gateway&sr=8-5)
* Hide complexity. There are way too many ways to screw up a position, or config. Capturing coordinates is way too easy. Make it more of a specific action than easy to do accidentally.
* Conveyors and SMEMA are trivial. Add native support.
* Dropping support for different boards in the same job would simplify things and probably affect no one. I don't think anyone ever uses this feature.
* Nozzle Tips should be global and assigned to packages, and overridable by parts: #183 
* Formalize the error system and bubble it up to alerts. It should be possible to ignore most errors until the end of a job.

bergyla, 2019-05-13:
I totally agree to the following point:
- Hide complexity. There are way too many ways to screw up a position, or config. Capturing coordinates is way too easy. Make it more of a specific action than easy to do accidentally.
I often touch the wrong icon and skrew up the part position for example when I simply like to set the camera to that position.. There should be a way of write protecting capturing coordinates and Part-entry itself (so also the selector should not be longer visible since this will every time flip out whenever I will try to simply select a part at the machine to travel the camera to the position...)
Since a part could be either Top or Bottom side. I separated this as two different boards (since I can´t) switch them without stopping the run at all - Almost all PCB Cad-Systems could either select exporting either Top or Down located parts..

markmaker, 2019-05-14:
> Hide complexity. There are way too many ways to screw up a position, or config. 

I think those two should be distinguished.

1. _"Hide complexity"_
2. _"There are way too many ways to screw up a position, or config."_

I'm reluctant to agree with the first one, I'm suffering enough as it is from dumbed-down products. As long as we still need to wire machines, tweak our `config.txt` and `machine.xml` etc. by hand, the GUI is fine in comparison, I think :-)

But for the second one there is potential. 

I think one thing that is cumbersome with OpenPNP in general, is the handling of forms/views. The user assumes that entered data will not be active before "Accept" is pressed. Including when you press the capture buttons. So when you accidetally press it, you think you can revert it using "Reset". But we all know that ist not the case. The capture buttons write directly to the model and the view is only refreshed with the changes. Other buttons inside the form (e.g. "Test Align" in Bottom Vision or "Calibrate" in Runout Compensation) work with the old model data, not the newly entered one in the form that is not yet accepted. A third variant is the "Reset" for the feed count in the feeder. It directly accepts the form with all changes without asking. A fourth variant is the camera settings (sliders) that change the model directly while other data is still not accepted. Furthermore a form is also not accpeted (or the user prompted to) when you switch to another wizard/form/view. So changes are not written back and may even be lost when you exit OpenPNP (I think). All this is unexpected and confusing. 

A proposal could be:

1. Snapshot the old model data when a Wizard is started, Because OpenPNP already has XML-Persistence this should be relatively easy to accomplish with an XML-Fragment (I think). This shuld be done each time the Wizard gets the focus i.e. also after returning from another Wizard. The snapshots should be saved in a reasonably size-limited stack (only push if the least shapshot has changed).  
2. Always directly accept data when it is entered i.e. when a field/control changes or when a location is captured or when a function button manipulates data (in many cases it already works that way). 
3. Change the "Reset" button programming to restore the old model state from the last snapshot i.e. the XML fragment saved in step 1. Pop the stack. Perhaps rename the "Reset" into "Undo". 
4. Remove the "Accept" button.

This way an accidentally captured coordinate could be reverted. The same with accidently changinge the camera sliders. Accidently pressing "Reset" on the feeder feed count, etc. pp.

Changing and reverting data should be worry-free. However sending the nozzle tip to a location may actually be dangerous, if the user confused the buttons. Yes, I was smacked by my machine once :) It might be a good idea to provide an optional confirmation message box there, or perhaps sound a warning beep and wait a sec before the motion. 

_Mark

cncmachineguy, 2019-05-14:
"Formalize the error system and bubble it up to alerts. It should be possible to ignore most errors until the end of a job."

To some degree this is what Marek and I have been working on. Instead of stopping the run for a failed vision or pick, discard and try again. After a user assignable number of tries, mark the feeder as empty and continue building.

JulienDevillers, 2019-05-21:
Hi

> This is sort of a tongue in cheek 

I needed google translate to understand :D

> * Touchscreen interface, if done well, is great. It's hard to do well.

Yes, hard to do well and will require a minimum (who said fixed? ) screen definition.

> * Probably very little value in showing cameras in color.

Is there an "unvalue" to show in color? Color may help the user to discern things on the board that luminance may not.

> * Camera does not need to be shown all the time - just when targeting something.

Agree if you have a reliable machine. But when setting up or learning, it's good to see them. Or maybe the relevant one : top for picking, bottom for ... bottom...

> * Graphical previews are gold. Parts, packages, boards, jobs, feeder locations, etc. A picture really is worth a thousand table rows.

Of course.

> * The placement -> part link should be soft, not hard. Store the text, not the ID. If the part doesn't exist in the database, warn the user but leave the text.

Not very clear to me what you mean.

> * Get rid of all the crazy saving of data constantly, saving config after every change, etc. If users need that level of safety, buy a [cheap UPS](https://www.amazon.com/AmazonBasics-Standby-UPS-400VA-Outlets/dp/B073Q48Z95/ref=sr_1_5?keywords=ups&qid=1557457854&s=gateway&sr=8-5)

Yes

> * Hide complexity. There are way too many ways to screw up a position, or config. Capturing coordinates is way too easy. Make it more of a specific action than easy to do accidentally.

Agree with Mark.I think there should be setup stuffs somewhere and production stuffs elsewhere.
I don't understand what the "apply" button does so I click it all the time to be sure that what I did is taken into account. It's not "standard UI".

> * Conveyors and SMEMA are trivial. Add native support.

Why not? No usage for me but maybe for others

> * Dropping support for different boards in the same job would simplify things and probably affect no one. I don't think anyone ever uses this feature.

Yes

> * Nozzle Tips should be global and assigned to packages, and overridable by parts: #183

Yes

> * Formalize the error system and bubble it up to alerts. It should be possible to ignore most errors until the end of a job.

Yes

vonnieda, 2019-05-21:
> > * Probably very little value in showing cameras in color.
> 
> Is there an "unvalue" to show in color? Color may help the user to discern things on the board that luminance may not.

My thinking here is that all, or nearly all, of the vision algorithms operate on grayscale images so showing color may lead the user to see something that the computer doesn't. Ultimately, what matters is that the computer can do what it needs to.

I'm also basing this on the fact that every commercial machine I see shows the camera in gray :)

> 
> > * Camera does not need to be shown all the time - just when targeting something.
> 
> Agree if you have a reliable machine. But when setting up or learning, it's good to see them. Or maybe the relevant one : top for picking, bottom for ... bottom...

Yea, there would always be a way to see it, but I don't think it deserves the permanent screen real estate it gets now.

> > * The placement -> part link should be soft, not hard. Store the text, not the ID. If the part doesn't exist in the database, warn the user but leave the text.
> 
> Not very clear to me what you mean.
>

This is really the biggest one, and the reason I wrote this post. Currently, in the board.xml there is a part-id field. This is the ID of the part in the Parts tab. OpenPnP looks up each part as the job is loaded and links it internally to the part object. If no part is found it effectively nulls out the part-id, and you lose data.

In addition, when you import a board from CAD data, since each placement must reference a part, if the part doesn't exist the importer either creates it or sets it to null. The value from the file is lost. 

For example, say your parts tab has one part: "0805-10k-1%". Now you load a CSV like this:

Ref,Part,X,Y,Rotation
R1,0603-10k-10%,10,10,0

OpenPnP will try to look up 0603-10k-10% in the parts database, will not find it, and so will set the part field to null. Now when you look at that board you can't tell what the part was originally supposed to be.

So, the change I want to make is that instead of the placement part field being a part object, it's just a string. We would store the string "0603-10k-10%" in the board file and we would only try to find the part from the library when it's needed. This would allow you to import a board that you don't have part definitions for yet, and then add the part definitions separately. The field would be shown in red if no part is defined, so you can easily browse through the placements and fix any that are missing parts.

In other words, we decouple the placement.part field from the parts database and "lazy load" it when needed.

pnpff, 2019-05-26:
@openpnp 2.0:
As we should have a component database with all necessary parameters we also would need a pool of pipelines. Nearly all passives like resistors or caps with same package could use same pipeline selectable from a pipeline-database. It would be much easier to maintain only a few pipelines instead of hundreds when a small change is made to bottom camera setup. 
The GUI needs a lot of improvements regarding easier handling. Userdefable Buttons like Discard (only a single Click) should allow access to scripting etc. and should be allowed to be positioned on main page for fast interaction. Camera switching could be done with single clicks/touches. All sensible decisions like deleting a nozzle tip with risk of loosing a lot of setup work should be double checked. Other changes like on feeder setup should not require a confirmation. 
Java seems to be a beautiful language for developing multiplatform apps but it is far from perfect compared to other solutions.

gxurma, 2019-05-28:
Some suggestions, if there is room / time for it:
- easier preview of parts and packages . Currently I have to switch to packages and click on one of them to show them. but who wants to figure out always which package a part has?
- Somehow sequencable / programmable nozzletip changer. The currently implemented 4 positions are OK, and mainly useful, but I already have troubles and would need at least 5 positions and intermediate actions/events too.
- somehow being able to set / measure nozzletip heights, so tips with different lengths can be calibrated.
- Calibrate table and board heights, maybe with a touch sensor, so we need a touch input.

I am thinking of an adaptive (deep-)learning for the loose part feeders. Maybe we need to add at least the possibility to start this.

vonnieda, 2019-05-28:
Hi Martin,

I agree with all your points. I'll add that regarding "adaptive
(deep-)learning for the loose part feeders" you could do this by
implementing a new feeder class. OpenPnP is designed to make adding new
feeders easy. Take a look at the current LoosePartFeeder or
AdvancedLoosePartFeeder, make a copy of it and change it to your liking.

Jason

gxurma, 2019-05-28:
Another important thing to have:
- calibration of squareness, or measurement and adjustment to the shear of the x-y table. (i know, the high accuracy machines also have this, but how they really do this is hard to figure out without a glass table measurement part)

vonnieda, 2019-05-28:
We have squareness compensation now. See
https://github.com/openpnp/openpnp/wiki/GcodeDriver#non-squareness-compensation

Jason

wiegleyj, 2019-05-31:
> Hide complexity. There are way too many ways to screw up a position, or config. Capturing coordinates is way too easy. Make it more of a specific action than easy to do accidentally.

The reason this is happening is because you also don't work on CNC equipment or deal with robotic kinematics and are inconsistently utilizing the concepts of coordinate frames and their transformations. The position and calculation of everything should be defined according to orthogonal coordinate transforms. See a book titled "Robotic Kinematics" by John Craig.

vonnieda, 2019-05-31:
@wiegleyj Can you expand on that a bit? Perhaps provide an example of how we're doing it wrong, and how we could do it right?

The problem I am describing is a UI problem, not a kinematics problem, but perhaps I don't understand what you are referring to.

pfried, 2020-01-19:
Maybe a more workflow oriented design which could enable different workflows. Right now the whole GUI is basically around operating the machine manually. If you want to use the software in production you want to follow the steps along:

* Machine setup hidden for most users (having other persons doing placing usually does not require them to be able to change machine configuration)
* Setting up feeders and the PCB for a certain job
* Running a job multiple times

Maybe also other things like Inspection or Dispensing

Having specific screens for some steps enables better visibility of the important features for that step (e.g. during production errors or warnings, time passed, parts running low, etc.)

Touch interfaces get quite common, but usually vendors will only display a specific small portion of the software because you cannot display as much.