openpnp/openpnp #834: Job will not Run

nfsaisi, 2019-05-10:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Please describe the issue in as much detail as possible. Without a good description it's impossible for developers to understand the problem.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Jog the machine 10mm to the right.
4. The machine fails to jog and shows an error.

## Expected Result
What did you expect to happen? Be descriptive.

## Actual Result
What actually happened? Be descriptive.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

nfsaisi, 2019-05-10:
Hi, my PNP Hardware still in construction , i figured today i give the software a try , installed the windows ver. and followed  the quick start with the example test boards , i'm have an issue makinig the job run , it will give a bunch of java error , i will attach a screenshot ,
Secondly , the driver tab for machine is (NULL) will not let me choose a driver as per Wiki instructed to do, please help to get me started
I appreciate all the work  you guys did , it's a dream coming true to own an SMD PNP Machine.
Regards
Naji
[image]

vonnieda, 2019-05-11:
Hi @nfsaisi, what version of Windows are you running?

nfsaisi, 2019-05-11:
Hi, Win 7 64bit

vonnieda, 2019-05-12:
@nfsaisi There some missing libraries in Windows 7 that are causing OpenPnP to not work correctly. Please see https://github.com/openpnp/openpnp/wiki/FAQ#openpnp-is-not-working-in-windows-7 for more information.

closed 2019-05-12 by vonnieda

nfsaisi, 2019-05-12:
@vonnieda  call me old school , i really hate graphical interface windows , its a real disappointment not to be able to use Win7, i downloaded the missing library and installed it , much better difference , no its only showing  "Job Error "  when i press the job processor play button , the parts status now Green , i attach
[image]
[image]

 the log and the error, you think there is a work around it  or i'm wasting my time ?

thanks

vonnieda, 2019-05-12:
Copy the pnp-test files out of the installation directory and to somewhere
in your home directory, like desktop. The problem is that the files are
read only because they are in program files.

I’ll add this to the FAQ too - it has been coming up a lot.

Jason

nfsaisi, 2019-05-12:
Thank you very much , it's working now 
 one last thing,  there is NO camera crosshair showing ,  i'm missing something or there is a command to activate it.

vonnieda, 2019-05-12:
Right click on the camera view and look at the options under Reticle. There
are a variety of reticles you can show. Also make sure you have set your
camera units per pixel:
https://github.com/openpnp/openpnp/wiki/Setup-and-Calibration%3A-General-Camera-Setup#set-units-per-pixel

Jason

nfsaisi, 2019-05-13:
Thanks for everything , i do have 2 more issues right now i don't think they are covered in the manual, if i need to open a new post please let me know
1st  :  i did choose a new job to import my own board into the job but still showing old test board and old parts etc.   is that normal till the new board actually loads up ?

2nd :  i'm trying to import a Gerber files to the new board but its not working , it's weird because the import worked before when i had the issue with Win 7 missing libraries now after the PNP is working the import is not .  by the way the files are 274xx from Protel 99SE , i know the PNP doesn't support the Protel .pcb file , it would be much easier.
if there is a conversion program you can recommend  , i'm all ears 

Thank you very much

vonnieda, 2019-05-13:
Please post these questions to the discussion group at
http://groups.google.com/group/openpnp. In general, that is the best place
to post questions as lots of people will be able to help. If a question
turns into a "this looks like a bug" then filing an issue is appropriate.

Thanks,
Jason