openpnp/openpnp #839: 0402 part placement with referencestrip Feeder with Openplacer - optimizing feeder pick

bergyla, 2019-05-14:
I started with a 0402 project on a Openplacer hardware.
In general I´m pretty impressed to get some positive results out of it.

But as I see, the referencestrip pick process should be optimized a bit:
Z-movement should slow down some millimeters before touching the strip.
Slow down for the lasting length (avoiding that parts will jump out...)
get really sure that movement stopped before open solonoid
wait some time until vacuum is safely holding the part before lifting.

Could you help me where to start with codereading to jumpstart with codeediting...?

cncmachineguy, 2019-05-15:
Hi bergyla, You can do most of this in the pick command assuming you are using GCODE driver. I am not sure exactly how, but you can define the speed for Z, and add some G4Pxxx for delays. So your command might look like :
G1ZxxF(whatever speed you want here)
G4P(how long you want to wait before turning on Vac)
Mxxx(Assuming you can turn on vac with M-Code
G4P(how long you want to wait after turning on Vac before lifting)
G0z0 (assuming Z0 is your top)

vonnieda, 2019-05-15:
@bergyla As @cncmachineguy, some of this you can do in Gcode. Other parts I think are already there: You can set pick dwell time and place dwell time on the Nozzle.

For other things, you will want to look in ReferenceStripFeeder.java, and in ReferencePnpJobProcessor.java. ReferencePnpJobProcessor.java is where the movement in Z happens.

closed 2019-05-25 by vonnieda