openpnp/openpnp #843: FR : manage a list of pipelines definition at a system level and affect them where needed.

JulienDevillers, 2019-05-16:
# Feature Request
## Describe the Feature
Instead of having a default pipeline and a specific pipeline for each part.
We could have a system that allows has many pipeline settings as the user needs.
Then everywhere where a pipeline is needed (feeder, camera runtout calibration...)
the user can assign one of the predefined pipelines or a specific one.

## How Will It Look?
A new tab in the gui with an incrementable list of named pipelines and the ability to modify any of them.
As I suspect that there won't be a lot of pipelines, maybe the pipeline setup window might be integrated to this tab (using the current camera view(s) would be a nice integration).

For each pipeline, we will be able to see the list of items to which the pipeline is assigned to.
In feeders, it will replace the tape type.
The user will still be able to modify the pipeline in the feeders but this will affect all feeders using this pipeline.

Adding a checkbox to say "this particular feeder is using a specific pipeline or one of the generic pipelines defined in the pipelines tab.

## Why Do We Need It?
1. copy / paste pipelines beetween feeders in not very fun to do and error prone.
3. You never know if 2 feeders are sharing the same pipeline settings.
2. Easier to add new feeder, if an already present feeder uses the same packaging, just need to 
select it in the list. (Generally, we will use (for strip), white, black and clear).

vonnieda, 2019-05-17:
Agreed - I think this is a good improvement. This concept has been discussed a bit here: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/JyZvwIf9A0U/C5lq4VtTAgAJ

One thing I'd change from the above is that I don't think we really need "Adding a checkbox to say "this particular feeder is using a specific pipeline or one of the generic pipelines defined in the pipelines tab.". If a user has a feeder that requires a specific pipeline they can always create one in the global list and assign it. I'm open to arguments on that.

This would be a very large change, so if you are going to go ahead with it I'd suggest breaking it into small chunks, i.e. create the new pipeline manager area (Machine Setup -> Vision -> Pipelines), then convert one class to use it, like Strip Feeder. We'd merge that and let users try it and see what they think before converting everything that uses pipelines.

pfried, 2020-05-23:
> One thing I'd change from the above is that I don't think we really need "Adding a checkbox to say "this particular feeder is using a specific pipeline or one of the generic pipelines defined in the [...]

I do agree with that, simplifies the setup

> This would be a very large change, so if you are going to go ahead with it I'd suggest breaking it into small chunks, i.e. create the new pipeline manager area (Machine Setup -> Vision -> [...]

I did as suggested (with small teaks as the camera)

pfried, 2020-06-02:
Ok we discussed the specification of separate classes should not be there for the implementation, instead one general class should be used.

* We need to solve the "defaults" issue somehow. (I would suggest a "catalogue", e.g. loading them from a file could be an option).
* Do we save the defaults on creation?

I would remove the BoardSide thing and we would assign a camera directly (default is up camera).

markmaker, 2020-06-02:
Hi

Before #1013 I was also thinking about a solution. 

The difference in my approach was to not only include the Pipeline, but the whole Vision task i.e. **including** additional Vision settings aside from the Pipeline (see the two screenshots below for examples) **and** the behavior i.e. the class providing the Vision task operation. IMHO those belong together and this is also how Jason designed Bottom and Fiducial Vision, if you look closely.  

OpenPnP already has the Vision branch in the Machine Setup tree. Currently, only one instance per Vision Type is allowed, but internally this is already an object with interface `PartAlignment` and `FiducialLocator` respectively and implementations `ReferenceFiducialLocator` and `ReferenceBottomVision`.

These Vision task objects could be given a common base interface and proper instantiating dialog etc. For better overview, the different interfaces could be separate Machine Setup Tree branches.

[image]

Furthermore, Jason already envisioned that these Vision task objects be dynamically assigned to parts and there is already infrastructure in place, but dormant:

https://github.com/openpnp/openpnp/blob/556e6f02492fc43f49aa0877e4909b2fc0451d75/src/main/java/org/openpnp/spi/base/AbstractPnpJobProcessor.java#L84-L89

I would build on top of that and add a rule based assignment. Of course, you could still assign the Vision task reference to a single package or even part (even though IMHO the latter is a design mistake) but you could also leave the assignment open in the package/part and let more powerful automatic rules take over. Those could look at properties like overall footprint size, pad count etc.  so if you later imporrt new packages/parts from the ECAD (and assuming the import includes the footprint), the matchup is automatic. Most importantly, we could add different class implementations for the same task, like footprint matching Bottom Vision for instance. Or Bottom Vision for large parts with multiple part corner shots. 

One thing is very clear for me: Those Vision tasks are integral recipes and the Pipeline should only be one of the ingredients. Btw. it can also be easily imagined that some Vision task will even have multiple Pipelines. 

Vision settings from Feeders etc. would have to be transplanted into separate Vision object classes and their interfaces and classes registered with the machine. Most of the time it would only be a Pipeline i.e. simple to implement a "delegate" style class and do the transplant. But for some feeder classes it would be much more, e.g. the OCR settings etc. for the `PushPullFeeder`. I would of course take over the work for that one: 

[image]

The transition could be made one by one, starting with Bottom and Fiducial Vision. 

Looking at #1013 much of the work could still be quite easily refactored, as **the basic idea remains the same,** i.e. the **reference semantics** remains the central issue. An abstract  Vision provider base class could even add an interface that does not much more than delegate to the Pipeline so the interface would remain the same. 

_Mark

pfried, 2020-06-03:
@markmaker I agree that vision is more than the Pipeline, I already figured that out when doing this PR.

Doing Vision like you propose requires a kind of "runnable" interface where Vision is a step in the state machine of placing and gets called on different objects (e.g. part, fiducial, etc). I guess it would be best to just create a flexible event system where those things happen based on events (at a certain priority). Otherwise we end up stuffing in new steps all the time. (Sry @vonnieda I havent read your refactoring plans for the JobProcessor / other parts, maybe there are better ideas).

The vision would "take over" the machine for a while and would return some kind of result to a result interface of some form?

markmaker, 2020-06-03:
Hi @pfried 

Sorry, I don't understand. I was just thinking in terms of use case specific methods on these Vision task classes, i.e. the interfaces will be specific.  I think most will not be employed directly by the JobProcessor/Step but by another object, like a feeder. The Vision task / interface is specific to the use case i.e. one Feeder will do some positional calibration (StripFeeder), the other determines whether a cover is open or closed (BlindsFeeder), the third reads an OCR label (PushPullFeeder), etc.

_Mark

pfried, 2020-06-07:
@markmaker Okay maybe a misunderstanding from my side. So objects owning the vision object would perform actions like you described, but the vision itself does not interact with the machine?

I thought about how the "take a picture of all four corners" would work, since there the vision itself would be responsible for performing a machine movement. How would you address this?

markmaker, 2020-06-07:
Hi @pfried 

That interface already exists:

https://github.com/openpnp/openpnp/blob/5131a7483863394f55ce70249cff3a145bfed80e/src/main/java/org/openpnp/spi/PartAlignment.java

You create a new class to implement the interface and override `findOffsets()`. Everything then happens inside there:

For multiple corners bottom vision you would just move the nozzle to four slightly different locations so that shot for shot the part corners are in the center of the bottom camera (the system would need to know the footprint or at least the part size for that e.g. imported from the E-CAD). The pipeline would probably recognize individual pads/balls/pins and those would be matched against the (partial) footprint of that corner. From (up to) four shots, the true center of the part can then be calculated. The part can be much larger than the camera view. 

It's something I'll definitely need if I want to place my largest parts. My camera view is too small for those. But it's just an idea...

_Mark

pfried, 2020-06-12:
Ok , tbh I cannot oversee all consequences of your vision proposal. I discussed the PR with @vonnieda on May 26th on the discord channel and one concern was that I derived spezialized classes for e.g. the Feeders. This is contrary to your idea since the proposal from @markmaker would definitly need such classes as we cannot have a one fits all approach here (matching parts, vs matching fiducials etc.)

So a vision object would solve the referencing issue in the first place (but only for their respective subclasses or at least compatible subclasses), but all custom behaviours would then need to be subclassed implementing the specific behaviours like PartAligment etc.

For the most classes this would be passing through the pipeline, but for others it could be some custom stuff like Part detection ans so on.

Following @markmaker I call them `Vision` instead of `Pipeline`

* Vision
  * PartAlginmentVision (implementing `findOffsets`)
  * PipelineVision (as a drop in replacement for pipeline only vision)
  * PushPullFeederVision
  * BottomVision
  * PartVision
  * ...

Part detection matching and so is doable, but this would also go into the subclass `PartVision`.

For the defaults restoration I would say the subclasses should load their defaults from a File like before.

@vonnieda Would you agree? Migration would be step wise as you proposed.

markmaker, 2021-12-30:
The most important steps towards this are in the Pull Request #1318.