openpnp/openpnp #844: FR : OK / cancel button on pipeline window

JulienDevillers, 2019-05-16:
# Feature Request
## Describe the Feature
Set the pipeline editor window modal and add ok/cancel button.

## Why Do We Need It?
A user edits a pipeline.
He does some changes.
Changes are a bad idea, he wants to get back to initial pipeline.
He can't do it.

It's a "standard" behaviour in softwares.

I can't see any reason to show several pipelines editrors at the same time. Does anybody has some?
And you can open several editors on the same pipeline. Can be confusing.

vonnieda, 2019-05-17:
Agreed on all points - sounds like a good improvement.

markmaker, 2021-04-25:
Seems to be modal for some time. Confirmation dialog was added by @tonyluken in #1098.

closed 2021-04-25 by vonnieda