openpnp/openpnp #846: linuxcnc connection problem

mylucille, 2019-05-17:
Hi, after I update to the new stable version from version 2018, I get this following error:

[image]

linuxcnc version: 2.7.0

Best Regards.

vonnieda, 2019-05-18:
Hi @mylucille, it looks like LinuxCNC dropped the connection after the "set mdi G21" command. Can you check your LinuxCNC logs and see if they show the dropped connection?

mylucille, 2019-05-18:
LinuxCNCrsh logs:

```
connected to x
linuxcncrsh: eof from client
linuxcncrsh: disconnecting client x (1.1)
```
Seems to be Openpnp bug, cause it was working fine before the update.

vonnieda, 2019-05-18:
Please post your question to the discussion group: http://groups.google.com/group/openpnp

Perhaps someone else has seen this issue there.

closed 2019-05-18 by vonnieda