openpnp/openpnp #851: Single physical camera for both top/bottom vision

darenschwenke, 2019-06-01:
# Feature Request
## Describe the Feature
I would like to use a single physical camera to feed both the top and bottom vision pipeline.  This is currently not possible (within OpenPNP).  The pipeline starts at the hardware, sets exposure and other parameters for the camera, and then the camera is streamed.  The pipeline can be fooled into thinking the one physical camera is two cameras using gstreamer, but that is not really the right place to solve this problem and does not address the issue of using different options per virtual camera.

I propose putting this capability into OpenPNP instead.  I am opening this issue to gather suggestions as to the best way to implement this, and then I will do what I can to make that happen.

## How Will It Look?
No changes to the user interface are needed, but a live view of both virtual cameras will probably not work, as each state may require a different exposure or resolution setting.

## Why Do We Need It?
1. What problem does it solve?
A machine which utilizes a bottom vision camera mounted to the head, which is used as the top vision camera as well.
2. Is it useful for everyone, or does it only solve a problem for your machine?
Probably just my target machine for the time being.

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.
Yes, there are commercial machines I have seen which utilize this.  No, I do not have examples.

vonnieda, 2019-06-01:
I'm working on this today actually. This is required for the CHMT36VA port which is a big focus of my work currently. In my case the two cameras are physically separate, but they share a single interface and are switched with a relay. I'm creating a MuxCamera that will let two Camera instances share the same source camera.

Having separate exposure and other parameters might be pretty difficult, so that is definitely phase 2, but we'll see how it goes. Expect to see some progress on this by Sunday.

darenschwenke, 2019-06-02:
Thank you!  I've worked around the exposure issue thus far via careful control of lighting, but it is far from perfect.

vonnieda, 2019-06-13:
This is now available in the develop branch. See https://github.com/openpnp/openpnp/wiki/SwitcherCamera

closed 2019-06-13 by vonnieda