openpnp/openpnp #856: Visual homing successful only the first attempt

tdarlic, 2019-06-14:
# Problem Report
Visual homing causes distance to homing fiducial from mechanical homing position to be added every time the machine homes

## Steps to Reproduce
1. Start OpenPnP
2. Click the Power On button.
3. Home the machine - machine homes fine
4. Home the machine again and the machine searches for homing fiducial on x,y coodinates that are multiple of coordinates set up int the machine.xml. Every other attempt increases the distance the machine tries to home further away

## Expected Result
Machine should home to visual fiducial

## Actual Result
Machine tries to find homing fiducial location only during the first homing attempt. Every other attempt it just adds distance to the distance it searches the homing fiducial. Only way to get it to home correctly is to reset the OpenPNP and the Smoothieboard. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
https://youtu.be/lS0tALguZuE

https://pastebin.com/Baq8w8E8

markmaker, 2019-06-15:
Hi  tdarlic

It seems to be a Computer Vision pipeline problem.

`2019-06-14 23:11:54.202 ReferenceFiducialLocator DEBUG: No matches found!
`
Be sure to switch off automatic exposure of your camera. Computer CV needs repeatable exposure. The camera slowly adapts to the lighting level, so I suspect the second homing, coming from a different view has the camera "pumping" with auto-exposure.

Another problem source could be the camera settling time. May need to lengthen that for your camera. But try the manual exposure first, as manual exposure allows for a shorter settle time, obviously.

Hint: ask in the google group next time. Issues need only be opened if something cannot be solved with community help:
https://groups.google.com/forum/#!forum/openpnp

_Mark

tdarlic, 2019-06-15:
Hi. I am pretty positive that it is not the vision. Let me try to explain.
It homes mechanically ok. Then it moves the camera above homing fiducial
and homes visually perfectly. But only the first time. Next time it homes
it homes mechanically ok but then it moves over the homing fiducial and
overshoots it by far. Off course it fails to find the fiducial because it's
not there. Next homing in overshoots even more and so on.
I've checked the vision pipeline and it locates the fiducial perfectly
every time. Camera settings are not set to auto.

Sorry for posting to github. I was so positive that this is bug because I
was using machine for 3 days and this started once I've updated to new
version.

markmaker, 2019-06-15:
Hi 

the log shows normal behaviour on the second homing, it issues `G0 X10.4875 Y2.5000` as expected (log section below).

I also have a Smoothie and I don't need the `M82` G-Code, `G90` does the extruder/rotation axes too.  Try removing the M82 from your CONNECT_COMMAND.

```
2019-06-14 23:11:51.330 ReferenceMachine DEBUG: homing machine
2019-06-14 23:11:51.330 ReferenceMachine DEBUG: setHomed(false)
2019-06-14 23:11:51.331 ReferenceHead DEBUG: H1.home()
2019-06-14 23:11:51.331 GcodeDriver DEBUG: sendCommand(G28 ; Home all axes, -1)...
2019-06-14 23:11:53.367 GcodeDriver DEBUG: sendCommand(serial://COM6 G28 ; Home all axes, -1) => [ok]
2019-06-14 23:11:53.369 ReferenceFiducialLocator DEBUG: Looking for FIDUCIAL-HOME at (10.500000, 2.500000, 0.000000, 0.000000 mm)
2019-06-14 23:11:53.370 ReferenceHead DEBUG: H1.moveToSafeZ(0.99)
2019-06-14 23:11:53.370 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.99)
2019-06-14 23:11:53.371 ReferenceCamera DEBUG: Head.moveToSafeZ(0.99)
2019-06-14 23:11:53.371 ReferenceActuator DEBUG: A1.moveToSafeZ(0.99)
2019-06-14 23:11:53.371 ReferenceCamera DEBUG: moveTo((10.500000, 2.500000, NaN, 0.000000 mm), 0.99)
2019-06-14 23:11:53.379 GcodeDriver DEBUG: sendCommand(G0 X10.4875 Y2.5000   F11880, 15000)...
2019-06-14 23:11:53.380 GcodeDriver DEBUG: sendCommand(serial://COM6 G0 X10.4875 Y2.5000   F11880, 15000) => [ok]
2019-06-14 23:11:53.381 GcodeDriver DEBUG: sendCommand(G1 X10.4875 Y2.5000   F1188, 15000)...
2019-06-14 23:11:53.383 GcodeDriver DEBUG: sendCommand(serial://COM6 G1 X10.4875 Y2.5000   F1188, 15000) => [ok]
2019-06-14 23:11:53.384 GcodeDriver DEBUG: sendCommand(M400 ; Wait for moves to complete before returning, 15000)...
2019-06-14 23:11:53.696 GcodeDriver DEBUG: sendCommand(serial://COM6 M400 ; Wait for moves to complete before returning, 15000) => [ok]
2019-06-14 23:11:53.700 ReferenceCamera DEBUG: moveTo((10.500000, 2.500000, 0.000000, 0.000000 mm), 0.99)
2019-06-14 23:11:53.702 ReferenceHead DEBUG: H1.moveToSafeZ(0.99)
2019-06-14 23:11:53.702 ReferenceNozzle DEBUG: N1.moveToSafeZ(0.99)
2019-06-14 23:11:53.703 ReferenceCamera DEBUG: Head.moveToSafeZ(0.99)
2019-06-14 23:11:53.704 ReferenceActuator DEBUG: A1.moveToSafeZ(0.99)
2019-06-14 23:11:53.704 ReferenceCamera DEBUG: moveTo((10.500000, 2.500000, NaN, 0.000000 mm), 0.99)
2019-06-14 23:11:53.705 ReferenceCamera DEBUG: moveTo((10.500000, 2.500000, 0.000000, 0.000000 mm), 0.99)
2019-06-14 23:11:54.202 ReferenceFiducialLocator DEBUG: No matches found!

```

_Mark

vonnieda, 2020-04-21:
Closing due to no further response. Please reopen if the bug still exists.

closed 2020-04-21 by vonnieda