openpnp/openpnp #861: NullPointerException with USB camera

jymoulin, 2019-06-25:
# Problem Report
## Describe the Issue

After configuring an OpenPnpCaptureCamera, log fills with:

2019-06-25 16:41:46.643 SystemLogger ERROR: java.lang.NullPointerException
2019-06-25 16:41:46.643 SystemLogger ERROR:     at org.openpnp.machine.reference.camera.OpenPnpCaptureCamera.run(OpenPnpCaptureCamera.java:199)
2019-06-25 16:41:46.643 SystemLogger ERROR:     at java.lang.Thread.run(Thread.java:748)

## Steps to Reproduce
1. Plug USB Camera
2. Start OpenPnP
3. Go to 'machine setup' -> Cameras
4. Click '+' to add a OpenPnpCaptureCamera
5. Select newly added camera -> 'Device settings'
6. Choose USB camera and click 'Apply'.

## Expected Result
Image from camera in camera view.

## Actual Result
Camera view stays black and log filled with NullPointerException.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
OS: macOS Mojave 10.14.5
OpenPnp: 2019-06-01_21-53-14.3891a53
USB Camera (x2): unknown USB model (from OpenPlacer) but working with other software.
USB Hub: tried two different hub (two different brand)

markmaker, 2019-06-25:
Hi

what version of OpenPNP do you use?

_Mark

jymoulin, 2019-06-25:
It's written at the end :-) 

OpenPnp: 2019-06-01_21-53-14.3891a53
(I downloaded it a couple of hours ago)

markmaker, 2019-06-25:
Sorry, I guess the version _is_ somehow encoded into that timestamp but I don't know how to interprete that.

What I meant to ask is what version (stable or latest) did you download and/or which branch did you checkout and build.  

_Mark

jymoulin, 2019-06-25:
Oh, ok sorry. This is the stable (1.0) version downloaded this morning from openpnp.org.

vonnieda, 2019-06-25:
Probably related to a similar issue in https://github.com/openpnp/openpnp/issues/860. 

@markmaker For the version, the hex digits after the last period are the Git SHA shortcode for the commit. If you run `git log 3891a53` you'll see the commit info.

vonnieda, 2019-06-25:
Oh, I forgot Github will helpfully link the commit for you too: 3891a53

markmaker, 2019-06-25:
Cool!

But 3891a53 is OpenPNP 1.0 so #860 should not apply.

The line of the exception points to the `stream ` inside the camera:
https://github.com/openpnp/openpnp/blob/3891a530f007fb85007164d02bd503986abd43bf/src/main/java/org/openpnp/machine/reference/camera/OpenPnpCaptureCamera.java#L199

It seems it can't open the stream:
https://github.com/openpnp/openpnp/blob/3891a530f007fb85007164d02bd503986abd43bf/src/main/java/org/openpnp/machine/reference/camera/OpenPnpCaptureCamera.java#L296-L302

Exceptions are in deed swallowed. But they should appear further up in your log @jymoulin. Please send the full log. 

_Mark

jymoulin, 2019-06-25:
Here is a 'fresh' log (I quit OpenPNP, removed all log, then start OpenPNP and added usb cam). Unfortunately, there is not much more information.

[OpenPnP.log](https://github.com/openpnp/openpnp/files/3326087/OpenPnP.log)

markmaker, 2019-06-25:
I think you need to 

1. remove the camera, 
2. increase the log level. 
[image]
3. exit OpenPNP
4. restart OpenPNP
5. add the camera again
6. wait a bit
7. exit 
8. send the log.

Note that the log is automatically cycled. No need to delete anything.

Go to your home/.openpnp/log

_Mark

jymoulin, 2019-06-25:
Here is the new log with all level in TRACE:

[OpenPnP.log](https://github.com/openpnp/openpnp/files/3326195/OpenPnP.log)

markmaker, 2019-06-25:
Hmmm.. have you really deleted **all** the cameras? It seems you have a Open**Cv**Camera still there. If it connects to the _same_ USB camera it might explain the problems.

_Mark

markmaker, 2019-06-25:
Note that cameras can be both on the machine and on the head. Clean them all out.

_Mark

jymoulin, 2019-06-26:
Sorry, this was a test meanwhile. The log with openCvCamera removed is almost the same :

[OpenPnP.log](https://github.com/openpnp/openpnp/files/3328889/OpenPnP.log)

markmaker, 2019-06-26:
It seems the `stream` cannot be opened. And OpenPnpCaptureCamera.run() does not guard against `stream == null` as it probably should. 

Looking at the code I would expect some log entries from openpnp-capture:
https://github.com/openpnp/openpnp-capture/blob/5ebafd8b805f63b4f48019148ad1af5a9b83afc5/win/platformstream.cpp#L169

Do you have "System Output" enabled in the "Log" tab? 

If not, enable it and then restart OpenPNP and send the log. 

@vonnieda am I right to assume this switch does add the openpnp-capture log entries to the OpenPNP Log display and file?

_Mark

jymoulin, 2019-06-26:
The 'System Output' is already enabled

markmaker, 2019-06-26:
I see.

Can you try to start OpenPNP from a console? There seems to be a way, starting with a script/batch:
https://github.com/openpnp/openpnp/wiki/Quick-Start#install-and-run

Personally I start from the build environment. When I do, I get some output from openpnp-capture on the console. 

_Mark

jymoulin, 2019-06-26:
Started it from console, and it works. But my shell asked me to allow camera access...

So I checked in "Security -> Privacy" systems settings if OpenPnp was allowed too. It wasn't... I don't know why. I don't remember it asking for camera access.

Anyway, after allowing OpenPnp camera access, I can see camera too in OpenPnp (started normally).

Perhaps can we add a check for this setting somewhere ?

Thank you guys for your help !

markmaker, 2019-06-26:
@jymoulin could you please exit OpenPNP, then disallow camera access in "Security -> Privacy", then try again with the console and see if the error message by openpnp-capture would have pointed you to the **source** of the problem?

@vonnieda I guess the most versatile solution would be to capture error output from openpnp-capture in the OpenPNP log. But I'm afraid I havent the slightest idea how to do that. :-(

_Mark

jymoulin, 2019-06-26:
Ok, with camera access disabled, here is what I get on the console :

objc[42023]: Class PlatformAVCaptureDelegate is implemented in both /var/folders/dj/brlgyj7j3zd7ddfdsd1glhnr0000gn/T/jna-105726/jna6987893077435353616.tmp (0x150bd5bb0) and /var/folders/dj/brlgyj7j3zd7ddfdsd1glhnr0000gn/T/jna-105726/jna5204774583976095871.tmp (0x150bf9bb0). One of the two will be used. Which one is undefined.
OSX Unique ID is not exactly 18 characters - wrong format to extract location.
We might have trouble identifying the UVC control interface.
2019-06-26 12:06:37.484 Main DEBUG: Bienvenue, Bienvenido, Willkommen, Hello, Namaskar, Welkom, to OpenPnP version 2019-06-01_21-53-14.3891a53.
2019-06-26 12:06:37.518 Scripting TRACE: Scripting.on Startup
[ERR ] Error opening native device Cannot use USB2.0 PC CAMERA #2
[ERR ] Could not open stream for device USB2.0 PC CAMERA #2 (Unknown)
2019-06-26 12:06:57.030 MainFrame INFO: Shutting down...
2019-06-26 12:06:57.407 ReferenceMachine DEBUG: setEnabled(false)
2019-06-26 12:06:57.408 ReferenceMachine DEBUG: setHomed(false)
2019-06-26 12:06:57.412 MainFrame INFO: Shutdown complete, exiting.

The OpenPnP.log is the same as before, full of NullPointerException.