openpnp/openpnp #868: OpenPnP 2: Vaccum system totally unusable

hukedev, 2019-07-01:
# Problem Report
## Describe the Issue
The current vaccum system seems to be totally unuseable for my machine.

## Steps to Reproduce
Run a job with enabled vaccum actuators.

## Expected Result
- place part (close vac, open exhaust)
- move to safe-z
- measure vac
- proceed

## Actual Result
- place part (close vac, open exhaust)
- move to safe-z
- performs a pick command ? (close exhaust, open vac)
- measure vac, we get unusable values, job fails

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Tested with actual openpnp2.
Ccompiled with eclipse jdk 1.8_212, eclipse 2019-06 @ windows x64.

Had to disable vac system to be able to finish some jobs.
Clicking in machine setup panel on items caused few exception errors again.

vonnieda, 2019-07-01:
The pick (third bullet point in Actual Result) is intended to turn on the vacuum to check if the part is off. The idea is that without the vacuum being on we can't measure vacuum levels.

To fix this, you should just need to set your Part Off vacuum level, as shown here:

[image]

hukedev, 2019-07-01:
I know how to setup the four values - normally two values would be fairly enough for all normal vaccum procedures and so much setup parameters makes everything much more complicated.

What i can not understand is: why are this _completely useless **solenoid life time killing** pick & place commands_ inserted after a part placement. This make absolutely no sense. It costs only valuable time as vacuum measurement decay times are added. 

Normally when part is placed, after nozzle+tip dwells and z retract, the vaccum decay time is short enough to get good values for a check if a part sticks on nozzle and that happens on my machine only after a some thousand placements - so not very often.

Vaccum checking in a job makes only sense after pick from empty feeder to prevent moving empty all the way to bottom vision and or after discards.

Without enabled vaccum actuators everything runs fine with new openpnp2 except this new gui exception errors from machine setup panel (tested with jdk 1.8 & 11, for jdk 12 there are issues with simple-xml).

hukedev, 2019-07-01:
After some adaptions of `ReferencePnpJobProcessor.checkPartOff()` it seems to be working now as expected.

closed 2019-07-01 by hukedev