openpnp/openpnp #870: Memory leak in nozzle tip calibration

vonnieda, 2019-07-07:
Reported on IRC:

﻿i'm seeing a mem leak somewhere, not sure if it's only local, running nozzle tip calibration a few times will eat 100s of MB, even editing its pipeline and refreshing the image makes the memory usage go up and never down
linux, latest develop

vonnieda, 2019-07-08:
Possibly related to this other memory leak report: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/RKWw2q63wzI/nGOpCfarBAAJ

vonnieda, 2019-07-08:
I would guess there is a leak in one of the pipeline stages, or in the pipeline processor itself.

CircuitSetup, 2019-08-15:
I just noticed this on the latest dev branch, after bottom vision was taking a really like time, sometimes not working at all.

markmaker, 2019-08-15:
Hi

Having worked on both nozzle tip calibration and bottom vision I feel somewhat addressed... :-/

Though I can't remember having done anything to its vision. Other than adding CircleMask center properties. 

Am I right to assume that with Java's GC and seeing that Mat has a finalize() that calls release(), a true leak is not possible?

So either the leak is in the C code or a collection or linked list (or similar) is keeping objects alive?

_Mark

vonnieda, 2019-08-15:
> Hi
>
> Having worked on both nozzle tip calibration and bottom vision I feel somewhat addressed... :-/
>
> Though I can't remember having done anything to its vision. Other than adding CircleMask center [...]

Unfortunately no: https://github.com/opencv/opencv/issues/8683

Most likely we're just missing a release() somewhere. I just haven't had a chance to look at it.

> 
> So either the leak is in the C code or a collection or linked list (or similar) is keeping objects alive?
> 
> _Mark

markmaker, 2019-08-15:
I'm rubbing my eyes. Then how on earth is one supposed to use native ressources reliably in GC + exceptions + non-RAII environment?

_Mark

markmaker, 2019-08-15:
Ahh there is a closet RAII in Java:
https://www.baeldung.com/java-try-with-resources

It breaks the whole GC promise because it spreads along ownership like a virus. 

Information hiding - broken.
Code reuse - broken.
OOP - broken.
Java - broken. 

_Mark

markmaker, 2019-08-16:
Hi @vonnieda 

I've looked at 

- https://github.com/openpnp/openpnp/pull/825/files
- https://github.com/openpnp/openpnp/pull/815/files

and I haven't found any relevant changes.

I first thought that doing bottom vision in a loop here is a problem:

https://github.com/markmaker/openpnp/blob/a2fa6358cd74333293ba0acbd3a41a3c00c3eaea/src/main/java/org/openpnp/machine/reference/vision/ReferenceBottomVision.java#L122-L124

The `AutoCloseable` pipeline is in a "try with resources" **outside** the loop. The same pipeline is reused multiple times (the old code reused it once). But `release()` is called here:

https://github.com/openpnp/openpnp/blob/6aabf9e8170a4044c8126697685cc9c5d192e62a/src/main/java/org/openpnp/vision/pipeline/CvPipeline.java#L193-L195

So overwriting the result multiple times here is OK:

https://github.com/openpnp/openpnp/blob/6aabf9e8170a4044c8126697685cc9c5d192e62a/src/main/java/org/openpnp/vision/pipeline/CvPipeline.java#L240

There are "old" cases, where setters of references that may (indirectly) contain ownership of resources will not `release()` the former object.  Such as here:

https://github.com/markmaker/openpnp/blob/3bf7080da3cb730e4ec34cbd6f22c73194a1e7ac/src/main/java/org/openpnp/machine/reference/ReferenceNozzleTipCalibration.java#L856

The same is (probably) true, when the whole Wizard is overwritten with a new object selected in the machine tree.
 
https://github.com/openpnp/openpnp/blob/6aabf9e8170a4044c8126697685cc9c5d192e62a/src/main/java/org/openpnp/gui/MachineSetupPanel.java#L169-L173

With `finalize()` missing, those  are probably leaks. But that's **not** new. 

_Mark

vonnieda, 2019-08-16:
@markmaker For what it's worth, I don't think this is something you introduced. I think there is probably an error in one of the CvStage implementations that causes the memory leak, and that it's been there for a while. I don't have time to dig into it right now, but it's on my radar.

sadman, 2019-08-17:
i've spent some time today trying to figure out the source of the massive memleaks i'm seeing that appeared vision related and after few hours reading through vision code, adding release()s here and there, by dumb luck i was staring at the console, full of those "nashorn will be deprecated lines" and thought "hm, what if..."
you can guess the rest, even empty js scripts for before/after capture cause massive memleaks on my system and since they're called when doing vision, it fooled me to think it's vision related

i'll just hardcode my light handling directly into openpnp for now and try to make some boards to see whether the issue is really gone when scripting is not used

javnub, 2019-08-19:
On my production machine running w7x64/java8 i'm facing not any worryingly memory leaks with latest OpenPnP2, even after some dozen tip cals and hundreds placements. 

A complete problem report with provided machine.xml would be fine to be able to reproduce this hiccup.

CircuitSetup, 2019-10-10:
> EDIT: i've rewritten my event scripts to beanshell, no more leaks (or at least not that massive to easily notice using simple tests)

Can you share what you did here? This bug is really annoying me lately.

sgjava, 2019-12-15:
If you are relying on OpenCV Java wrappers it's OpenCV that's broken, not Java that is alluded to in an earlier post. https://github.com/sgjava/opencvmem details how OpenCV handles native memory and why it leaks. My OpenCV install script https://github.com/sgjava/install-opencv allows patching of the generated Java wrappers. https://valgrind.org can help you find native memory leaks while Netbeans profiler or JMC can find Java memory leaks.

In any even it's easier to track down memory issues by profiling instead of guessing if you missed a release().

CircuitSetup, 2020-04-20:
Has anyone figured anything out with this? I occasionally get major lag right after Camera.BeforeCapture is executed, even though I do not have anything for this. I tried creating a blank .bsh file with the same results. The bottom camera seems to freeze for, sometimes, minutes at a time. I cant find a pattern to why this happens. 

Restarting OpenPnp does not seem to help. Maybe it has something to do with usb bandwidth, and it just happens to manifest when Camera.BeforeCapture is executed.

CircuitSetup, 2020-04-21:
I observed the same behavior with nozzle tip calibration off. Although, it was happening less frequently. Here is the relevant snippet in the logs. Note that Camera.BeforeCapture.bsh is blank:
```
2020-04-21 08:13:33.992 GcodeDriver DEBUG: sendCommand(M811, 10000)...
2020-04-21 08:13:33.992 GcodeDriver TRACE: [serial://COM11] >> M811
2020-04-21 08:13:33.993 GcodeDriver TRACE: [serial://COM11] << ok
2020-04-21 08:13:33.993 GcodeDriver DEBUG: sendCommand(serial://COM11 M811, 10000) => [ok]
2020-04-21 08:13:34.244 Scripting TRACE: Scripting.on Camera.BeforeCapture
2020-04-21 08:13:34.245 Scripting TRACE: Scripting.on found Camera.BeforeCapture.bsh
2020-04-21 08:13:42.991 Scripting TRACE: Scripting.on Camera.AfterCapture
2020-04-21 08:13:42.992 Scripting TRACE: Scripting.on found Camera.AfterCapture.bsh
2020-04-21 08:13:42.995 ReferenceActuator DEBUG: UpCamLights.actuate(false)
2020-04-21 08:13:42.995 GcodeDriver DEBUG: sendCommand(M813, 10000)...
2020-04-21 08:13:42.996 GcodeDriver TRACE: [serial://COM11] >> M813
2020-04-21 08:13:42.996 GcodeDriver TRACE: [serial://COM11] << ok
2020-04-21 08:13:42.997 GcodeDriver DEBUG: sendCommand(serial://COM11 M813, 10000) => [ok]

```

doppelgrau, 2021-04-09:
Solved with PR #1007 or still open?

vonnieda, 2021-04-18:
@doppelgrau Likely solved - can be reopened if not.

closed 2021-04-18 by vonnieda