openpnp/openpnp #871: Pick retry error after deferred error

vonnieda, 2019-07-08:
Reported on IRC by AneoX:

AneoX: All placement are set to defer error handling in my job. So if some pick troubles occurs, openpnp tries to retry, very nice feature. But if retries count exeeded, even if i restart job with reset all placed, retries count become 1 for all problem placements. To reset retries count, i have to restart openpnp. I cant find bug in code, looks like its in deep of java exceptions.
*retries count become 0*

vonnieda: AneoX: When you say retries count becomes 0, do you mean the retries number gets changed on the feeder? Or it's just not resetting the number properly?
[09:25:32]  <AneoX>	@vonnieda Actually feeder settings do not change and retries dictionary in job planner class works ok. But openpnp will not retry anymore, if retries count exeeded in previous job run, just discards part and skip placement.
[09:26:53]  <@vonnieda>	AneoX: Thanks. I'll take quick look now and see if there is something simple.
[09:27:02]  <AneoX>	i think, some error noted in other memory, not retries dictionary, but i didnt find it in deep of exceptions
[09:34:17]  <@vonnieda>	I bet the issue is that the error state needs to be reset, and it's not getting reset. I'll take some notes and look at it more tonight. 
[09:38:07]  <@vonnieda>	AneoX: After you restart the job, does it even try to pick? Or does it skip right past everything?
[09:38:15]  <AneoX>	yes
[09:38:18]  <AneoX>	it tries
[09:38:23]  <AneoX>	but only 1 time
[09:38:52]  <@vonnieda>	Very strange

vonnieda, 2019-07-08:
Test emails subject "openpnp retries bug"

aneox, 2020-04-02:
https://github.com/openpnp/openpnp/blob/944b650dd64bd7cd8651d0a3f32fc05a18d2d10a/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L550

this comment says, that "If either postPick or checkPartOn fails we discard and then cycle back to feed up to pickRetryCount times.", but the function returns the same result in both cases(pick ok and pick fail). My machine doesn't retry pick with this code never. It doesn't matter, is it first run, or not. My research shows, that problem not in the placement error state. I think this [line](https://github.com/openpnp/openpnp/blob/944b650dd64bd7cd8651d0a3f32fc05a18d2d10a/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L569) should not return same result as normal function return.

My java skill not good, and I do not understand how this code should cycle back.
So for this moment, I have added some for cycle code in pick function to retry pick, but in this case, I can't stop the job if the feeder stop working, until retries count exceed. Please help me make it work, as expected)

vonnieda, 2020-12-03:
Also see discussion in https://groups.google.com/g/openpnp/c/kbnMKnDArIk/m/Sk7XRC2IBgAJ

vonnieda, 2020-12-03:
WIP branch: https://github.com/openpnp/openpnp/tree/feature/fix_pick_retry

vonnieda, 2020-12-24:
#1070 related

vonnieda, 2020-12-29:
This has been fixed in https://github.com/openpnp/openpnp/commit/c7f0a313e0434e026ca5c9568b232d01eb000dd4. It is currently available to test in the test branch at https://openpnp.org/test-downloads/.

closed 2020-12-29 by vonnieda