openpnp/openpnp #874: Head doesn't manually move in Z with camera selected.

wiegleyj, 2019-07-16:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
Z axis doesn't respond to manual movement requests when a head camera is selected in the pendant/movement panel.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Select a head camera.
2. Click on Z up or down.

## Expected Result
I expect the camera to move so that I can focus on parts/feeders/boards at different heights.

## Actual Result
Movement request is ignored.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
I sort of get a concept of "don't move the camera" up or down because you may not remember where the nozzle is and crash it. But this is a manually requested move and I think should be serviced. Especially since X/Y moves are serviced and those have a far greater chance of Nozzle problems if you forgot where your nozzle is. Preventing Z movements just prevents moving to a proper focus distance when working with the camera and its coordinate frame directly.

vonnieda, 2019-07-16:
Are you sure you don't have your axis mapping set to ignore the camera on Z moves? Most people do have that set: https://github.com/openpnp/openpnp/wiki/GcodeDriver#nozzle-is-moving-or-turning-when-camera-is-moved

wiegleyj, 2019-07-16:
Thank you!!! Yes the Z axis had "N1" I changed it to "*" and now the camera moves as expected when requesting manual Z movements.

wiegleyj, 2019-07-16:
This also fixes the camera moving to focus! Thank you! I'll comment there and then close that issue as well.

closed 2019-07-16 by wiegleyj