openpnp/openpnp #875: Soft Limits are calculated incorrectly.

wiegleyj, 2019-07-16:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
When attempting to move to (10,0,32) OpenPNP responds with "Can't Move Head to (10.00000, 0.000000, 32.0000000, 0.000000mm), outside of soft limits on head H1." DRO is indicating correct current position of (0,0,32) since my nozzle has an offset of (0,0,0). My machine limits are (303,190,32) I have plenty of space to move 10mm to the right.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Set proper offsets for the head camera and your camera should be to the Positive side of the nozzle. (This issue is the result of the nozzle offsets being relative to the camera and not the head.)
3. Attempt to move in the X direction.
4. The machine fails to jog and shows an error.

## Expected Result
If the nozzle is showing (0,0,0) and it's offsets are (0,0,0) then I should just be requesting an absolute head movement and it should work.

## Actual Result
It somehow thinks the desired coordinate is outside of the workspace and refuses to move.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Sorry I'm submitting too many issues. They're all related to offsets on how offsets are thought of in this system. One more tonight and then I'll stop.

markmaker, 2020-04-17:
Hi @wiegleyj 

I have worked extensively with soft limits, and I haven't encountered any issues. 

However, you must be aware that soft limits are related to camera coordinates (like most things in OpenPNP). So if you have a camera offset other than (0, 0, 0), it may be hard to follow.

Note, it is recommended to set the camera offsets to (0, 0, 0) and have a nozzle offset instead. 

_Mark

surabibio, 2021-01-01:
Hi Mark,
I don't see any soft limit for Z. I have banged the nozzle many times in the past week and the nozzle is nearly broken. 
Ravi Ganesh

markmaker, 2021-01-01:
Hi @surabibio 

See here:
https://github.com/openpnp/openpnp/wiki/Machine-Axes#kinematic-settings--axis-limits

But you must have the newest OpenPnP 2.0 Version for this.

See also this:
https://github.com/openpnp/openpnp/wiki/Advanced-Motion-Control

_Mark

markmaker, 2021-04-25:
I believe the original concern does no longer apply. With #1035, soft limits are now raw axis coordinates with their own special capture buttons. 

[image]

This was moved from the head to the axes to add support for the Z axis (of which there can be many per head) and to cover cases where axes are shared i.e. soft limits are transformed to different HeadMountables in reversed/transformed form and must still be consistent and make sense when violations are reported to the user. The raw axis is the only fixed frame of reference where the user can know what the "lower limit" means. 

See also:
https://github.com/openpnp/openpnp/wiki/Machine-Axes#referencecontrolleraxis

vonnieda, 2021-04-29:
Agreed - thanks @markmaker

closed 2021-04-29 by vonnieda