openpnp/openpnp #877: nozzle Offset Wizard calculates Z wrong

wiegleyj, 2019-07-16:
# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
## Describe the Issue
The nozzle offset wizard is not doing the same calculation for the Z offset that is doing for the X/Y offsets.

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Zero all offsets in nozzle and camera.
3. move the nozzle to a mark position.
4. Click "Store nozzle mark position". Nozzle Mark position is recorded at current machine coordinates (no offsets applied). [my case: (32.7,7,4).
5. Move camera above mark and to proper focus.
6. Click "Calculate nozzle offset"
7. Z offset is incorrect.

## Expected Result
If the nozzle mark position is N and the camera position is C, then nozzle offsets should be C-N.
My nozzle mark position is (32.7, 7, 4). My Camera position is (0.7, 8, 17). That means my camera is to the left  and above my camera. Its offsets should be (-32,1,-13). (left and below camera).

## Actual Result
Offsets are calcuated as (-32,1,-4). The Z is wrong.

It is not the result of C-N, but rather (Cx, Cy, 0.0) - N.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
This is my original problem from yesterday that you asked me to file an issue for. Sorry for many issues at once. They all came up as a result of testing this and having a very difficult time with lots of Z axis items. There may be a stronger underlying problem with Z calculations that is related to and causing all these issues.

wiegleyj, 2019-07-16:
Ok.... I think this may have also been solved by the "axis z" being mapped to only N1. Since changing it to "*" (all axis) and trying again It seems to work correctly.

vonnieda, 2019-07-30:
Thanks @wiegleyj, based on the above comment I'll close this, but if you find it's still happening please feel free to reopen.

closed 2019-07-30 by vonnieda