openpnp/openpnp #879: Question: OpenPnP structure

hukedev, 2019-07-27:
@vonnieda  
Dear Jason !

Regarding  #864 and #843 i am trying with my very modest knowledge of java to give every part an additional id that links to a pipeline. In the pipeline editor new ones can be added, renamed or a already existing could be selected. 
[image]

I would like to try storing all pipelines in a separate file (pipelines.xml), each pipeline would get an user defineable name and an unique id.

When looking at current config files, there are
- parts.xml
- packages.xml
- machine.xml

Inside machine.xml there is a block called `part-settings-by-part-id`.

Trying to understand the structure of OpenPnP i would like to ask you what is (or was) the reason for keeping parts.xml in a separate file? Is this for setup's without any bottom-vision?

_Postscript: I'm just trying to make it fun - so please do not waste too much of your precious free time with my beginner questions - thanks._

hukedev, 2019-07-29:
Made some progress. Main config file machine.xml is smaller now...
Sorry for any inconvenience.

closed 2019-07-29 by hukedev

vonnieda, 2019-07-30:
Apologies for not answering before you figured it out, it's been a very busy few weeks. If you have any further questions about it please feel free to ask and I will try to be more responsive. And please consider sharing what you've written!

Thanks,
Jason

vonnieda, 2019-07-30:
And in response to the question: I originally wanted part and package definitions to be sharable across the community, so I tried to keep machine specific information always in machine.xml, and things that might be useful to others in parts and packages.

I never really got around to making the information sharable, and it would have been relatively easy to remove the information at the time of sharing, so all in all it wasn't worth the complexity.

javnub, 2019-08-13:
Thank you for your kind reply and a big Sorry for my short term memberships here and there, especially on google based services ...

After playing around for several evenings i realize how nontrivial and complex this simple feature request can get. I altered every class that needs a pipelineId, stored the pipelines in a `LinkedHashMap `(`configuration.java`) and got the modifed `CvPipelineEditorDialog` working so far. 

But i'm still fighting with java classes - e.g. `PartAlignment`: The list is accessed with 

`Configuration.get().getMachine().getPartAlignments()`

How can a single `PartAlignment` be altered (maybe as one-liner) when i have got `part.getId()` only?

vonnieda, 2019-08-13:
You can use `AbstractPnpJobProcessor.findPartAligner(Machine, Part)`. This is a simple method that just loops through `Configuration.get().getMachine().getPartAlignments()` calling `canHandle(Part)` to see which one works for the Part.

For intuition on how you could find that code, what I normally do is find a class I'm interested in, in this case `PartAlignment` right click in Eclipse and select References -> Workspace. This will show all references to that class in the workspace and you can use that to find methods that work with that class. This also works with methods, variables, etc. It's one of the most powerful tools for finding your way through the codebase.

If you don't use Eclipse, you can always use grep, or your editors search function, but it's not as aware of context as the reference search.

javnub, 2019-08-13:
Yes, i'm using eclipse/ubuntu lastest versions for playing with code.
For production i prefer a rock-solid windows machine.

Thanks to your help I was finally able to change the pipelineId in the corresponding class:
[image]
Now i have got to add all the other `sourceType` cases (it's a parameter passed to `CvPipelineEditorDialog`) like Fiducuals, NozzleTips and Feeders.

javnub, 2019-08-18:
Regarding sharing i want to show you what i have done so far.

I created seven default pipeline-ID's that can not be deleted or renamed.
When a pipeline is added then it gets an unique id, when an added pipeline is deleted, then all items (parts, tips, feeders) that uses this pipeline-ID are set back to their default pipeline-ID.

There have been made a lot of changes - so the code is not intended for use in production - it's just a faint attempt to find a way to solve this long-cherished desire.
[pipelinepool.zip](https://github.com/openpnp/openpnp/files/3513015/pipelinepool.zip)