openpnp/openpnp #881: Board location not updated via fiducial check when panelized

CircuitSetup, 2019-08-15:
# Problem Report
After panelizing a board, and running a fiducial check, the board location is not updated. This goes for any board in the panel.  It seems as though the location is saved somewhere, however since it does tell the machine to go to the new set location. 

## Steps to Reproduce
1. Load a board using version 2019-07-30_02-44-25.6aabf9e
2. Autopanelize that board
3. Run a fiducial check on the first board
4. Note that X Y and Rot. are not saved

## Expected Result
Board location is updated after running fiducial check.

## Actual Result
Board location is not updated at all.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
Here is the log file output of doing a fiducial check:

`2019-08-15 15:00:44.394 ReferenceFiducialLocator DEBUG: Locating FD2
2019-08-15 15:00:44.395 ReferenceFiducialLocator DEBUG: Looking for FIDUCIAL-MICRO-FIDUCIALUFIDUCIAL at (206.532389, 130.487244, -20.000000, 0.144356 mm)
2019-08-15 15:00:44.395 ReferenceHead DEBUG: H1.moveToSafeZ(1.0)
2019-08-15 15:00:44.395 ReferenceNozzle DEBUG: N1.moveToSafeZ(1.0)
2019-08-15 15:00:44.395 ReferenceNozzle DEBUG: N2.moveToSafeZ(1.0)
2019-08-15 15:00:44.395 ReferenceCamera DEBUG: Down Camera.moveToSafeZ(1.0)
2019-08-15 15:00:44.395 ReferenceActuator DEBUG: NozzleDragN1.moveToSafeZ(1.0)
2019-08-15 15:00:44.395 ReferenceActuator DEBUG: Vac1Sense.moveToSafeZ(1.0)
2019-08-15 15:00:44.395 ReferenceActuator DEBUG: Vac2Sense.moveToSafeZ(1.0)
2019-08-15 15:00:44.396 ReferenceActuator DEBUG: FeederTape.moveToSafeZ(1.0)
2019-08-15 15:00:44.396 ReferenceActuator DEBUG: NozzleDragN2.moveToSafeZ(1.0)
2019-08-15 15:00:44.396 ReferenceActuator DEBUG: FeederTape_long.moveToSafeZ(1.0)
2019-08-15 15:00:44.396 ReferenceActuator DEBUG: NozzleDragN2_double.moveToSafeZ(1.0)
2019-08-15 15:00:44.396 ReferenceCamera DEBUG: moveTo((206.532389, 130.487244, NaN, 0.144356 mm), 1.0)
2019-08-15 15:00:44.396 GcodeDriver DEBUG: sendCommand(G0 X207.1990 Y130.2872   F60000, 10000)...
2019-08-15 15:00:44.397 GcodeDriver DEBUG: sendCommand(serial://COM11 G0 X207.1990 Y130.2872   F60000, 10000) => [ok]
2019-08-15 15:00:44.397 GcodeDriver DEBUG: sendCommand(G1 X207.2990 Y130.4872   F12000, 10000)...
2019-08-15 15:00:44.398 GcodeDriver DEBUG: sendCommand(serial://COM11 G1 X207.2990 Y130.4872   F12000, 10000) => [ok]
2019-08-15 15:00:44.399 ReferenceCamera DEBUG: moveTo((206.532389, 130.487244, -20.000000, 0.144356 mm), 1.0)
2019-08-15 15:00:44.399 ReferenceHead DEBUG: H1.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceNozzle DEBUG: N1.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceNozzle DEBUG: N2.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceCamera DEBUG: Down Camera.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceActuator DEBUG: NozzleDragN1.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceActuator DEBUG: Vac1Sense.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceActuator DEBUG: Vac2Sense.moveToSafeZ(1.0)
2019-08-15 15:00:44.399 ReferenceActuator DEBUG: FeederTape.moveToSafeZ(1.0)
2019-08-15 15:00:44.400 ReferenceActuator DEBUG: NozzleDragN2.moveToSafeZ(1.0)
2019-08-15 15:00:44.400 ReferenceActuator DEBUG: FeederTape_long.moveToSafeZ(1.0)
2019-08-15 15:00:44.400 ReferenceActuator DEBUG: NozzleDragN2_double.moveToSafeZ(1.0)
2019-08-15 15:00:44.400 ReferenceCamera DEBUG: moveTo((206.532389, 130.487244, NaN, 0.144356 mm), 1.0)
2019-08-15 15:00:44.400 ReferenceCamera DEBUG: moveTo((206.532389, 130.487244, -20.000000, 0.144356 mm), 1.0)
2019-08-15 15:00:44.408 ReferenceActuator DEBUG: UpCamLights.actuate(false)
2019-08-15 15:00:44.409 GcodeDriver DEBUG: sendCommand(M813, 10000)...
2019-08-15 15:00:44.662 GcodeDriver DEBUG: sendCommand(serial://COM11 M813, 10000) => [ok]
2019-08-15 15:00:44.663 ReferenceActuator DEBUG: DownCamLights.actuate(true)
2019-08-15 15:00:44.663 GcodeDriver DEBUG: sendCommand(M810, 10000)...
2019-08-15 15:00:44.664 GcodeDriver DEBUG: sendCommand(serial://COM11 M810, 10000) => [ok]
2019-08-15 15:00:44.829 ReferenceActuator DEBUG: UpCamLights.actuate(false)
2019-08-15 15:00:44.829 GcodeDriver DEBUG: sendCommand(M813, 10000)...
2019-08-15 15:00:44.830 GcodeDriver DEBUG: sendCommand(serial://COM11 M813, 10000) => [ok]
2019-08-15 15:00:44.830 ReferenceActuator DEBUG: DownCamLights.actuate(true)
2019-08-15 15:00:44.830 GcodeDriver DEBUG: sendCommand(M810, 10000)...
2019-08-15 15:00:44.831 GcodeDriver DEBUG: sendCommand(serial://COM11 M810, 10000) => [ok]
2019-08-15 15:00:44.993 ReferenceFiducialLocator DEBUG: FIDUCIAL-MICRO-FIDUCIALUFIDUCIAL located at (206.468889, 130.153869, 0.000000, 0.000000 mm)
2019-08-15 15:00:44.993 ReferenceCamera DEBUG: moveTo((206.468889, 130.153869, 0.000000, 0.000000 mm), 1.0)
2019-08-15 15:00:44.993 GcodeDriver DEBUG: sendCommand(G0 X207.1335 Y129.9539   F60000, 10000)...
2019-08-15 15:00:44.994 GcodeDriver DEBUG: sendCommand(serial://COM11 G0 X207.1335 Y129.9539   F60000, 10000) => [ok]
2019-08-15 15:00:44.994 GcodeDriver DEBUG: sendCommand(G1 X207.2335 Y130.1539   F12000, 10000)...
2019-08-15 15:00:44.996 GcodeDriver DEBUG: sendCommand(serial://COM11 G1 X207.2335 Y130.1539   F12000, 10000) => [ok]
2019-08-15 15:00:44.996 ReferenceFiducialLocator DEBUG: FIDUCIAL-MICRO-FIDUCIALUFIDUCIAL located at (206.468889, 130.153869, 0.000000, 0.000000 mm)
2019-08-15 15:00:44.996 ReferenceCamera DEBUG: moveTo((206.468889, 130.153869, 0.000000, 0.000000 mm), 1.0)
[... 180 lines cut ...]
2019-08-15 15:00:48.013 ReferenceCamera DEBUG: moveTo((159.174625, 179.891792, 0.000000, 0.000000 mm), 1.0)
2019-08-15 15:00:48.014 GcodeDriver DEBUG: sendCommand(G0 X160.1315 Y179.6918   F60000, 10000)...
2019-08-15 15:00:48.016 GcodeDriver DEBUG: sendCommand(serial://COM11 G0 X160.1315 Y179.6918   F60000, 10000) => [ok]
2019-08-15 15:00:48.017 GcodeDriver DEBUG: sendCommand(G1 X160.2315 Y179.8918   F12000, 10000)...
2019-08-15 15:00:48.020 GcodeDriver DEBUG: sendCommand(serial://COM11 G1 X160.2315 Y179.8918   F12000, 10000) => [ok]
2019-08-15 15:00:48.020 ReferenceFiducialLocator DEBUG: Found id FD1, location (7.620000, 60.960000, 0.000000, 0.000000 mm), side Top, part id FIDUCIAL-MICRO-FIDUCIALUFIDUCIAL, name 1x1 Fiducial, heightUnits Millimeters, height 0.000000, packageId (FIDUCIAL-MICRO), type Fiducial at (159.174625, 179.891792, 0.000000, 0.000000 mm)
2019-08-15 15:00:48.021 ReferenceFiducialLocator INFO: Fiducial results: scale (0.9986427763181628, 0.9986427763181628), translate (151.1857265852831, 119.06193307970813), shear (0.0062211361412815944, -0.0062211361412815944)
2019-08-15 15:00:48.021 ReferenceHead DEBUG: H1.moveToSafeZ(1.0)
2019-08-15 15:00:48.021 ReferenceNozzle DEBUG: N1.moveToSafeZ(1.0)
2019-08-15 15:00:48.022 ReferenceNozzle DEBUG: N2.moveToSafeZ(1.0)
2019-08-15 15:00:48.023 ReferenceCamera DEBUG: Down Camera.moveToSafeZ(1.0)
2019-08-15 15:00:48.023 ReferenceActuator DEBUG: NozzleDragN1.moveToSafeZ(1.0)
2019-08-15 15:00:48.024 ReferenceActuator DEBUG: Vac1Sense.moveToSafeZ(1.0)
2019-08-15 15:00:48.024 ReferenceActuator DEBUG: Vac2Sense.moveToSafeZ(1.0)
2019-08-15 15:00:48.025 ReferenceActuator DEBUG: FeederTape.moveToSafeZ(1.0)
2019-08-15 15:00:48.025 ReferenceActuator DEBUG: NozzleDragN2.moveToSafeZ(1.0)
2019-08-15 15:00:48.026 ReferenceActuator DEBUG: FeederTape_long.moveToSafeZ(1.0)
2019-08-15 15:00:48.026 ReferenceActuator DEBUG: NozzleDragN2_double.moveToSafeZ(1.0)
2019-08-15 15:00:48.027 ReferenceCamera DEBUG: moveTo((151.185727, 119.061933, NaN, -0.356925 mm), 1.0)
2019-08-15 15:00:48.027 GcodeDriver DEBUG: sendCommand(G0 X151.7852 Y118.8619   F60000, 10000)...
2019-08-15 15:00:48.030 GcodeDriver DEBUG: sendCommand(serial://COM11 G0 X151.7852 Y118.8619   F60000, 10000) => [ok]
2019-08-15 15:00:48.030 GcodeDriver DEBUG: sendCommand(G1 X151.8852 Y119.0619   F12000, 10000)...
2019-08-15 15:00:48.033 GcodeDriver DEBUG: sendCommand(serial://COM11 G1 X151.8852 Y119.0619   F12000, 10000) => [ok]
2019-08-15 15:00:48.034 ReferenceCamera DEBUG: moveTo((151.185727, 119.061933, -20.000000, -0.356925 mm), 1.0)
`

kaklik, 2022-10-04:
That seems to be an outdated issue, which is not present in recent OpenPnP versions. I think it should be closed as resolved.

tonyluken, 2023-08-02:
@CircuitSetup,

Can we close this issue as I believe it was address by [PR1507](https://github.com/openpnp/openpnp/pull/1578)?

tonyluken, 2023-08-19:
Since I haven't heard from @CircuitSetup, I'm closing this issue as resolved.

closed 2023-08-19 by tonyluken