openpnp/openpnp #908: Feature Request: Do not allow ReferenceStripFeeder to go out of bounds

CircuitSetup, 2019-10-21:
# Feature Request
## Describe the Feature
Lately I've had a few instances where ReferenceStripFeeders have been causing issues. If they run out of parts, or enough parts are skipped over, and the feed count increments enough, this causes the head to try to pick a part outside of the bed, and for me, turn off the machine when an endstop is hit. Preventing this from happening should be a standard feature.

## How Will It Look?
I think the simplest way to do this would be to add a Tape Length or Part Count field to the Tape Settings. Once the Tape Length or Part Count limit is reached, the Feed Count is not allowed to increment, and the feeder is either disabled or throws an error, perhaps based on the Pick Retry Count

## Why Do We Need It?
This will prevent potential damage to a machine, and keep jobs running. If a machine uses ReferenceStripFeeder at all, this will be useful.

gxurma, 2019-11-28:
Oh yeah. I need this too!

CircuitSetup, 2019-11-28:
> Oh yeah. I need this too!

The new soft limits helps with three first partIDF this, but I think a count for parts on the strip feeder would be useful to prevent the job from stopping and/or go to the next feeder with that part available automatically.

vonnieda, 2019-11-28:
This is a good idea, but also with the recent vision fixes you could try
re-enabling vision. The vision does detect the end of the strip.

Jason

CircuitSetup, 2019-11-28:
@vonnieda that may be part of my problem since I have strips going off of the end of the strip feeder holder in most cases, so it's a bit easier to 'reload'.

tyllurius, 2020-02-04:
> > Oh yeah. I need this too!
> 
> The new soft limits helps with the first part of this, but I think a count for parts on the strip feeder would be useful to prevent the job from stopping and/or go to the next feeder with that part available automatically.

Yep, would be a nice feature, and it would be even nicer if it's used to jump to the next available strip feeder with same parts when the currently used strip is empty.

vonnieda, 2020-02-04:
For what it’s worth, if you use vision it already does jump to the next
available feeder.

duncanellison, 2020-02-28:
Understand that vision is helpful here, but a hard stop on running out of components on the strip like the trayfeeder would be nice.

markmaker, 2020-04-17:
Ok, I'm biased, but why not just use the BlindsFeeder instead ;-)
https://makr.zone/new-openpnp-blindsfeeder/353/
[image]

_Mark

CircuitSetup, 2020-04-18:
Thanks @markmaker. I was considering printing your design to replace what I have and just using those. I'm having some issues with tape holes getting detected anyway (mostly because I have black plastic tape on black strip feeders).

CircuitSetup, 2020-05-22:
Closing this since it's now handled by the soft limits

closed 2020-05-22 by CircuitSetup