openpnp/openpnp #912: Wrong visualization binding algorithm.

bacrua, 2019-11-03:
# Invalid visual snap algorithm
## The exact location of the board fails due to the incorrect binding algorithm. The location of the board is determined by the position of the camera and the component over which the camera is located. For calculations, two components are used, the coordinates of the camera and the coordinates of the placement of the component on the board. In this case, the rotation of the nozzles should not affect the position of the chamber. The rotation of the nozzles around the axis should not offset the crosshair. Otherwise, the position of the nozzle forces you to move the head to set the crosshair in the center of the selected component. If the nozzle is recalibrated, then if the position of the board is not changed, the location of the board must be adjusted again.

## Playback actions
1. Launch OpenPnP
2. Press the power button.
3. Install the circuit board.
4. Calibrate nozzle (select nozzle with maximum runout)
5. Move the head so that the crosshair in the camera window is in the center of any component on the board.
6. Set the change value to 10
7. Rotate the nozzle around the axis.

## Expected Result
It was expected that the nozzle will rotate, the crosshair in the camera window will rotate in the wake of the rotation of the nozzle. But the bias will not occur. Axial runout is expected only in displacement calculations to ensure accuracy.

## Actual result
Rotating the nozzle, the crosshair moves from the center of the selected component. To return the crosshair back to the center, you need to move the head. The camera position will be different. This is important when positioning and specifying the location of the board. After recalibration, the board is shifted and the position of the board must be re-specified for accuracy, although the board was not physically moved.

markmaker, 2019-11-04:
@bacrua 

Note that when you are using the machine controls, all commands are directed at the selected tool, not to the selected camera. Remember that you can also display **all** camera views at once, so OpenPNP cannot in general know at which you are currently looking. 

[image]

If you change the selected tool to the camera and then rotate, what happens?

_Mark

markmaker, 2019-11-04:
@vonnieda 

Thinking about this one a bit further, have you ever contemplated to switch the currently selected tool automatically over to the positioning tool when using one of the "Position" buttons?

[image]

When I developed/tested calibration I had that same issue all the time, having to (remember to) switch the selected tool over manually again and again. Before I figured this out, I kept being frustrated by thinking (mistakenly) that I broke something in the code. 

_Mark

bacrua, 2019-11-04:
Yes indeed, if you point out the "camera: DownLooking H1" and adjusting the placement of the board everything works fine. I made this mistake while indicating the StripFeeder.
What are my difficulties in using:
1. I did not know this nuance and did not even guess that to adjust the position of the board you need to select a camera.
2. It is not convenient, you always need to install it and do not forget.
I still think that the choice of nozzle and its position should not affect the visualization. I am in the place where the crosshairs of the camera are looking. The camera has the same coordinates regardless of how installed and which nozzles are selected. And displacements and anticipations are needed exactly at the time when the nozzles are turned on. But when the nozzle is in operation, there is no difference where the camera is looking.
As a result, it seems to me more logical to get the absolute coordinates of the camera in all cases when the main tool is the camera function. And shift to calibration values ​​at a time when the function is touching the nozzle.
-----------
Anyway, this is a cool program and it works. For me, this is an unreachable level. You have energy and prosperity.
--------
There are some suggestions for paneling:
1. Indicate the number of repeating boards per row and columns.
2. Do not indicate the distance between the boards in the panel. Use component binding instead. For example: "R1" left bottom board, same "R1" right bottom board, same "R1" left top board. On the diagonal, component "R2" is lower left and component "R2" is for upper right. And computations.
----
And thanks again for your work.

bacrua, 2019-11-04:
[image]

closed 2019-11-04 by bacrua

markmaker, 2019-11-04:
> I still think that the choice of nozzle and its position should not affect the visualization. 

If you think this through carefully, you might notice that what you ask is impossible. :-)

If the nozzle tip has runout, the offset between camera and nozzle tip is _variable_. When you rotate the nozzle tip, the camera view will always wobble, as it is mechanically connected. 

The only remedy that I can think of, would be to **pan** the camera view slightly depending on what tool is currently selected (and the runout compensation applied). That in turn would introduce all sorts of new effects, like your head jumping when you switch the selected tool. I'm sure people would hate that even more.  

I think what I proposed - to automatically switch the selected tool with the position buttons - is not so bad. Of yourse it would have to include all the special purpose positioning buttons on the job panel and on feeders etc.

Come to think of it, clicking or dragging on the camera views should also select the camera. 

Most of the time the right tool would automatically be selected. I can't think of a plausible scenario where this does not result in the right tool being selected. Can you?

**To make it foolproof, the cross hairs could be switched off unless the camera is the selected tool.** 

_Mark

vonnieda, 2019-11-04:
I think this all ties in with something I mentioned a while back, which is that I'd like to remove the concept of selected tool (almost) entirely. It would only be used during machine setup, and not during job setup or production. The user's interface to the machine should be via the camera, and they should not have to think about the tools at all - with the expectation that if the tools are configured correctly they will go where the camera is pointing when needed.

> Thinking about this one a bit further, have you ever contemplated switching the currently selected tool automatically over to the positioning tool when using one of the "Position" buttons?

I'm not 100% sure what you mean here. Do you mean something like "If the user is interacting with N1 for calibration, for instance, then N1 becomes selected."? If so, yes, I think that is how it should work with the caveat that rather than "selecting" the tool globally, the wizard just interacts directly with the tool being configured rather than using the global selection.

There are small amounts of this already. When you configure a camera, OpenPnP will try to select that camera automatically.

The ability to select a tool was mostly added so that you could see the DRO coordinates for the tool, and to help with capturing offsets. By making the offset wizards do the work for you we really don't need tool selection at all.

Jason

bacrua, 2019-11-04:
I apologize. I live in Ukraine. My English is so weak that I use a translator. Therefore, difficulties in explanation will arise.
There are several stages to the whole process.
1. Initial setup of the entire machine. Everything is good here.
2. Operation of the machine. Also has several stages.
2.1 setting interchangeable nozzles. Everything is correct here too.
2.2 setting up the flow of components. Manual adjustment of component location in !! tape section. It’s about setting the location visually. There should be an absolute crosshair position of the camera. Beats, tool displacements should not affect. (Positioning with a nozzle is not considered)
2.3 setting the location of the board. It’s about setting the location visually. There should be an absolute crosshair position of the camera. Beats, tool displacements should not affect. (Positioning with a nozzle is not considered)
2.4 installation of components. Precision in mechanics is important here. This is where all displacements, beats are important, everything that affects the accuracy of the installation. Here you need to calculate the relative position. Interchangeable tools will only change the relative position of the component.

In this way. Turning on, off, parking, going to the home position and recalibrating the nozzles will not result in the naming of the visual position and absolute position. (after all, nothing physically moved), but what has changed is a new nozzle or a new nozzle position. If correctly calibrated, the relative position will be accurately positioned.
Working with visual settings through the camera (it is not physically interchangeable) absolute coordinates (interchangeable elements should not affect), interchangeable and calibrating calculate relative, the arrangement according to relative values.

markmaker, 2019-11-05:
>I think this all ties in with something I mentioned a while back, which is that I'd like to remove the concept of selected tool (almost) entirely. 

Yes, I remember. There remained some use cases where the "almost" in that sentence is essential (e.g. bare nozzle calibration). 

> The user's interface to the machine should be via the camera, and they should not have to think about the tools at all.

The problem starts with the possibility to show multiple cameras on the GUI. Unless you have eye tracking, OpenPNP would not know _which_ camera. ;-)

So a telling feature would be to hide the crosshairs unless a camera is the selected tool. Clicking (or dragging) on a camera would select it. 

But some things are done without the camera. If you wanted to capture the Z height of a feeder, you might want to use one specific nozzle with the right tip loaded. But OpenPNP cannot guess what you're about to do, before you actually press that capture button. So it would not know which nozzle to use to jog to the feeder.  

So I still think the selected tool should stay but with a smart auto-selection and the crosshairs' status feedback. 
 
_Mark