openpnp/openpnp #913: Nozzle tip Toolchanger blank third position line and not shown in machine.xml

gxurma, 2019-11-11:
# Problem Report
## Describe the Issue
When adding new nozzle tip, on the Toolchanger page it show a blank line for Third location.
Also saving entered data does not work. button never greys out when hitting Apply

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start fresh OpenPnP
2. add new Nozzle Tip
3. go to page Toolchanger
4. try to enter and apply plausible data. It will not work.
5. change to the other Nozzle tip, and back to the new tip.
6. there will be no change in the list, blank line remains.

## Expected Result
entered data shall get accepted and written to the machine.xml

## Actual Result
entered data does not  get accepted and written to the machine.xml

## fault mitigation:
manually edit machine.xml with plausible data

gxurma, 2019-11-11:
[image]

gxurma, 2019-11-11:
Tried to solve this by modifying the initialization constructor in the 
src\main\java\org\openpnp\machine\reference\ReferenceNozzleTip.java
from
private Location changerMidLocation2;
to
private Location changerMidLocation2;  new Location(LengthUnit.Millimeters);

But that breaks the tests. 
I dont yet understand the testing system.

vonnieda, 2019-11-21:
Fixed in #915. Thanks @gxurma!

closed 2019-11-21 by vonnieda