openpnp/openpnp #916: jetson nano

wireless-road, 2019-11-16:
hello!
Trying to run openpnp on jetson nano gives me error:
`Caused by: java.lang.UnsupportedOperationException: Architecture "aarch64" is not supported.`
Does it mean that there is no way to run OpenPnp on Jetson Nano?
Thank you.

vonnieda, 2019-11-17:
This happens because we don't have an OpenCV binary for aarch64. You can try installing OpenCV 3.4.2 manually, make sure it's in your dynamic library path, and OpenPnP should find it and start.

K2AAE, 2019-12-29:
OpenCV comes pre-installed with LT4, some have replaced 3.xx with 4.xx but the default currently is 3.xx - When I fix my SD card reader so I can create new image I'll check it out.

solhuebner, 2020-02-01:
> OpenCV comes pre-installed with LT4, some have replaced 3.xx with 4.xx but the default currently is 3.xx - When I fix my SD card reader so I can create new image I'll check it out.

Could you fix your SD card reader?

brainstorm, 2020-03-15:
I'm currently facing similar issues but getting closer to resolution (I think), my platform is a Raspberry Pi3, see:

https://github.com/brainstorm/openpnp-balena

@vonnieda Will the upstream OpenCV 4.2.0 code work with OpenPNP or we still have to rely on this fork?:

https://github.com/openpnp/opencv

I've found issues with [`libjpeg.8` vs `libjpeg.62`](https://github.com/brainstorm/openpnp-balena/commit/dc425b4aeb12553d333fe5f70cccb07e0c17174c) when using [this docker container with OpenCV support in it](https://github.com/brainstorm/openpnp-balena/commit/ae40fd871d9b8db85130842d133bf3f120632898#diff-3254677a7917c6c01f55212f86c57fbfR2)... I am now testing with the pre-packaged Debian OpenCV, but it does not seem to be willing to work for `aarch64`/`arm64`, so I guess I'll have to fallback to local compilation instead as I was toying with in here:

https://github.com/brainstorm/openpnp-balena/commit/1fa732802e521e4e729a37742bccbc436ccf5ace

Perhaps it's as easy as patching OpenCV up as @vonnieda was doing here?:

https://github.com/openpnp/opencv/commit/5ccd9e716bb76f870e1197324b2fafbe024a5520#diff-cc2450ddb977c98e67b0eaa70c4aaa92R65

I'll try that next...

brainstorm, 2020-03-15:
Oh, nice, @vonnieda seems to have fixed it up here:

https://github.com/openpnp/opencv/commit/6ce3db900df9283f09ae1c0f2916438d5c65c257

I'm running on 64-bit rpi3, hope that works, I'll try to use that opencv fork then instead of upstream...

brainstorm, 2020-03-15:
@vonnieda, you mentioned:

> This happens because we don't have an OpenCV binary for aarch64. You can try installing OpenCV 3.4.2 manually, make sure it's in your dynamic library path, and OpenPnP should find it and start.

So I went with [this hack](https://github.com/CCHS-Melbourne/openpnp-balena/commit/d4a95264bd3ee084f5add1a7cf8499ae8c05cfc8) and got different results than the simple `aarch64` error:

```
 Caused by: java.lang.UnsatisfiedLinkError: /usr/src/app/libopencv_java342.so: /usr/src/app/libopencv_java342.so: wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mismatch)
```

So this simply means that the ARMv8 library you are including in OpenPnP-OpenCV fork is 32-bit only, I need the 64bit one... I'm getting closer :)

brainstorm, 2020-05-07:
Folks, see https://github.com/CCHS-Melbourne/openpnp-kiosk for a working proof of concept ;)

I would appreciate those having Jetson Nanos to test it out a bit you you can (I just have a RPi3 available at the moment).

G-Pereira, 2021-02-10:
> This happens because we don't have an OpenCV binary for aarch64. You can try installing OpenCV 3.4.2 manually, make sure it's in your dynamic library path, and OpenPnP should find it and start.

Hey, I am attempting to run on a Jetson Nano. How can I tell if it is on my dynamic library path (the correct version)?

brainstorm, 2021-02-10:
@G-Pereira You can mention this same issue in several other repositories, but the answer is still the same: until CI is not fixed, it's not practical to pursue this further since it's a lot of manual work for every release and it does need to be automated, see attempts missing windows here: https://github.com/openpnp/opencv/pull/59 and https://github.com/openpnp/opencv/pull/56 ... 

Recently @ohbus [had plans to start afresh](https://github.com/openpnp/opencv/pull/59#issuecomment-770192055). Have some patience ;)

brainstorm, 2022-04-16:
See https://github.com/openpnp/openpnp-capture/pull/47#issuecomment-1100576011 for progress updates ;)