openpnp/openpnp #917: trying to run demo job gives non informative error

wireless-road, 2019-11-16:
Hello.
Going step by step as explained here:
https://github.com/openpnp/openpnp/wiki/Quick-Start
1. install Openpnp for windows 64 bit.
2. launch Openpnp.
3. load `pnp-test.job.xml` demo job
4. Enable machine.
6. Get this error on pressing to `Run ` button
[image]

vonnieda, 2019-11-16:
>    5. Get this error on pressing to Run button

Try copying the pnp-test folder to your desktop and load it from there.
This is an issue where Windows does not allow writes to the install
directory. Thanks for reminding me - this needs to get fixed.

Jason

K2AAE, 2019-12-02:
Installed today, 2/12/2019, samples directory still located in protected C:\Program Files\openpnp\sample, the quick start guide still instructs users to look in the install directory. Also need to have the application expand to full screen when it starts up, when a job is loaded it doesn't appear due to the collapsed state of the right side area. Google Groups notes this problem in May. Half hour of my life I won't get back .....

vonnieda, 2020-02-01:
Duplicate of #836

closed 2020-02-01 by vonnieda