openpnp/openpnp #918: Incorrect bottom-side placement locations after Nov. 6th 2019 update

maholli, 2019-11-17:
# Problem Report
## Describe the Issue
Bottom-side placement locations are incorrect when importing from KiCad. Setting board width/length also yields start results. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Add new board (width/length = 0.000)
2. Import KiCad .pos for top and bottom (create missing parts)
3. Set board side to TOP
4. Set REF1***,REF2***,REF3*** as fiducials 
5. Physically place bottom left corner of PCB at board home location 
6. Manually perform fiduial check through Boards menu
7. ✔ Note that fiducials are correctly located (albeit XYZ & Rot. don't update for the board)
   * note camera is able to correctly position itself over a random placement 
8. Set board side to BOTTOM
9. Set REF1***,REF2***,REF3*** as fiducials
10. Physically flip the PCB about the Y axis (illustrated below)
11. Manually perform fiducial check through Boards menu
   * note that camera cross-hairs originally center ~5 - 10 mm off-target of the fiducials, but vision corrects this and centers the cross-hairs on the fiducial. 
12. ❌ fiducial check fails after two fiducials after looking in the wrong location
13. Physically rotate board to put the 3 bottom-side fiducials where tool looked for them in #12 above
14. Perform manual fiducial check again
15. fiducial check succeeds
16. request camera navigate to random placement 
17. ❌ placement position is entirely incorrect 
   * note no combination of 90deg rotations is able to remedy this
18. Alter board width/height to reflect actual values (width=100mm, height=100mm)
19. Perform manual fiducial check again
20. Tool looks for fiducial at ~2X where it should be (fails)

## Expected Result
Prior to the 2019-11-06 version, a board with width/height=0 was able to correctly pass bottom-side fiducial checks and camera was able to correctly position over placements. Actual width/height values have never worked for me. 

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

KiCad screenshot showing top-side fiducials and board origin (bottom left of PCB)
[image]

KiCad screenshot showing bottom-side fiducials and board origin (bottom left of PCB) as perceived in KiCad with the "flip-board" option. This reflects what the actual board looks like. 
[image]

maholli, 2019-11-23:
Work around for others with the issue (can't be just me.. right?) is to feed OpenPnp a modified bottom-side position file and treat it like the a top-side position file.

Basic steps:

1. With your layout open in KiCad, go to `View` -> `Flip Board`
2. Go to `Place` -> `Drill and Place Offset`
3. Use your cursor to select the bottom left corner of the board as viewed from the bottom (shown below).
4. Export the .pos file (top and bottom separately) 
5. Open the bottom-side .pos file in your text editor of choice
6. Change all instances of `bottom` to `top` and save
7. Import your new bottom-side .pos file into OpenPnP **as the topside .pos file.** 
8. Set board origin in OpenPnP as the bottom left corner (as viewed from the bottom side)
9. Proceed with fiducial alignment as usual. 

Here's what my board from the initial issue post looks like given the "modified" treatment described above:
[image]

tonyluken, 2023-08-25:
@maholli,

Have you tried this with the latest test branch of OpenPnP? This should have been fixed with [PR#1507](https://github.com/openpnp/openpnp/pull/1507).

Tony

maholli, 2023-08-25:
@tonyluken yes I can confirm this has been fixed for some time as I have been using kicad .pos files without needing any workarounds for about a year now.

Thanks all!

closed 2023-08-25 by maholli