openpnp/openpnp #928: Vacuum Sensing cannot be read

lei111, 2019-12-03:
I connected the vacuum sensor to the T0 port of marlin. I try to use  T: (? <Value>-? \ D +)  to read the temperature value of MARLIN firmware as a vacuum value。
Click (read) but does not work

2019-12-03 10:57:59.367 GcodeDriver DEBUG: sendCommand(M105, 5000)...
2019-12-03 10:57:59.368 GcodeDriver TRACE: [serial://COM8] >> M105
2019-12-03 10:57:59.664 GcodeDriver TRACE: [serial://COM8] << ok T:11.04 /0.00 (997.75) T0:11.04 /0.00 (997.75) T1:11.04 /0.00 (997.75) @:0 @0:0 @1:0
2019-12-03 10:57:59.664 GcodeDriver DEBUG: sendCommand(serial://COM8 M105, 5000) => [ok T:11.04 /0.00 (997.75) T0:11.04 /0.00 (997.75) T1:11.04 /0.00 (997.75) @:0 @0:0 @1:0]
2019-12-03 10:57:59.665 ReferenceActuator DEBUG: N1VAC.read(): null

lei111, 2019-12-03:
The sensor I use will output a voltage change of 0v-5v. It should be possible to use this port to get the vacuum value

vonnieda, 2019-12-03:
Change the regex to exactly `ok T:(?<Value>-?\d+.\d+).*`

lei111, 2019-12-03:
Thank you very much for your help, the first port can be extracted
Excuse me, how to write can extract the data of the second, which is T0.
thank you very much

vonnieda, 2019-12-03:
Try `ok.*T0:(?<Value>-?\d+.\d+).*`

lei111, 2019-12-03:
Can work, thank you

vonnieda, 2019-12-03:
You're welcome. For reference, I like to use regexr.com/4pvqj to test regexes. If you click the Explain tab it shows exactly how it works. Very helpful for debugging complex regexes.

closed 2019-12-03 by vonnieda