openpnp/openpnp #933: N2 nozzle cannot control rotation and up and down movement

lei111, 2019-12-17:
Thank you very much for your work
I encountered another problem, JOB page, select N2 can not control rotation and Z movement.
I have configured GcodeDriver: Axis Mapping
`<axis name="z" type="Z" home-coordinate="0.0">
		   <head-mountable-ids class="java.util.HashSet">
			  <string>N1</string>
			  <string>N2</string>
		   </head-mountable-ids>
		   <transform class="org.openpnp.machine.reference.driver.GcodeDriver$NegatingTransform">
			  <negated-head-mountable-id>N2</negated-head-mountable-id>
		   </transform>
		</axis>
		<axis name="c1" type="Rotation" home-coordinate="0.0">
		   <head-mountable-ids class="java.util.HashSet">
			  <string>N1</string>
		   </head-mountable-ids>
		   <pre-move-command><![CDATA[
		   T0S1
		   G92E{Coordinate:%.4f}
		   ]]></pre-move-command>
		</axis>
		<axis name="c2" type="Rotation" home-coordinate="0.0">
		   <head-mountable-ids class="java.util.HashSet">
			  <string>N2</string>
		   </head-mountable-ids>
		   <pre-move-command><![CDATA[
		   T1S1
		   G92E{Coordinate:%.4f}
		   ]]></pre-move-command>
		</axis>
	 </axes>
  </driver>`
`2019-12-17 14:01:58.976 ReferenceNozzle DEBUG: N1.moveTo((6.000000, 8.000000, -1.000000, 0.000000 mm), 0.89)
2019-12-17 14:01:58.978 GcodeDriver DEBUG: sendCommand(G1   Z1.0000  F2670, 5000)...
2019-12-17 14:01:58.998 GcodeDriver DEBUG: sendCommand(serial://COM8 G1   Z1.0000  F2670, 5000) => [ok]
2019-12-17 14:01:58.998 GcodeDriver DEBUG: sendCommand(M400 ; Wait for moves to complete before returning, 5000)...
2019-12-17 14:01:59.078 GcodeDriver DEBUG: sendCommand(serial://COM8 M400 ; Wait for moves to complete before returning, 5000) => [ok]
2019-12-17 14:02:02.828 ReferenceNozzle DEBUG: N1.moveTo((6.000000, 8.000000, -1.000000, -1.000000 mm), 0.89)
2019-12-17 14:02:02.830 GcodeDriver DEBUG: sendCommand(T0S1, 5000)...
2019-12-17 14:02:02.926 GcodeDriver DEBUG: sendCommand(serial://COM8 T0S1, 5000) => [echo:Active Extruder: 0, ok]
2019-12-17 14:02:02.927 GcodeDriver DEBUG: sendCommand(G92E0.0000, 5000)...
2019-12-17 14:02:03.137 GcodeDriver DEBUG: sendCommand(serial://COM8 G92E0.0000, 5000) => [X:380.00 Y:490.00 Z:1.00 E:0.00 Count X:30400 Y:49000 Z:80, ok]
2019-12-17 14:02:03.138 GcodeDriver DEBUG: sendCommand(G1    E-1.0000 F2670, 5000)...
2019-12-17 14:02:03.159 GcodeDriver DEBUG: sendCommand(serial://COM8 G1    E-1.0000 F2670, 5000) => [ok]
2019-12-17 14:02:03.160 GcodeDriver DEBUG: sendCommand(M400 ; Wait for moves to complete before returning, 5000)...
2019-12-17 14:02:03.187 GcodeDriver DEBUG: sendCommand(serial://COM8 M400 ; Wait for moves to complete before returning, 5000) => [ok]
2019-12-17 14:02:06.932 ReferenceNozzle DEBUG: N2.moveTo((24.881893, 17.468509, -1.000000, 0.000000 mm), 0.89)
2019-12-17 14:02:09.180 ReferenceNozzle DEBUG: N2.moveTo((24.881893, 17.468509, -1.000000, 0.000000 mm), 0.89)
2019-12-17 14:02:10.452 ReferenceNozzle DEBUG: N2.moveTo((24.881893, 17.468509, 0.000000, -1.000000 mm), 0.89)
2019-12-17 14:02:11.540 ReferenceNozzle DEBUG: N2.moveTo((24.881893, 17.468509, 0.000000, -1.000000 mm), 0.89)
`

lei111, 2019-12-17:
I am using marlin, hope someone knows how to solve it, thank you very much

vonnieda, 2019-12-17:
Please post help requests on the mailing list: http://groups.google.com/group/openpnp

closed 2019-12-17 by vonnieda