openpnp/openpnp #944: Feature Request: ScalingTransform for axis

riaancillie, 2020-01-29:
# Feature Request
## Describe the Feature
An axis transform (similar to the existing NegatingTransform and CamTransform), that will simply scale the axis value a user sees into a transformed value for the motion controller using a simple multiplication factor. 

## How Will It Look?
The transform will have only one property, "scale-factor", which is a multiplier from the transformed value to the raw value.

`<transform class="org.openpnp.machine.reference.driver.GcodeDriver$ScalingTransform" scale-factor="0.1"/>`

## Why Do We Need It?
Short version: to compensate for the difference between maximum speeds of various axes and allow the speed slider to uniformly adjust all axes.

Longer version: Using a rotation axis as an example, OpenPnp will send the required number of degrees as millimeters to the motion controller. E.g. 270 degrees rotation will ask the motion controller to move the axis to 270mm. For an X- or Y-axis, a 270mm move means many rotations possibly. For a nozzle rotation axis, this is simply a 270 degree turn. A rotation axis is almost always going to be able to be faster than a linear axis. 
In GRBL's configuration, I have to a use a very high max feed rate for my rotation axes - multiple times the max speed of my X and Y axes. 
E.g. Max-X speed = 30 000mm/min and Max-A speed = 400 000mm/min

Because of the difference in speed, I have to pick which maximum speed to use as GcodeDriver's "Max Feed Rate". Using 30 000, my rotations are painfully slow. Using 400 000, the speed slider only affects the X,Y and Z when the slider is almost at 0%. 

Currently, I set my GcodeDriver's max feed rate to 30 000mm/min and then simply perform the rotation as a seperate gcode line using G0 and omitting the feedrate. This way, at least the Speed slider works on my linear axes.

Ideally, I would like to be able to transform the rotation axis by a factor of 10, e.g. 270 degrees becomes 27.0mm sent to the motion controller. This way, I can match the maximum speeds in GRBL's configuration for all axes. By doing so, I can still get the maximum speed out of all movements and have the slider work on all axes.

vonnieda, 2020-01-29:
Good idea! This might mean we need to support multiple transforms per axis
in case the axis already has one.

riaancillie, 2020-01-29:
Thank you @vonnieda . I suppose when supporting multiple transforms, the order of the transforms are going to be an issue?

vonnieda, 2020-01-29:
Yep, but we can just use an ordered list. That can be done later - no
reason not to add the scale transform now to fix your issue.

vonnieda, 2020-04-25:
Fixed by #994

closed 2020-04-25 by vonnieda