openpnp/openpnp #949: Panelization: Miscalculated Placements

tyllurius, 2020-02-03:
# Problem Report
## Describe the Issue
The X position of parts from some panelized boards for the job are calculated wrongly. For the GUI the positions are calculated correctly (means for example the camera moves to all parts of all panelized boards correctly).

See [thread](https://groups.google.com/forum/?pli=1#!topic/openpnp/Bss8ciLELgE) for discussion.

## Steps to Reproduce
1. Take a board, bottom side
2. Panelize it
4. Start the Job
5. The parts of the first board are misplaced (see picture), part of the other boards are placed correctly.

## Expected Result
Part position with correctly applied x and y offset per board.

## Actual Result
X position of some boards is wrong.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
[image]

tyllurius, 2023-08-03:
@tonyluken 
I will test next week and report back. Thanks for addressing this issue!

tonyluken, 2023-08-19:
@tyllurius,

Any chance to test this?

tonyluken, 2023-09-09:
Since I haven't heard anything to the contrary, I'm going to close this issue. You can always reopen it or start a new one if you are still having problems.

closed 2023-09-09 by tonyluken