openpnp/openpnp #950: OpenPnP on Raspberry Pi 4

dronecz, 2020-02-06:
I was able to setup OpenPnP on RPi 4 with 4 GB RAM. I´m following setup steps from wiki and i can not setup any camera. I know that ARM platform is not supported, but I was hopping that somebody will at least point me some solution. 

# Read This First
To submit an issue for OpenPnP you must use this template or it will be deleted. Please fill out either the Problem Report or Feature Request section below and delete the other section.

-----------------------------------------------------------------------

# Problem Report
I´m unable to setup any camera. 

## Steps to Reproduce
How can a developer reproduce the issue you are experiencing? Use a step by step format like this:
1. Start OpenPnP
2. Click the Power On button.
3. Jog the machine 10mm to the right.
4. The machine fails to jog and shows an error.

## Expected Result
What did you expect to happen? Be descriptive.

## Actual Result
What actually happened? Be descriptive.

## Notes, Log Files, Screen Captures, Videos, etc. to Show the Issue
The more information you can provide about the issue the more likely it is that a developer will be able to respond.

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-----------------------------------------------------------------------

# Feature Request
## Describe the Feature
What do you want it to do? Be as descriptive as possible.

## How Will It Look?
If the feature will include changes to the user interface, how would they look? Include drawings or mockups if possible.

## Why Do We Need It?
1. What problem does it solve?
2. Is it useful for everyone, or does it only solve a problem for your machine?

## Provide Examples
Is this something that exists in other pick and place software? Provide examples and references if possible.

vonnieda, 2020-02-06:
You can try the OpenCvCamera or WebcamCaptureCamera. The
OpenPnpCaptureCamera has not been ported to ARM.

Jason

dronecz, 2020-02-06:
> You can try the OpenCvCamera or WebcamCaptureCamera. 

I tried to add OpenCVCamera and I can use RPI camera module at slot 0 and USB camera at slot 1 😍. I should test it more before I posted a question, but I hope it will help somebody in future. Many thanks for prompt reply.

closed 2020-02-06 by dronecz

G-Pereira, 2021-01-31:
Hi @dronecz I am thinking going this route as well. Can you share your RAM and eMMC size recommendation?

dronecz, 2021-02-10:
> Hi @dronecz I am thinking going this route as well. Can you share your RAM and eMMC size recommendation?

Hi, I have RPi 4 with 4 GB RAM and I used SD card.. 😉

olovskos, 2021-04-25:
Will OpenPnpCaptureCamera be ported to ARM? And if so, is there an estimated release plan?

olovskos, 2023-05-28:
So?

markmaker, 2023-05-28:
Have you tried?

This says it should now work on the OpenPnP `test` version...

https://github.com/openpnp/openpnp-capture/pull/47#issuecomment-1100576011

... it was integrated after a long delay:

https://github.com/openpnp/openpnp/pull/1544

But I'm only talking about USB cameras. The PI cameras may not work yet:

https://github.com/openpnp/openpnp/issues/1437

_Mark

olovskos, 2023-05-29:
Thanks for the hint, will try it. Usb camera is all I need :)

darkomenz, 2023-05-31:
Would love to see the CSI cameras working on the CM4.

https://github.com/openpnp/openpnp-capture/issues/60