openpnp/openpnp #961: Tab switches eventually freeze the GUI (AWT-EventQueue-0 null pointer exception)

psychogenic, 2020-03-05:
## Issue
While using the latest OpenPnP builds (OpenPnP-linux-develop.deb under Ubuntu, jre1.8.0_24), the system will eventually cease responding to tab switch requests and the tab pane is effectively frozen, though you can still thankfully save job/config and close the program to restart.

The terminal spews out a
`ERROR: Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException`
on the click to the tab.  It seems to have to do with the logging, attaching a log file with a full trace of the error.

I actually managed to have this happening while running a sort of REPL shell in Python.  In that instance, I forced the tab switch to the logs tab, like so:

`
t = gui.getTabs()
t.setSelectedIndex(5)
`

Which caused a storm of same _AWT-EventQueue-0_ null pointer exceptions, until I reset the selected index to 0.

## Steps to Reproduce
1. Start OpenPnP
2. Use it for a while.
3. If you switch tabs often enough, you get the freeze and uninformative error.  

Seems to happen more during a job, though maybe I just noticed it more.

## Expected Result
Being able to switch tabs at will.

## Actual Result
Eventual freeze and error.  See attached log.
[nullpointererror.txt](https://github.com/openpnp/openpnp/files/4294021/nullpointererror.txt)

vonnieda, 2020-03-05:
Thanks for the report @psychogenic. Do you know if this is working correctly on older versions by chance?

I suspect this has to do with not removing property change listeners from wizards, which is something I've actively working on.

psychogenic, 2020-03-05:
Sorry, haven't installed/tried it.

I've been using it with logging turned off for about 20-30 mins, now, without issue--maybe it's a work-around...

vonnieda, 2020-03-05:
Thanks - how many tab switches would you think to reproduce it? I'd like to be able to see if it happens on other operating systems, too.

psychogenic, 2020-03-05:
I couldn't figure it out, seemed inconsistent/hard to replicate, maybe some race condition... but it also seemed to happen a lot more during a job (when I had to go and disable some feeder because it was empty--wishlist item: if there are multiple feeders for component X and one is empty, use the other).

So I think that also points to logging--as there's lots happening/being logged.  So maybe anything that increases log output would help trigger it faster (I was set to TRACE, so very verbose).

brainstorm, 2020-03-14:
Did you try to run with a different JDK? Java AWT queues might be bugged, I reported it to Corretto (AWS JDK distribution) and they fixed it for me, see:

https://github.com/corretto/corretto-11/issues/46#issuecomment-544404829

Could be unrelated though, but worth a try, imho.

psychogenic, 2020-03-14:
I'd actually started running openpnp with openjdk stuff that was already installed--it was failing miserably, as it would crash everytime the filemanager dialogs were accessed.
So I installed the oracle jre1.8.0_241 and that worked without issue--but for this particular thing.
The problem is completely sidestepped if I turn off the logging so, so far, I've been using it this way without any other problems.

markmaker, 2020-05-14:
I can confirm this bug. I fixed it in my upcoming PR, improving it to remember the selected tab by class, so if you switch from Camera to Nozzle and back, you're back in the right Cameras tab.

Will break it out as a separate PR as soon as I find some time. 

_Mark

tonyluken, 2020-05-16:
I've seen something similar but it didn't involve switching tabs.  I was experimenting with a strip feeder and was just rapidly clicking between the Command feed and Move camera to pick location buttons.

Tony

vonnieda, 2020-05-16:
There is a known issue where wizard event handlers do not get released, but
I would expect it take thousands of tab switches to make an impact.

Jason

tonyluken, 2020-05-16:
I don't know if this is related, but I also occasionally see index out of bounds exceptions in AWT-EventQueue-0.  In the example below I was just running a muli-placement alignment to make sure it still works (it does):

`2020-05-16 15:48:24.142 GcodeDriver DEBUG: sendCommand(null, 250)...
2020-05-16 15:48:24.184 GcodeDriver TRACE: [serial://COM5] << qr:31, qi:0, qo:1
2020-05-16 15:48:24.346 GcodeDriver TRACE: [serial://COM5] << posx:177.043,posy:100.973,posz:0.000,posa:321.000,feed:8000.00,vel:0.00,unit:1,coor:1,dist:0,frmo:1,stat:3
2020-05-16 15:48:24.347 GcodeDriver TRACE: Position report: posx:177.043,posy:100.973,posz:0.000,posa:321.000,feed:8000.00,vel:0.00,unit:1,coor:1,dist:0,frmo:1,stat:3
2020-05-16 15:48:24.347 GcodeDriver TRACE: [serial://COM5] << qr:32, qi:0, qo:1
2020-05-16 15:48:24.392 SystemLogger ERROR: Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException: Index 4562 out of bounds for length 4562

2020-05-16 15:48:24.393 GcodeDriver DEBUG: sendCommand(serial://COM5 null, 250) => [qr:31, qi:0, qo:1, posx:177.043,posy:100.973,posz:0.000,posa:321.000,feed:8000.00,vel:0.00,unit:1,coor:1,dist:0,frmo:1,stat:3, qr:32, qi:0, qo:1]
2020-05-16 15:48:24.393 ReferenceCamera DEBUG: moveTo((176.824862, 100.973367, -36.000000, 321.000000 mm), 1.0)
2020-05-16 15:48:24.393 SystemLogger ERROR: 	at java.desktop/javax.swing.plaf.basic.BasicListUI.updateLayoutState(BasicListUI.java:1446)

2020-05-16 15:48:24.393 SystemLogger ERROR: 	at java.desktop/javax.swing.plaf.basic.BasicListUI.maybeUpdateLayoutState(BasicListUI.java:1389)

2020-05-16 15:48:24.393 SystemLogger ERROR: 	at java.desktop/javax.swing.plaf.basic.BasicListUI.getPreferredSize(BasicListUI.java:648)

2020-05-16 15:48:24.393 SystemLogger ERROR: 	at java.desktop/javax.swing.JComponent.getPreferredSize(JComponent.java:1680)

2020-05-16 15:48:24.393 SystemLogger ERROR: 	at java.desktop/javax.swing.ScrollPaneLayout.layoutContainer(ScrollPaneLayout.java:795)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.layout(Container.java:1537)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.doLayout(Container.java:1526)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1722)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1731)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1731)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1731)

2020-05-16 15:48:24.394 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1731)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1731)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validateTree(Container.java:1731)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.desktop/java.awt.Container.validate(Container.java:1657)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.desktop/javax.swing.RepaintManager$3.run(RepaintManager.java:745)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.desktop/javax.swing.RepaintManager$3.run(RepaintManager.java:743)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.base/java.security.AccessController.doPrivileged(Native Method)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)

2020-05-16 15:48:24.395 SystemLogger ERROR: 	at java.desktop/javax.swing.RepaintManager.validateInvalidComponents(RepaintManager.java:742)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.desktop/javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1889)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.base/java.security.AccessController.doPrivileged(Native Method)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)

2020-05-16 15:48:24.396 SystemLogger ERROR: 	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)

2020-05-16 15:48:24.397 SystemLogger ERROR: 	at org.openpnp.gui.MainFrame$19.dispatchEvent(MainFrame.java:464)

2020-05-16 15:48:24.397 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)

2020-05-16 15:48:24.397 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)

2020-05-16 15:48:24.397 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)

2020-05-16 15:48:24.397 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)

2020-05-16 15:48:24.397 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

2020-05-16 15:48:24.398 SystemLogger ERROR: 	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

2020-05-16 15:48:31.202 ReferenceHead DEBUG: H1.moveToSafeZ(1.0)
2020-05-16 15:48:31.202 ReferenceNozzle DEBUG: Nozzle1.moveToSafeZ(1.0)
`