openpnp/openpnp #966: Feature Request: YUV color conversion option

psychogenic, 2020-03-14:
Greetings,

I've been having a hard time using HSV masking and came to realize it's because my cams are outputting **YUV**.  Converting using Bgr2HsvFull gives a very nice rainbow effect, but makes _MaskHsv_ less than optimally effective.

So I made a one-line change to _java/org/openpnp/vision/FluentCv.java_ to add 
`  Yuv2Bgr(Imgproc.COLOR_YUV2BGR)`
and gave that a shot.  Now I can do

1. ConvertColor Yuv2Bgr
2. ConvertColor Bgr2HsvFull
3. MaskHsv

To much (much) better effect for bottom vision processing.

My questions:

1. I see that _org.opencv.imgproc.ImgProc_ has an abundance of COLOR_ options, probably too many to be worthwhile including, but what criteria are use to choose which ones wind up in the FluentCv ColorCode enum?
2. Is it worth submitting my one-line patch for YUV support and, while I'm at it, are there other options that would be worth including?

scott-t-wilson, 2020-03-14:
I know effectively nothing about the internals of vision processing work, but I hope you're on to something. I gave up using MaskHsv and effectively only do black and white processing. After upgrading my lighting I get 100% success, but color processing was extremely frustrating.

vonnieda, 2020-03-14:
Hi Pat, if you want to send a patch I think that’d be fine, but I wanted to
ask which OpenPnP camera type are you using? I thought we were converting
RGB in all cases.

Jason

psychogenic, 2020-03-14:
Greets, Jason.

You helped me out, recently through IRC, for getting access to the reference bottom vision scripting, and this is for the same machine--short version, it's the USB cams that came with a recent LitePlacer kit.

Am attaching screen and dump from v4l2-ctl (which make me realize I might be better off with COLOR_YUV2BGR_Y422...)

[caminfo.txt](https://github.com/openpnp/openpnp/files/4332862/caminfo.txt)
[image]

vonnieda, 2020-03-14:
Ah, interesting. An uncommon use case because most people use MJPEG due to
USB bandwidth restrictions.

Feel free to submit a patch but I will also be looking into whether it
makes sense to explicitly convert all images to RGB or BGR since OpenPnP
does make some assumptions about that.

Jason

psychogenic, 2020-03-14:
Ok, will do.  
Can it do this conversion automagically?  
I've had a issues with masking, especially when trying to do a "bandpass" where I want to mask out hues from [0,n] and then chain that with a mask from [m,255] and the filter dies with a null pointer exception.  I thought it might be because the resulting image was no longer HSV (and the system was assuming some HSV/couldn't tell the format).

vonnieda, 2020-03-14:
Yes, it would be done behind the scenes. Honestly, I thought it already was
so I need to look in to how it’s getting through :)

psychogenic, 2020-03-14:
All right, let me know if you have tests for me to run--platform is ubuntu on an i7-8550U, if that matters.  And if you have a clue as to how to enable MJPEG, I can try that too.

psychogenic, 2020-03-14:
meh, cancel that last part... `v4l2-ctl --list-formats-ext` says all I have is 'YUYV' (YUYV 4:2:2)

psychogenic, 2020-03-14:
Yeah, works nicely.  Converting to BGR assuming various formats gives interesting effects that might actually be useful. 
The one I expected to work, bottom left of the attached image, seems to gag, but then the masking is finally doing what I expect it to and is working beautifully with this 0402 cap.
[image]

psychogenic, 2020-03-14:
Actually, I'm realizing I forgot to do the conversion to HSV and it's working anyway.  Maybe this is all pointless and we don't need to convert to HSV somehow? 
Though enabling the conversion that generates the exception seemed to visually affect the image, disabling it completely still leads to the same nice masking, so maybe it's doing nothing and all I had to do was _not_ convert to HSV to get decent masking?  
Every other color conversion leads to psychedelic rainbows, and it doesn't even seem required.  a bit confused...

vonnieda, 2020-03-14:
My thought with the HSV was that it’s easier to specify a range of hues
than it is to specify a range of RGB when looking at a color in the real
world. MaskHSV isn’t tied very tightly to HSV - it just applies a mask on a
channel by channel basis, so if you find a set of mask values that work
with your default image format then there’s no reason not to do it that
way. Should make your pipeline faster too.

Jason

psychogenic, 2020-03-14:
Ok, cool.  Also, do you know the reason why chaining two MaskHsv seems inevitably to lead to a java.lang.NullPointerException ?
I often seem to want to band-pass but it never works.

vonnieda, 2020-03-14:
Can’t think of why that would happen. Send me an example pipeline that does
it?

psychogenic, 2020-03-14:
Here, trimmed this down to get rid of some fluff but it causes the exception on the last stage:

`
<cv-pipeline>
   <stages>
      <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="imgcap" enabled="true" settle-first="true" count="1"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.BlurGaussian" name="blur" enabled="true" kernel-size="9"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="circleofinterest" enabled="true" diameter="400"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskHsv" name="masklow" enabled="true" auto="false" fraction-to-mask="0.0" hue-min="0" hue-max="30" saturation-min="0" saturation-max="255" value-min="0" value-max="255" invert="false" binary-mask="false"/>
      <cv-stage class="org.openpnp.vision.pipeline.stages.MaskHsv" name="crashme" enabled="true" auto="false" fraction-to-mask="0.0" hue-min="60" hue-max="255" saturation-min="0" saturation-max="255" value-min="0" value-max="255" invert="false" binary-mask="false"/>
   </stages>
</cv-pipeline>
`

markmaker, 2020-03-14:
> conversionsandmasking

Hi

I always struggled with MaskHsv, and used trial and error to get what I
wanted.

But after this exchange I had a deeper look and I think I found one
possible source of confusion:
https://github.com/openpnp/openpnp/blob/73ad73ddb9b66eb37c240351f9276eea4ab8b617/src/main/java/org/openpnp/vision/pipeline/ui/ResultsPanel.java#L123-L131

The order of the HSB bytes is wrong, it should be Big Endian i.e. H and
B are swapped. To capture a hue, you have to go back to the image
*before conversion*, the read off the *third *value as the hue, i.e. the
one mistakenly labeled as "B".

*After the conversion *it seems R is mapping the hue, though it seems to
change a bit through the conversion.

Maybe that is due to the difference between the Java  Color.RGBtoHSB()
and the OpenCV HSV conversion model

NOTE: I first thought HSB is not the same as HSV but rather HSL.

https://en.wikipedia.org/wiki/HSL_and_HSV#Basic_principle

But it seems that is not true. HSB *is the same as HSV*:
https://www.oodlestechnologies.com/blogs/Convert-Color-Code-into-HSV-to-RGB-and-Hex-Color-Code-in-Java/

I will enclose a fix in my upcoming PR. I'll also label it "HSV" instead
of "HSB" to match the stages and conversions. That's how it'll look:

I had no problems using MaskHSV for "green screening" once I guestimated
the right hue range. It is very resilient against all kinds of
influences, shadows etc. and reliably masks just the green stuff! Much
better than any edge based solution.

I use it extensively in the BlindsFeeder
<https://makr.zone/new-openpnp-blindsfeeder/353/> and my upcoming reel
feeder.

_Mark

psychogenic, 2020-03-14:
Ok, now the pipeline I sent isn't crashing... So am attaching a full version that actually still does.
[chainedmasks.txt](https://github.com/openpnp/openpnp/files/4332980/chainedmasks.txt)

markmaker, 2020-03-14:
_[Sorry, E-Mail ate the images and won't let me edit the post]_

Hi

I always struggled with MaskHsv, and used trial and error to get what I wanted.

But after this exchange I had a deeper look and I think I found one possible source of confusion:
https://github.com/openpnp/openpnp/blob/73ad73ddb9b66eb37c240351f9276eea4ab8b617/src/main/java/org/openpnp/vision/pipeline/ui/ResultsPanel.java#L123-L131

The order of the HSB bytes is wrong, it should be Big Endian i.e. H and B are swapped. To capture a hue, you have to go back to the image before conversion, the read off the third value as the hue, i.e. the one mistakenly labeled as "B".
[image]

After the conversion it seems R is mapping the hue, though it seems to change a bit through the conversion.
[image]

Maybe that is due to the difference between the Java  Color.RGBtoHSB() and the OpenCV HSV conversion model

NOTE: I first thought HSB is not the same as HSV but rather HSL.

[image]

https://en.wikipedia.org/wiki/HSL_and_HSV#Basic_principle

But it seems that is not true. HSB is the same as HSV:
https://www.oodlestechnologies.com/blogs/Convert-Color-Code-into-HSV-to-RGB-and-Hex-Color-Code-in-Java/

I will enclose a fix in my upcoming PR. I'll also label it "HSV" instead of "HSB" to match the stages and conversions. That's how it'll look:
[image]

I had no problems using MaskHSV for "green screening" once I guestimated the right hue range. It is very resilient against all kinds of influences, shadows etc. and reliably masks just the green stuff! Much better than any edge based solution.

I use it extensively in the BlindsFeeder and my upcoming reel feeder.
[image]

_Mark

psychogenic, 2020-03-14:
Hey Mark, thanks for the info.  I've been going pretty blind, just tweaking values until what I don't want goes away but this should make things easier.

markmaker, 2020-03-14:
> My thought with the HSV was that it’s easier to specify a range of hues than it is to specify a range of RGB when looking at a color in the real world. 
>

HSV is _essential_ to capture a specific color regardless of its brightness. You simply couldn't do that with RGB. 

_Mark

psychogenic, 2020-03-14:
I've had tons of trouble because of changing amounts of daylight.  But there's something really off with the HSV setup in my case
[image]
.  Here's a queue that does nothing but 
1. Capture
2. Bgr2HsvFull
3. MaskHsv
And it's a java.lang.NullPointerException with nothing mentioned in the logging (level trace).

psychogenic, 2020-03-14:
mk, and with a restart the exception goes away.  Maybe something to do with the new stage creation/addition.

markmaker, 2020-03-14:
Hmmm. The documentation of `java.awt.Color.RGBtoHSB(int, int, int, float[])` says the HSB byte order is correct in the previous code. But it does not match experience. Very strange.

```
 /**
     * Converts the components of a color, as specified by the default RGB
     * model, to an equivalent set of values for hue, saturation, and
     * brightness that are the three components of the HSB model.
     * <p>
     * If the <code>hsbvals</code> argument is <code>null</code>, then a
     * new array is allocated to return the result. Otherwise, the method
     * returns the array <code>hsbvals</code>, with the values put into
     * that array.
     * @param     r   the red component of the color
     * @param     g   the green component of the color
     * @param     b   the blue component of the color
     * @param     hsbvals  the array used to return the
     *                     three HSB values, or <code>null</code>
     * @return    an array of three elements containing the hue, saturation,
     *                     and brightness (in that order), of the color with
     *                     the indicated red, green, and blue components.
     * @see       java.awt.Color#getRGB()
     * @see       java.awt.Color#Color(int)
     * @see       java.awt.image.ColorModel#getRGBdefault()
     * @since     JDK1.0
     */
    public static float[] RGBtoHSB(int r, int g, int b, float[] hsbvals)
```
Investigating further...
_Mark

markmaker, 2020-03-14:
No, I was definitely wrong about the HSB array order. I was tricked into believing that because the OpenCV conversion is different from the Java conversion.

- Java uses the full 0 .. 255 range to code hue.

- OpenCv uses only a 0 .. 180 range to code hue, because it is an angle after all and it scales the 360° down by 2.

See the doc:
https://docs.opencv.org/2.4/modules/imgproc/doc/miscellaneous_transformations.html#cvtcolor
How it is an angle:
[image]

This does the trick:
```
                    matStatusLabel.setText(String.format("RGB: %03d, %03d, %03d HSV: %03d, %03d, %03d XY: %d, %d %s",
                            color.getRed(),
                            color.getGreen(),
                            color.getBlue(),
                            (int) (180.0 * hsb[0]),
                            (int) (255.0 * hsb[1]),
                            (int) (255.0 * hsb[2]),
...
```

Now the 50 .. 70 range I was obtaining from experimenting with the mask is finally there:
[image]

And it is coded in the "B" channel after conversion (which is logical, because RGB is actually coded BGR in a Java BufferedImage):
[image]

Actually, I think something like 

```
                            (int) (179.999 * hsb[0]),
                            (int) (255.999 * hsb[1]),
                            (int) (255.999 * hsb[2]),
```

is more correct, because 255 should get its slice of the range (`int` cast rounding down). It does explain the difference in the conversion result (see screenshots). With the change they do match perfectly.

This will be in the PR.

_Mark

vonnieda, 2020-03-14:
Mark, I’m not able to check at the moment, but I think you will find that
the default pipelines and the docs on the wiki recommend BGR2HSV_Full,
which uses the full 8 bit space for the Hue.

Jason

markmaker, 2020-03-14:
OK. Now I feel stupid. Confirmed and changed. Same mistake in the BlindsFeeder pipeline.

_Mark

vonnieda, 2020-03-15:
It does make me think that MaskHsv would be a lot more user friendly if it checked the incoming color model and if it's not HSV_Full, convert it automatically and then back at the end. This would reduce the pipelines that use it by two stages and make it much easier for users to understand.

markmaker, 2020-03-15:
The mistake is simply all mine. I should have read the MaskHSV description more carefully (RTFM to myself).

> This stage expects the input to be in HSV_FULL format, so you should do a ConvertColor with Bgr2HsvFull before this stage

Pipelines are not for the faint-hearted. It would be very hard to change that fundamentally. And it would probably come at considerable cost in vision speed. 

_Mark

markmaker, 2021-07-02:
@psychogenic,

what is the status of this Issue?

The discussion somehow veered of course (and I'm partly to blame for that), in the end it is unclear to me whether this was just a usage problem, or a real problem. 

Also note that @tonyluken has added color conversion in the Editor, see #1098, so some issues might be obsolete.

_Mark

closed 2021-10-11 by markmaker