openpnp/openpnp #1009: Board locations using multiple component

bacrua, 2020-05-15:
It is not possible to set the location and rotation of the board using the function "placement and rotation of the board using multi components"

1. I install the board at a point with the coordinates x = 300mm Y = 100mm. with a 45 degree angle
2. Move the camera head to the board.
3. I click on the button "location and rotation of the board for multi components"
4. I select 4 components from the list of this board, which will determine the position of the board.
5. I click on "next"
6. The head leaves for the coordinates x0 y0 in the dialog text field indicates the name of the component on which you want to point the camera.

EXPECTED. As in the previous version 2, which I used before the update, that after the steps taken, the head will remain in place. I point the camera at the listed component. I point the camera to the center of the component, click "next." The camera head remains in place. The following component is indicated in the text. I point the camera to the center of the component, click "next." The head remains in place, the text with the following component is displayed in the field. The procedure is repeated until it reaches the last component from the list and after the location of the board is calculated.

1. Perform the previous paragraphs from the list.
2. After the head has moved to the position x0 y0.
3.. Move the head to the center indicated in the text dialog box of the component.
4. Click "next"
5. The head moves to the coordinate not x ??? and u ???
6. Repeat steps 3 and 4.
7. The head moves each time as to a coordinate not x ??? and u ???
8. After the last click "next", the dialog box "an error due to incorrect placement."

Expected. As in the previous version of the board definition for 2 components of the left and right corners. This bearing will only differ in the number of components for accuracy. spontaneous movement of the head after pressing "next" will not be. The board will receive the coordinates and angle after indicating all the points.
It is not possible to use control fuducial.

vonnieda, 2020-05-15:
@tonyluken Could you take a look at this?

bacrua, 2020-05-15:
Do not understand the question? What do you have in mind?

bacrua, 2020-05-15:
Tell me, there are archives with earlier versions of PnP_v2

vonnieda, 2020-05-15:
@bacrua I was asking @tonyluken to take a look at your question because he wrote the new version, so he might know more about how it works.

No, there are no archives of previous versions. You can build from source if needed.

bacrua, 2020-05-15:
Thank. I will wait.

tonyluken, 2020-05-16:
@bacrua,

On the Job tab in the Boards panel, did you actually set the board location to x = 300mm Y = 100mm at 45 degree rotation before you began?  And are you sure that you actually have the board (CAD) origin located at those machine coordinates?  You need to have the board origin and rotation set approximately correct before beginning.

Tony

bacrua, 2020-05-16:
@tonyluken

bacrua, 2020-05-16:
physical layout of boards. Prior to using the multi-component function, no coordinates were set. Everything is as in the previous version. Installed, pointed to the components, click "next". After each click, the feeling is that the carriage is moving to the position of the intended location of the component. BUT IT IS NOT THERE !!! And after returning to the component, nothing is calculated. The algorithm in the new version is completely changed. There were no problems in the previous version. This function worked like a clock. The carriage only moved when mounted on the center of the component.
This project has already worked, starting from version 1 and 2.1 until it is updated to a new version. Nothing has changed for me, I repeat the same steps (according to the new version, I select 3 or 4 components). But to establish the location of the board is not possible.

tonyluken, 2020-05-16:
@bacrua,
You need to set the coordinates of your board before you begin.  At step 2 above where you move the camera to the board (also be sure to jog the camera to the correct angle), click the "Set board's location to camera's current location" button.  You can then save the job so you don't need to do it again in the future. Then perform the multi component function.

Tony

bacrua, 2020-05-16:
@tonyluken You wrote this function laying down the wrong algorithm.
1. The program should facilitate the life and work of the user. Yours makes it harder. I would not say this if it were not for the correct algorithm and the operation of this function in pnp_1.
2. And what if the board has a complex configuration ????
3 Why enter the coordinates if you are now doing something through the "multi component" function?
4 The board always has to be arranged as conveniently.
5 The function in the first version differs from the one you wrote only in the number of components by which the position is determined (I’m taking it for greater accuracy). But this was not intended to complicate the life of the user.
CONCLUSION: In increasing stupid bindings, manual placement becomes faster. You are not offended, but indicating the coordinates before calculating the exact position of the board is stupid.

tonyluken, 2020-05-16:
@bacrua,
1.  Sorry if you think this makes life harder for you.  Did you actually try to use it?  Note that you don't need to set the coordinates of the board very precisely for this to work, probably just within a couple of millimeters.  I thought it is easier to manually jog the camera to one location and capture the initial board location than it is to manually jog the camera over several different placements. 
2.  I don't know what you mean.  Complexity of the board shouldn't make any difference.  You just need to capture the approximate location of the board's CAD origin.
3.  This is no different than if you were using fiducials to position the board.  Remember the purpose of both the fiducial check and the multi-place alignment is to not just compute the location of the board but to also compensate for different axis scaling and axis misorthogonality between the board and the machine.  Both need a starting point so that they can move the camera to the approximate location of the fiducial or placement.  In addition, the starting coordinates also provide a safety check against which the results of the alignment process is compared to verify nothing went seriously wrong.
4.  You are still free to arrange the board however you wish.
5.  Yes, it is different but I believe it is easier to use than it was before.

aneox, 2020-05-16:
Hi! @tonyluken you have done a great job. I love the new algorithm, it is super fast and easy now to set up board with affine interpolation if you do not have fiducials. Thank you!

@bacrua you should try it in the correct way.

bacrua, 2020-05-16:
@tonyluken and @aneox 
First of all, I want to thank for your work and the work of your like-minded people. I really appreciate this huge work without which I didn’t have PNP. But still about the main thing.
As I do in pnp_1 and I like it. The result is clear and light.
1. I turn on the machine and create a job. (I do not set the coordinates of the position of the board). No anchor points.
2. I install the board in a place where it is allotted in my car (x300 u100), but the board has a complex profile and there are no angles, and the center of the board is missing.
3. I turn on the "multi-component location" function, select the component, point the camera to the center component on the board. I perform the procedure to the end. Board position and coordinates, components are all attached and everything is accurate. No extra measurements of installations, adjustments. Everything is fast and clear and without nerves.
4. It remains to establish the coordinate Z.
5. Next, if the feeders are configured and the nozzles are aligned, we dial the board.
NOTE: since the board has a complex profile in the panel, they are not only multiplied, but also rotated, mirrored and offset. Therefore, I failed to perform the "panelization" function. But there is NOTHING DIFFICULT to quickly specify 2 binding components for the next board in the panel. And in 4 clicks I get new coordinates with sneak, etc. As easy as pie. Yes, if there are a lot of components for binding, then manually moving the head with the camera is boring. Here, your idea would be useful to move the camera to a pre-estimated location of the component. BUT only when the board is called to coordinates after the first two components.
Meaning in your functional, the first 2 components make a binding as in pnp_1, and the rest can be used to move the head for accuracy.

In version pnp_2.1, this worked as in pnp_1. It so happened that I did not save the installation and had to download the latter. For the floor for I could not adjust the position of the board. Even indicating the preliminary approximate position of the board. The head eats anywhere, only not at the specified point. Demolished pnp_2.xx set pnp_1 and did the job.

This is your brainchild and you cherish it, but I still insist that entering coordinates, bindings, twist camera grip to determine degrees, when I described to you how easy it is in PNP_1. So we will soon get to the point that instead of pressing a single button, you will need to do the calculations on the calculator and enter all the data.

bacrua, 2020-05-16:
@tonyluken Over time, for relief, I adapted to do so. I click "the position of the board on the multi components." As the first binding, I take 2 adjacent components. I'm finishing the binding. This gives a binding. Then I click "the position of the board on the multi components" and make an exact reference to the most distant components. And I’m moving it to the components not manually, but "install the camera on the component" since there is already a binding.

tonyluken, 2020-05-17:
@bacrua,
Ok, I think understand your use case.  You have multiple boards panelized in a non-rectangular grid.  That's probably not a very common use case but I'll see what I can do to add an option to allow it to work as you desire.  In the meantime, you could move your CAD origin to something on the board that is easily identifiable (like one of the placements).  You would just need to jog the camera over that location, adjust the angle, click the "Set board's location to camera's current location" button and then proceed with the multi-placement alignment.

Tony

bacrua, 2020-05-18:
@tonyluken скачал версию 
17.05.2020 изменений в моей проблеме нет.

bacrua, 2020-05-18:
[image]

tonyluken, 2020-05-18:
Sorry, in English please.

bacrua, 2020-05-18:
@tonyluken Sorry. My English so bad and I use google translator. Downloaded version 05/17/2020 there are no changes in my problem. I still can not use.

tonyluken, 2020-05-18:
@bacrua,
I'm working an a fix for it right now.  Could you send me a copy of your .board.xml file for testing purposes?

Thanks,
Tony

bacrua, 2020-05-18:
@tonyluken 
[Autostart_V4_p2.board.zip](https://github.com/openpnp/openpnp/files/4647236/Autostart_V4_p2.board.zip)

bacrua, 2020-05-18:
This is a file for one board. I want to remind you that in my case, the boards in the panel are both mirrored and 180 degrees. If in pnp_1 everything works without problems, then in pnp_2 there is no new algorithm. A preliminary indication of the approximate location did not solve the problem.

bacrua, 2020-05-18:
While messing around with "installing a board with multi-components," another inconvenient use of the program appeared. This is a quick movement of the head to a point with the mouse (while holding the left button) through the camera window. Again, in PNP_1 and PNP_2 versions (before the “setting the angle” through the camera) this inconvenience was not. I will not describe it yet, maybe version 1 will remain so. Alas, I did not save version 2 with which I liked working.

bacrua, 2020-05-18:
I think I could not convey the main essence of the problem. She's not even in weird paneling, and a malformed form of the board. The bottom line is measuring the position. To determine the position of the board, we need only 2 points on the board. We get a camera hover. Nothing more is needed for this. Everything else is in the * .board.xml file and the program. For greater accuracy, more points are possible. You can get the coordinates of additional points again with a camera. And nothing more. Nothing is needed. Pointed the camera at the center of the selected component, got the coordinates. To facilitate the manual movement of the camera, you can use the automatic movement of the camera to the position of the component. BUT ONLY when the position of the board is determined by the first 2 points.

vonnieda, 2020-05-19:
Hi @bacrua. A few things:

1. Could you please record a short video showing you using the new version, and exhibiting why it does not work well for you?
2. I have posted a version of OpenPnP that uses the old 2 point board locations to https://openpnp.s3-us-west-2.amazonaws.com/index.html?prefix=test/2020-05-19_02-01-26.eeea8d6/. Would you please download this and record a short video showing us how you use it? This will help us understand the difference between the new and old version.
3. Regarding "another inconvenient use of the program appeared." - I agree. I had actually been working to revert this change for the last few days. I have now released the change to OpenPnP 2.0 and in the test version above.

Thanks,
Jason

bacrua, 2020-05-19:
Thank you for your titanic work. I will make a video.

tonyluken, 2020-05-19:
@bacrua,
First, I totally agree with you that two placements are enough to define a board's location and rotation.  However, it is also easy for an operator to make a mistake that will ruin all the placements on a board.  Consider if you intended to use R38 as one of the two placements but accidentally missed the line and clicked on R37 instead and then you still manually jog the camera to R38.  Everything would appear to be ok but when you run the job you will discover your placements are all totally incorrect.  On a densely populated board it is also very easy to accidentally jog to the wrong placement.

Most people set the location and rotation of their boards once and save them in the job.xml file.  Then they can just place a new board (or panel) on their machine without having to set its location and rotation again.  Then they just run the multi-placement alignment (or use fiducials) to do the final alignment before running the job.  This is not difficult to do.

Are you willing to try?  If not, why not?  If you are, I attached a link to an edited version of the board file you supplied me that moved the CAD origin to R4.  From the image you posted, it appears you have eight copies of the board on the panel so you need to add eight copies to your job.  Place a the panel on your machine, for each board manually jog the camera over R4 and align the crosshairs so that they match the angle of the placement, click on the "Set board's location to camera's current location" button, repeat that for all eight boards and then save your job.   Now you can select a board and start the multi-placement alignment process.  If you save the job, you don't need to set the board location and rotation ever again as long as you always place your board or panel in approximately the same location and orientation on you machine (most people use a jig or fixture to hold their boards/panels so that it is easy to do).

[Autostart_V4_p2_1.board.xml.zip](https://github.com/openpnp/openpnp/files/4652197/Autostart_V4_p2_1.board.xml.zip)

All that said, as I stated previously, I am working on a change that will allow you manually jog to the first two placements.

Tony

bacrua, 2020-05-19:
@tonyluken 
Thank you so much, Tony. I will certainly try. I will describe all the results. I think it’s not worth doing for me, although it’s good, I, like you, want to improve and simplify the procedure. As I saw this situation.
1. I turn on "post by multi component".
2 I choose 4 or more components.
3 I click "Next". The name of the first component appears in the text box of the dialog.
4 I bring the camera to the center, manually, the head does not move.
The list contains the second component.
5 I bring the camera to the center, manually, the head does not move, the program calculates the preliminary position of the board.
6 I click "nekst", the third component is in the list and for investment, the camera moves to the center of the component, since there is already a snap.
7 If necessary, we bring it manually and click "next", the fourth component in the list, the camera moves to the center.
8 We bring the camera to the center, click Next to complete the procedure.
NOTES. Perhaps for control you need to move the camera programmatically to point 1 and 2. To learn exactly.

Vasiliy

tonyluken, 2020-05-19:
@bacrua,
See pull request #1011

bacrua, 2020-05-20:
@tonyluken @vonnieda Thanks for you work. I will testing it.

bacrua, 2020-05-20:
@tonyluken When I tried to test the file you sent (Autostart_V4_p2_1.board.xml.zip), it turned out the following.
[image]

bacrua, 2020-05-20:
@tonyluken I tested the file Autostart_V4_p2_1.board you sent. The marker point for the board was selected by P4. and 5 points are selected for installation.
1. It turned out to adjust the position of the board.
2. Very excellent accuracy !!!!
3. The head as moved sideways, and moves. Since the binding marker is defined by the board marker. You have to move it manually. This is long because there are not 2 points but more. Therefore, manually it became even more time-consuming.
4. In my case, I can’t determine the correct position of the board because it is not square, it’s difficult for me to set the degree of the board and the board marker. I can do this using different methods, BUT this does not simplify the work.

bacrua, 2020-05-20:
@vonnieda 

> I have posted a version of OpenPnP that uses the old 2 point board locations to https://openpnp.s3-us-west-2.amazonaws.com/index.html?prefix=test/2020-05-19_02-01-26.eeea8d6/. Would you please [...]

> 
Excellent, just super. If in pnp_1 I always need to remember to set the camera control “in front of the multi component function”, now I haven’t installed anything and the accuracy is excellent !!!!! I’ll shoot a video.

bacrua, 2020-05-20:
@tonyluken 

> @bacrua,
See pull request #1011

> 
Thanks for trying, but this is not a solution to the problem. I won’t even try. I will explain the reasons. Imagine I am the person who works behind the machine and performs a sequence of simple operations. Machine and software are needed to simplify automation processes. And every time I told you how to sit down at the "piano" you need to call the tuner "pianos"? Work turns to hell.
I have the lasting impression that we are not hearing each other. I describe how I see the solution, you put patches or crutches in the program and give instructions on how to make the user's life even more difficult. Even with the corrected file "Autostart_V4_p2_1.board" that you sent, the installation took more time and unnecessary movements with accuracy just like the sent program from @vonnieda

bacrua, 2020-05-20:
@vonnieda https://www.youtube.com/watch?v=njg8X3yZk8Q

tonyluken, 2020-05-20:
@bacrua,
In your test of Autostart_V4_p2_1.board, you only needed to select two points instead of five.  It's not really fair to compare the time it takes to do five points against two points.

I watched your video.  I see that you use a jig to hold your panel so you should get fairly good position repeatability from panel to panel.  However I see you are starting a new job and the video ends before you save your job.  Do you save your job?  This may be the key point you are missing.  If you did, your process would be so much easier for the next panel.  You would just need to put the next panel on the machine, and load the saved job.  Then the boards would already be in the approximately correct position and rotation.  Then you could do the multi-placement alignment (as it currently works) very easily and much more quickly than the way you are currently doing it.

With respect to pull request #1011 - Once @vonnieda has merged it into the build, you will only need to switch it to behave like the old method the first time you start openpnp.  After that it will remember your setting so there is no need to call the piano tuner every time.  If you're not willing to try it then I have no choice other than to stop trying to help you.  

Tony

bacrua, 2020-05-20:
@vonnieda https://www.youtube.com/watch?v=NC1GKkcuOcM pnp_v2 Actions as with PNP_1

bacrua, 2020-05-20:
@tonyluken
 Tony, I in no case did not want to affect your feelings. I thank you for your work. But I could not get the point. Indeed, the situation is like communication between the deaf and the blind.
1. Yes, I save the changes and everything is ok.
2. ALL at 0 this is to simulate the first setting of the position of the board.
3 As soon as there is a need to redefine the coordinates, problems arise here.
 I think I'm not the only one with crooked hands. Sooner or later, someone else will have this problem. There is no question in my salvation. You have found a solution for me. I can perform my tasks v1 or V2 from @vonnieda or your method through the wire machine.xml. Then what do we want to achieve? If we want usability, simplicity, accuracy for all users who are strong in IT or just with a soldering iron, then I am a good example. V1 I work and in the new no.

tonyluken, 2020-05-20:
@bacrua,
How many of the same panel design do you build?

Tony

bacrua, 2020-05-20:
@tonyluken 

> @bacrua,
How many of the same panel design do you build?

> 
I do not quite understand your question. It is this board that is laid out here, all 8 copies in the panel. Regarding other 4x4 boards in the panel.

tonyluken, 2020-05-20:
@bacrua,

What you've shown is a single panel.  After you have finished placing all the components on a panel you remove it from the machine and run it through a reflow process.  I'm asking how many times do you do that for a single design?

bacrua, 2020-05-21:
Weird question. How does this affect the topic under discussion? This time 2 panels.

bacrua, 2020-05-21:
@tonyluken 
Tony, for correcting boards that have already been positioned works fine. What if you do not adjust the file the machine, hml, and a drop-down menu? In two or in many?

bacrua, 2020-08-04:
Hello. Again there was a need to make boards and again the same problems. I successfully used the 2-point setup program from @vonnieda last time. In the hope that in the new version the problem of "moving to a point" has been solved (at least the ability to turn off or select at will), but not everything either. It is REALLY problematic for me to create work in OpenPnP2-2points and then use the new version.
But here's the problem, the ".openpnp2" directory is overwritten by all versions, even if I put the programs in different places.
Really you are great. Honor and praise to you. BUT I CAN'T WORK QUIELY EVERY TIME. I am very nervous and angry. Before simply installing the board and getting started, I have to spend half a day in settings and installations.
When installing in different directories, make different configuration storage directories or make a selection button "place at two points without moving or with a forward move" And as an option, remove the check of the 5mm spread between the declared position of the board and the true one.
I DON'T KNOW YOU STILL HOW TO CARRY THAT THAT REALLY PASSES OUT OF YOURSELF.

markmaker, 2020-08-04:
> But here's the problem, the ".openpnp2" directory is overwritten by all versions, even if I put the programs in different places.

With that I can help:
https://github.com/openpnp/openpnp/wiki/FAQ#how-do-i-use-a-different-config-directory

But the other problems I don't understand. Are you using a language translation service? English is not my native language but I think parts of your text do not really make sense (and frankly, some sentences might even be misunderstood as offensive). Maybe try another translator? 
https://www.deepl.com/translator 

_Mark

tonyluken, 2020-08-04:
See PR 1011.  It was merged into develop back in May.  Just follow the instructions in the PR to setup openpnp to work the way you wish.  If language is a barrier, perhaps you could post a question on the the Google Groups openpnp forum and find someone there (maybe Igor Sharov) that can help with the translation.

bacrua, 2020-08-04:
@tonyluken @markmaker Thanks

bacrua, 2020-08-04:
@tonyluken 
Yes, you understood correctly and described the problem I faced. But it is more difficult for the user to change the text in the machine.xml file than to press the "old function" or "new function" selection button.
I really like the new positioning function. And I use it. BUT...
First I open the openpnp with 2-point positioning (old function), save the job, open the openpnp m2 with the new positioning function, open the job (saved with the approximate location). Now I am re-positioning the board, but more precisely.
I understand correctly that you suggest editing the file every time I switch from the old functionality to the new functionality and vice versa.
QUESTION-PROPOSAL. Adding a button to the user interface is tricky

bacrua, 2020-08-04:
@tonyluken 
Yes, you understood correctly and described the problem I faced. But it is more difficult for the user to change the text in the machine.xml file than to press the "old function" or "new function" selection button.
I really like the new positioning function. And I use it. BUT...
First I open the openpnp with 2-point positioning (old function), save the job, open the openpnp v2 with the new positioning function, open the job (saved with the approximate location). Now I am re-positioning the board, but more precisely.
I understand correctly that you suggest editing the file every time I switch from the old functionality to the new functionality and vice versa.
QUESTION-PROPOSAL. Adding a button to the user interface is tricky
-----------------
But even manual editing of the machine.xml file makes my life calm and easy.

bacrua, 2020-08-04:
> > But here's the problem, the ".openpnp2" directory is overwritten by all versions, even if I put the programs in different places.
> 
> With that I can help:
> https://github.com/openpnp/openpnp/wiki/FAQ#how-do-i-use-a-different-config-directory

Regarding overwriting the use of the "openpnp2" directory, I do not understand why the program does not create a different directory when specifying the installation in different places. After all, "installing the program to another location" implies a separate installation. In order not to affect the working version. In addition, it is much easier to copy the settings files to a new directory without losing the program's functionality. Strange and not clear to me. Am I the only one who looks at the world differently.

closed 2020-08-05 by bacrua