openpnp/openpnp #1012: Rotation angle -10440 after job is finished

aneox, 2020-05-20:
https://github.com/openpnp/openpnp/blob/bbe2e4c518af9430da27daf97bd6f7238a2d419f/src/main/java/org/openpnp/machine/reference/ReferenceNozzle.java#L364

I think this code was merged without the tests.
After I finished the job on my machine, I found the C axis position -10440 degrees, it's not normal.
Also if nozzle rotation is 270 and I click park C, nozzle rotation becomes 360.
I have tried limit to 180 rotation option, but from the time I see big angles too.

markmaker, 2020-05-20:
Yes I noticed too. I will try to fix that in the new motion code. I
intend to use the same method as in visiual homing (G92 on Smoothie) to
reset the rotation axis to the normed angle.

_Mark

tonyluken, 2020-05-20:
This was my change to limit nozzle pirouetting.  Since it is now always taking the shortest way around, there is the distinct possibility the nozzle will "wind-up" in one direction or the other.  It may be different behavior than you've seen in the past but does it hurt anything? I guess if there is a problem, maybe we could let the nozzle "un-wind" during moves when it isn't carrying a part.

Tony

aneox, 2020-05-20:
@tonyluken if you nozzle rotation accumulated 3600 and then click position camera over the location(feeders pick position or placement on the board) with rotation 0, 
the nozzle rotates 10 times per 360 degrees

I am sure this is not the shortest way))))) you have to test this)

and this huge-angle looks strange) what about nozzle runout calibration, is there any potential problems with it?

Of course, if we are at 270 degrees and we need to move to 0, the shortest way is to rotate forward 90 degrees, not 270 backward. In this case, you should found some way to reset 360 to 0 angle.
But!, I think if the stepper motor has some micro error, it will be accumulated maybe.
The best safest short way, in my opinion, is to use a 180-degree limit.

markmaker, 2020-05-20:
> @tonyluken <https://github.com/tonyluken> if your nozzle rotation

@aenox, are you sure?

I only ever noticed that when one is re-homing. Then it will unwind
whatever it has accumulated, because OpenPnP forfeits all knowledge of
the current rotation, but the controller still has it (because the
rotation axis does not have a homing endswitch, it keeps the current
coordinate). So as soon as OpenPnP sends the next moveTo() it will do
its endless dance.

It is particularly funny when one tests the nozzle tip calibration a
gazillion times, like I did. Calibration has the property of winding up
the nozzle a lot.

Like I said, I'll fix that, driving the axis in true wrap-around style
by using G92 (or whatever your controller has) to reset the accumulated
angle back to the normed +/- 180° angle.

http://smoothieware.org/g92-cnc

_Mark

aneox, 2020-05-20:
@markmaker 
openpnp shows me million of degrees if manually jog rotation million times) I am sure)

I do not use runout calibration now, because I can't find a stable pipeline for vision, so it was just a suggestion about calibration.

G92 must-have in this case.

tonyluken, 2020-05-20:
@markmaker, @aneox,

I think @aneox is correct.  Jogging the nozzle rotation to N degrees and then clicking the position camera button over some location that has zero rotation does indeed cause the nozzle to "un-wind" all the way to get to be back to that position.  I think the problem is that the rotation was only limited in the nozzle moveTo method but it probably needs to be done at the head level.  I assume Mark is on top of that with all his upcomming changes.

That said, that really won't fix the wind-up problem (if that really is problem).  I don't think a stepper motor can accumulate errors beyond one rotation as long as it's not missing steps.  I would think the only real issue would be in the numerical range that the controller can handle.  I tried a million degrees on my TinyG and it worked ok (although it took almost a half hour to get there).  I can't find anything in the TinyG documentation about numerical limits but I'm sure it must have a limit.  I assume all other controllers would as well.

> Of course, if we are at 270 degrees and we need to move to 0, the shortest way is to rotate forward 90 degrees, not 270 backward. In this case, you should found some way to reset 360 to 0 angle.

That would be ideal but you have to reset the controller and not just openpnp's angle.  That would require adding a new gcode command, having users populate it appropriately for their controller, and changing the code to send it to the controller at the appropriate times.

If this really is an issue and needs to be fixed, I think the best way to handle the un-winding would be to use the shortest way around when a part is on a nozzle and use rotate towards zero if a part is not on a nozzle.  In other words, if we are at 270 and we need to move to 0 and a part is on a nozzle we go +90 to 360, otherwise we go -270 to 0.  That should keep angles from winding-up in either direction.

Tony

markmaker, 2020-05-21:
@aenox, I think the wound up rotation is not an operations problem per
se. Except for the "human factor" ;-).

_Mark

aneox, 2020-05-21:
@markmaker in any case, huge angles are not readable and looks strange)

markmaker, 2020-05-21:
/> > In this case, you should found some way to reset 360 to 0 angle./

/> That would be ideal but you have to reset the controller and not just
openpnp's angle. That would require adding a new gcode command, having
users populate it appropriately for their controller, and changing the
code to send it to the controller at the appropriate times./

First, it is now on the axis and it has two check-boxes:

 1. Limit to +/-180°
 2. Wrap-around

For a combination of 1. + 2. users will need to add Gcode. But it will
be a new universal SET_GLOBAL_OFFSETS_COMMAND with per-axis semantics
that is also used for visual homing (the old POST_VISION_HOME_COMMAND
still works in deprecated "legacy mode").

See also:

http://www.linuxcnc.org/docs/html/gcode/coordinates.html#_the_g92_commands

http://smoothieware.org/g92-cnc

https://github.com/synthetos/TinyG/wiki/Coordinate-Systems#offsets-to-the-offsets-g92

_Mark

Am 20.05.2020 um 23:01 schrieb tonyluken:
> In this case, you should found some way to reset 360 to 0 angle.
>
> That would be ideal but you have to reset the controller and not just
> openpnp's angle. That would require adding a new gcode [...]

markmaker, 2020-05-21:
Mailing list conversion ate the image:
[image]

tonyluken, 2020-05-21:
@markmaker,
Just curious - what does it mean when both Limit to +/-180 degrees and Wrap Around are checked?

Tony

tonyluken, 2020-05-21:
@markmaker,
I thought I saw somewhere (but I can't find it right now) you asking a question about why normalizeAngle180 includes -180 but not +180 in its range.  The only real reason I didn't include both ends is I don't like having two different numeric values representing the same angle.  Now that's probably only of a concern if someone is comparing two angles - they would just need to be careful to consider 180 = -180.  I guess I included -180 rather than +180 in the range because of an old holdover from my hardware days when I represented angles using fixed point arithmetic.  For example, an 8-bit binary number could represent any angle with 360/256 ~= 1.41 degree resolution.  Interestingly you can look at the number as ether unsigned or two's-complement and it still represents the same angle:

As unsigned numbers:
11111111 = 255 ->  358.59 degrees
11111110 = 254 -> 357.19 degrees
…
10000000 = 128 -> 180.00 degrees
01111111 = 127 -> 178.59 degrees
…
00000010 = 2 -> 2.81 degrees
00000001 = 1 -> 1.41 degrees
00000000 = 0 degrees

As two's complement numbers:
11111111 = -1 ->  -1.41 degrees which is equivalent to +358.59 degrees
11111110 = -2 -> -2.81 degrees which is equivalent to +357.19 degrees
…
10000000 = -128 -> -180 degrees which is equivalent to +180 degrees
01111111 = +127 -> 178.59 degrees
…
00000010 = 2 -> 2.81 degrees
00000001 = 1 -> 1.41 degrees
00000000 = 0 degrees

Notice how the two's complement range includes -180 but not +180.  Also note how arithmetic overflow and underflow automatically wrap to the correct angle either way you look at it.

Tony

markmaker, 2020-05-21:
> @markmaker,
> Just curious - what does it mean when both Limit to +/-180 degrees and Wrap Around are checked?

That's when the G92 comes into play. Let's assume it's at 170° and you command it to go to -170°. It will then go to 190° first but immediately do

`G92 C-170`

So it turned the short way around but still has the modulo angle. 

_Mark

markmaker, 2020-05-22:
>  Now that's probably only of a concern if someone is comparing two angles - they would just need to be careful to consider 180 = -180.

Good point! In fact I just realized I had to improve my resolution aware coordinate comparison, thanks :-)

Having said that, I reworked all (known) places where code previously compared coordinates as bare double/IEEE 754. Guess we all know this is problematic, at least as soon as some transformations happen back and forth inside `moveTo()` and `getLocation()`. The coordinates will likely compare as artificially different due to IEEE 754 artifacts. 

The comparison is now handled on the axis and it can be configured with a resolution, so you won't get nonsensical sub-sub-microstep `moveTo()`s e.g. from runout compensation (and always also triggering backlash-comp!). 

Note, that coordinates were already factually rounded to precision in previous code with the %.4f format specifier in the Gcode fragment.   

So, _after having fixed the comparison thanks to your hint_, I would argue that it doesn't matter, if -180° and +180° were both allowed. If not, i'd rather keep +180° and get rid of -180°.

But it's not that important. @vonnieda, what do you think?

_Mark

tonyluken, 2020-05-22:
@markmaker, @vonnieda,

I have no objections to changing it to keep +180 rather than -180.

Tony

markmaker, 2021-02-21:
I believe the last issue relating to this was fixed in #1082. 
> Bug-fix: Rotation coordinates (angles) must not be length unit converted to/from driver units in the GcodeDriver.

closed 2021-02-21 by markmaker