openpnp/openpnp #1016: Z change in feeder pick location, not taken into account.

JulienDevillers, 2020-05-28:
Last version of develop [branch](url)

# Problem Report
## Describe the Issue
A change by typing Z coordinate of pick location is not taken into account when "move tool to
the selected feeder's current pick location".

## Steps to Reproduce
- open the joined machine.xml.
- connect and zero the machine
- goto to feeders panel
- click on feeder2
- see that Z for pick location is -26
- click on the button "move tool to the selected feeder's current pick location."
- check on tool coordinates, in status bar that  Z is  -26
- set  Z for pick location to -28.
- click apply
- select another feeder and go back to feeder2
- see that Z for pick location is -28 as it should
- click on the button "move tool to the selected feeder's current pick location."
- check on tool coordinates, in status bar, Z is -26 while it should be -28.

## Expected Result
tool should go to the new coordinates.

[machine.zip](https://github.com/openpnp/openpnp/files/4696263/machine.zip)
[OpenPnP.0.log.zip](https://github.com/openpnp/openpnp/files/4696267/OpenPnP.0.log.zip)

Whenever possible, please use the Menu -> Help -> Submit Diagnostics feature and include the URL here. Submit Diagnostics makes it easy to include log files, screen capture, debug images, configuration files, etc.

-> bad api key.

doppelgrau, 2021-04-18:
Duplicate of #1109 ?!
Pick location is only updated after a feed operation (can argue if that's good or not, but since a feed operation can be "expensive" it makes somehow sense.)

markmaker, 2022-07-23:
`pickLocation` should be made a local variable of `getPickLocation()` initialized to `location`. It is reset in each call to `feed()` I see no sensible use of this copy.

_Mark

markmaker, 2022-07-23:
But for the `ReferenceStripFeeder` that the OP referred to, I see no redundancy. Maybe it was solved in #1031. 

I also cannot reproduce it.

_Mark

markmaker, 2022-07-23:
See #1449.

markmaker, 2022-07-23:
Closing this as no longer reproducible for `ReferenceStripFeeder` and solved by #1449 for `ReferenceDragFeeder`. 
_Mark

closed 2022-07-23 by markmaker

surabibio, 2022-07-24:
Wow and Thanks