openpnp/openpnp #1028: Eclipse Error javassist.NotFoundException

yunxiang521, 2020-08-05:
javassist.NotFoundException: org.jdesktop.beansbinding.ELProperty
	at javassist.ClassPool.get(ClassPool.java:436)
	at org.openpnp.Main.monkeyPatchBeansBinding(Main.java:89)
	at org.openpnp.Main.main(Main.java:117)

The newly configured eclipse is set according to wiki settings, and the code has not been changed. Run will prompt such an error

vonnieda, 2020-08-05:
Probably you are using JDK 11+, try 10 or less.

ian-arkver, 2020-08-05:
I searched a bit and it seems this error is fixed in newer versions of javassist. Unfortunately the referenced one (3.12.1-GA) is the latest that's on maven central. I was able to fix this by putting a new version of javassist (3.27.0-GA) into a local repo and calling that up in the pom.xml. That seems very hacky though.

ian-arkver, 2020-08-05:
By the way, the blog article linked in comments to the Main.java that does this monkeyPatch has a comment on it that the underlying beansbinding performance issue has been fixed, so maybe a newer version of that doesn't need this hack any more?

http://blog.marcnuri.com/beansbinding-performance-issue-37/

yunxiang521, 2020-08-05:
> Probably you are using JDK 11+, try 10 or less.

Thank you very much for your help. According to the method provided by vonnieda, the program can run, but the error is still prompted

yunxiang521, 2020-08-05:
> By the way, the blog article linked in comments to the Main.java that does this monkeyPatch has a comment on it that the underlying beansbinding performance issue has been fixed, so maybe a newer [...]

OK, thank you. I'll try a new version

ian-arkver, 2020-08-05:
@yunxiang521 My comments were more intended for @vonnieda to be honest. If it works for JDK version 10 you're probably better to use that rather than hack in newer versions of dependencies which might break other things.

yunxiang521, 2020-08-05:
> @yunxiang521 My comments were more intended for @vonnieda to be honest. If it works for JDK version 10 you're probably better to use that rather than hack in newer versions of dependencies which might break other things.

Yes, thank you very much for your help. I'm trying to use javassist (3.27.0-ga),

yunxiang521, 2020-08-05:
Well, after I replaced javassist (3.27.0-ga), I did not find any other new problems. I would like to thank Ian arkver and vonnieda for their help

ian-arkver, 2020-08-05:
Good news. Thanks @yunxiang521 .