openpnp/openpnp #1029: Version openpnp2 does not have a choice of action if the component is missing.

bacrua, 2020-08-05:
The openpnp2 version does not have a choice of action if the component is missing or not captured by the nozzle.

In the version openpnp v1, if the component was not on the nozzle and the camera below it detected the process stopped until a decision was made. The dialog box that appeared offered skip and ignore options. This is a very useful feature.

1. Bottom rendering must be enabled globally and for components in the feeder.
2. Insert an empty tape into the feeder or remove the component from the tape to be installed. (ensure there is no component on the nozzle during visual capture at the bottom)
3. Create a job with grabbing 2 components. One component must not be caught by the nozzle, the other must be caught by the nozzle.
4. in the diological window, click on Ok.
5. Click on the icon to start work in the main program.
6. In the dialog that appears, click OK.
7. Press the stop button in the main program.

## Expected Result
The camera below will determine that there is no component, stop the process, display a dialog box with a warning which component is missing, as well as options for solving the situation - Skip, ignore or stop.

## Actual Result
The camera detects that the component is not present on the nozzle. The process stops, a dialog box appears with information about the missing component from the selection only "ok". After clicking on OK, the dialog box disappears. The vacuum pump continues to run. You can only stop work by clicking on stop. In this case, the component that was captured by the second nozzle will be dropped.

bacrua, 2020-09-19:
@vonnieda 
@markmaker 
I faced this problem. I had to manually pick up and attach the component to the nozzle from the feeder.

trailhead, 2020-10-17:
After a failed pick, try navigating to the feeders tab, then select the feeder that just failed.  Click the button to perform a feed of that part.  Then after a successful feed, click the green arrow button.  This has been working for me.

bacrua, 2020-10-18:
@trailhead.
Thanks. Alternatively, yes will solve the problem. But I would like the correct functionality of the program.

avalim, 2020-12-25:
I do agree with @bacrua. The openpnp v1 functionality was very practical. I am not familiar with the code, but what is needed to add the original functionality back?

markmaker, 2021-02-21:
@vonnieda do you agree that this was fixed by
https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2020-12-29
?

vonnieda, 2021-02-21:
@markmaker and @bacrua: A lot has changed since v1. I'm not intending to bring the v1 functionality back but I'd like to discuss this further to see if the current or planned changes will address the problem.

@bacrua In your original report you mentioned that a dialog would pop with Skip, Ignore, or Stop. Can you tell me which of these options you would likely be using here? In other words, how would you like to go about continuing the job?

avalim, 2021-02-21:
@vonnieda <https://github.com/vonnieda> and @bacrua
<https://github.com/bacrua> : An attempt at the question: dialog would pop
with Skip, Ignore, or Stop. Can you tell me which of these options you
would likely be using here? In other words, how would you like to go about
continuing the job?
Answer: In order, I would prefer "Skip" to be implemented (if not already);
"Stop" and "Ignore" would be cosmetic.

   - Skip - <critical> Behavior: identity missing part/issue, give the user
   a chance to skip that part, pick and place the next part (especially if the
   parts are already loaded on the other heads). The reason for skipping is to
   deal with all issues at the end of the job. As long as I have a list of the
   parts skipped, I can always review what happened, fix the issue, and run
   the job again to place skipped parts.
   - Stop - <cosmetic> In the case that I have a major issue that caused
   the missed part. There is a normal workaround to stop the job already in
   place.
   - Ignore - <not common in my setup> In my system, the cameras are quite
   accurate in identifying issues. At times, it happens that the parts are
   correct and the camera missed them. Meaning: ignore would just place the
   parts loaded in the heads and continue placement.

Please, does this answer your question?

Armando

avalim, 2021-02-21:
@vonnieda <https://github.com/vonnieda> and @bacrua
<https://github.com/bacrua> : <added a few clarifications> An attempt at
the question: dialog would pop with Skip, Ignore, or Stop. Can you tell me
which of these options you would likely be using here? In other words, how
would you like to go about continuing the job?
Answer: In order, I would prefer "Skip" to be implemented (if not already);
"Stop" and "Ignore" would be cosmetic.

   - Skip - <critical> Behavior: identity missing part/issue, give the user
   a chance to skip that part, place all parts loaded in the other heads,
   continue to pick and place the next parts. The reason for skipping is to
   deal with all issues at the end of the job. As long as I have a list of the
   parts skipped, I can always review what happened, fix the issue, and run
   the job again to place skipped parts.
   - Stop - <cosmetic> In the case that I have a major issue that caused
   the missed part. There is a normal workaround to stop the job already in
   place.
   - Ignore - <not common in my setup> In my system, the cameras are quite
   accurate in identifying issues. At times, it happens that the parts are
   correct and the camera missed them. Meaning: ignore would just place the
   parts loaded in the heads and continue placement.

Please, does this answer your question?

Armando

vonnieda, 2021-02-22:
Thanks @avalim - I think current functionality handles this, and it will improve some soon as well. Just set the placements error handling all to Defer. Then, when there is a failure OpenPnP will discard the part and continue to finish the job. At the end you can see which placements got skipped.

The improvement soon is that the align step will also support retry before continuing on. Right now it will just try once.

avalim, 2021-02-22:
Thank you for helping!

Armando

bacrua, 2021-02-22:
Skip and Stop

MatSpy, 2021-03-08:
Hello all, 

In my case retry after vision failure is very needed. 
I have projects where we use tiny parts with small spacing. It is impossible to 
place part low height part next to placed higher one. That means that
they need to be placed in certain order. (This excludes defer option). 

Second aspect is that, with not so accurate feeder and tiny parts, misspick
occur relatively often. That means a lot of manual click to repick components. 

Repick after vision failure would be very appreciated. 

Mateusz.

closed 2022-02-03 by bacrua